FreeRTOS/iot-reference-esp32 #7: Temperature task stops publishing after OTA job

aidiaz, 2022-06-04:
Hello! I've been trying to use this reference implementation to port coreMQTT-Agent flow into an standard `esp32dev` board. The code I'm working seems to execute well on the first loop but after an OTA job is successfully implemented the temperature MQTT publication stops. 

Here some outputs from the terminal (kinda incomplete, but shows no more pubs to `/filter/TempSubPubLED`):

```
I (1757) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (2197) core_mqtt_agent_manager: WiFi connected.
I (2197) app_wifi: Connected with IP Address:192.168.0.144
I (2197) esp_netif_handlers: sta ip: 192.168.0.144, mask: 255.255.255.0, gw: 192.168.
I (2757) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (3757) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (4757) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (5757) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (6577) core_mqtt_agent_manager: Session present: 0

I (6577) sub_pub_unsub_demo: coreMQTT-Agent connected.
I (6577) ota_over_mqtt_demo: coreMQTT-Agent connected. Resuming OTA agent.
I (6577) sub_pub_unsub_demo: Task "SubPub0" sending subscribe request to coreMQTT-Ageub0 with id 1
I (6587) temp_sub_pub_and_led_control_demo: coreMQTT-Agent connected.
I (6607) core_mqtt_agent_manager: coreMQTT-Agent connected.
I (6757) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (6767) temp_sub_pub_and_led_control_demo: Received subscribe ack for topic /filter/
I (6767) temp_sub_pub_and_led_control_demo: Sending publish request to agent with meskName": "TempSubPubLED", "temperatureValue": 65.000000, "iteration": 0}}" on topic "/
I (6787) temp_sub_pub_and_led_control_demo: Task TempSubPubLED waiting for publish 0 
I (7757) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (8757) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (9757) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (9957) ota_over_mqtt_demo: Subscribed to topic $aws/things/2462abddb070/jobs/notify

I (9957) AWS_OTA: Subscribed to MQTT topic: $aws/things/2462abddb070/jobs/notify-next
D (9967) AWS_OTA: MQTT job request number: counter=0
I (10757) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (11757) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (12757) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (13217) sub_pub_unsub_demo: Subscribe 1 for topic filter /filter/SubPub0 succeeded 
I (13227) sub_pub_unsub_demo: Task "SubPub0" sending publish request to coreMQTT-Agen "/filter/SubPub0" with ID 2.
I (13237) sub_pub_unsub_demo: Task "SubPub0" waiting for publish 2 to complete.
I (13757) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (14757) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (15757) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (16537) temp_sub_pub_and_led_control_demo: Rx'ed ack for QoS1 publish from Tx to /f
I (16597) temp_sub_pub_and_led_control_demo: Received incoming publish message {"temppSubPubLED", "temperatureValue": 65.000000, "iteration": 0}}
I (16757) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (17757) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (18757) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (19757) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (20017) ota_over_mqtt_demo: Sent PUBLISH packet to broker $aws/things/2462abddb070/

D (20017) AWS_OTA: Published MQTT request to get the next job: topic=$aws/things/2462
I (20017) ota_over_mqtt_demo: Received job message callback, size 635.

W (20037) AWS_OTA: OTA Timer handle NULL for Timerid=0, can't stop.
D (20047) AWS_OTA: OTA Event Sent.
D (20047) AWS_OTA: Executing handler for state transition:
D (20057) AWS_OTA: Added event message to OTA event queue.
I (20057) AWS_OTA: Current State=[WaitingForJob], Event=[RequestJobDocument], New sta
D (20067) AWS_OTA: OTA Event received
D (20077) AWS_OTA: Found valid event handler for state transition: State=[WaitingForJ
D (20087) AWS_OTA: JSON received: {"clientToken":":2462abddb070","timestamp":16543662bms-ota-we-1405","status":"IN_PROGRESS","statusDetails":{"self_test":"ready","updated366010,"startedAt":1654366045,"lastUpdatedAt":1654366192,"versionNumber":10,"executioa":{"protocols":["MQTT"],"streamname":"AFR_OTA-ec887f97-e528-41af-ab2d-0d800755085e",ates","filesize":845232,"fileid":0,"certfile":"/certs/root_cert_auth.pem","sig-sha2560Jzjl1zju17zz3zbqa128720AiAyGF6vt/wGw10m7EJqMiqH3Uik73Ho+zdSohog/QWEBQ=="}]}}}}
I (20147) AWS_OTA: Extracted parameter: [key: value]=[execution.jobId: AFR_OTA-bms-ot
D (20157) AWS_OTA: Identified parameter: [ execution.statusDetails.self_test ]
I (20157) AWS_OTA: Extracted parameter: [key: value]=[execution.statusDetails.updated
I (20167) AWS_OTA: Extracted parameter: [key: value]=[execution.jobDocument.afr_ota.s41af-ab2d-0d800755085e]
I (20187) AWS_OTA: Extracted parameter: [key: value]=[execution.jobDocument.afr_ota.p
I (20197) AWS_OTA: Extracted parameter: [key: value]=[filepath: /device/updates]
I (20197) AWS_OTA: Extracted parameter: [key: value]=[filesize: 845232]
I (20207) AWS_OTA: Extracted parameter: [key: value]=[fileid: 0]
I (20217) AWS_OTA: Extracted parameter: [key: value]=[certfile: /certs/root_cert_auth
I (20227) AWS_OTA: Extracted parameter [ sig-sha256-ecdsa: MEQCIDCG3B4A/IG8dei17mjO0J
I (20237) AWS_OTA: In self test mode.
[... 153 lines cut ...]
I (96787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (97787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (98787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (99787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (100787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (101787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (102787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (103787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (104787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (105787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (106787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (107787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (108787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (109787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (110787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (111787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (112787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (113787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (114787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (115787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (116787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (117787) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
```

Hope to get an answer here, thanks!!

jasonpcarroll, 2022-06-07:
Hi @aidiaz,
Sorry for the late reply. You will see on this (line)[https://github.com/FreeRTOS/iot-reference-esp32c3/blob/5586634458495dc6a94a638f1a9deaac6f04b67b/main/demo_tasks/ota_over_mqtt_demo/ota_over_mqtt_demo.c#L584], when OTA receives a job, it sends an event using the ESP-EVENT library to notify any event handler registered with this event to do something. Since OTA uses the same coreMQTT-Agent that all other demo tasks use to send MQTT messages, these tasks block themselves until OTA finishes, either on a success where the device restarts, or if OTA fails and sends out a fail event found on this (line)[https://github.com/FreeRTOS/iot-reference-esp32c3/blob/5586634458495dc6a94a638f1a9deaac6f04b67b/main/demo_tasks/ota_over_mqtt_demo/ota_over_mqtt_demo.c#L534]. The reason it behaves this way is because the sockets being used for the connection are currently blocking due to a bug in ESP-TLS's non-blocking sockets, which causes severe delay issues with OTA when there is a lot of MQTT traffic coming from other tasks, thus other tasks that use the same coreMQTT agent are blocked while OTA downloads the update. (This)[https://github.com/FreeRTOS/iot-reference-esp32c3/blob/5586634458495dc6a94a638f1a9deaac6f04b67b/main/demo_tasks/temp_sub_pub_and_led_control_demo/temp_sub_pub_and_led_control_demo.c#L609] is the handler for the temp pub sub demo. You will notice that this sets an event group bit, which is blocked on before any MQTT operation, (for example)[https://github.com/FreeRTOS/iot-reference-esp32c3/blob/5586634458495dc6a94a638f1a9deaac6f04b67b/main/demo_tasks/temp_sub_pub_and_led_control_demo/temp_sub_pub_and_led_control_demo.c#L543). The callback for job events can be found (here)[https://github.com/FreeRTOS/iot-reference-esp32c3/blob/5586634458495dc6a94a638f1a9deaac6f04b67b/main/demo_tasks/ota_over_mqtt_demo/ota_over_mqtt_demo.c#L500) and (this)[https://github.com/aws/ota-for-aws-iot-embedded-sdk/blob/666241d0f643b07d5146a3715b649d80f8135e0b/source/ota.c#L2333] is where the OTA library sends out the OtaJobEventReceivedJob event. Possible solutions here are to unblock the task by sending out the OTA stopped event when it receives an OtaJobEventStartTest, or when it receives OtaJobEventReceivedJob, to parse the JSON by casting pvData argument of the callback function to an (OtaJobDocument_t)[https://aws.github.io/aws-iot-device-sdk-embedded-C/202103.00/libraries/aws/ota-for-aws-iot-embedded-sdk/docs/doxygen/output/html/struct_ota_job_document__t.html#a5d8e1a7e8f8c32efebeaba6a119c98bc] then parsing the job document for if the job is in self-test. Let me know if this helps. I remember testing this fairly extensively for OTA success, but its definitely possible I missed this issue. Hopefully this helps - I will try this out on a C3 and see if its an issue on my end as well and come up with a proper fix that will probably be one of the two above. You mentioned this was for an esp32dev board, so I'm wondering what changes you may have made as well, so if possible, please share a repo with your code. Hope this helps and thank you for reaching out.

Best,
Jason Carroll

aidiaz, 2022-06-07:
@jasonpcarroll I've been reading the code and got to similar conclusion about coreMQTT-Agent event group blocking the other task and been trying to tell if this is desired behavior or not.

Your answer is pretty broad so I will review all the info. 

About the porting to `esp32dev` it was mostly straightforward except the wifi provisioning and digital signature components. Can you review this https://github.com/aidiaz/iot-reference-esp32c3 (dunno if public or private or need access). 

Thank you!!

jasonpcarroll, 2022-06-07:
So the behavior of blocking the other tasks is indeed desired because otherwise the tasks clog up the coreMQTT agent command queue while OTA is running, which can lead to massive delays in an OTA download, depending on the timing of operations since the socket is blocking. What I mean by this is that whenever a read or write request is made by coreMQTT Agent, the call blocks coreMQTT Agent from doing other operations until it successfully sends or receives or times out for each send or recv operation. So when a number of tasks are enqueing commands to the coreMQTT Agent to send an MQTT packet, each one of those requests blocks coreMQTT-Agent when it sends because the socket itself is blocking. If the socket is non-blocking, this becomes much less of an issue because coreMQTT agent can just send/recv without waiting for it to successfully go through, allowing it to get back to handling command callbacks for each of the commands given to it by other tasks. coreMQTT-Agent is essentially just a wrapper around coreMQTT that provides thread-safety for coreMQTT by serializing requests from tasks into a command queue - unfortunately this queue is not a form of priority queue where we can give specific tasks (like OTA) more priority on the connection. So if you have the temp pub sub demo, the subpubunsub demo, and OTA all heavily utilizing this command queue, each becomes throttled, especially OTA if its trying to download an update. I will take a look when I have a bit more time. Thank you!

aidiaz, 2022-06-07:
Absolutely desirable the use of event groups on this kind of system, no doubt. My question is about the exact event where after a successful OTA update won't reset the event group bits of the coreMQTT-Agent so the other tasks won't get past the wait (that's what I understood from your answer and the little exp with the code itself).

aidiaz, 2022-06-07:
@jasonpcarroll 

1. Unblock the task by sending out the OTA stopped event when it receives an OtaJobEventStartTest -> Seems to work just fine, now both subpub and temperature tasks working as expected.
2. Or when it receives OtaJobEventReceivedJob, to parse the JSON by casting pvData argument of the callback function to an [OtaJobDocument_t](https://aws.github.io/aws-iot-device-sdk-embedded-C/202103.00/libraries/aws/ota-for-aws-iot-embedded-sdk/docs/doxygen/output/html/struct_ota_job_document__t.html#a5d8e1a7e8f8c32efebeaba6a119c98bc) then parsing the job document for if the job is in self-test -> I'm not sure how to implement this, as a OTA started event is sent when a OtaJobEventReceivedJob event is received, should post both events (start then stop)?

Thanks for your time!

Andrés

jasonpcarroll, 2022-06-07:
Awesome - so for 2. when otaAppCallback is called for OtaJobEventReceivedJob, it also passes a reference to an OtaJobDocument_t which has a pointer to the jobs JSON data (OtaJobDocument_t.pJobDocJson). Unfortunately, this struct does not pass the information that the job document is in self test as a member, but you could parse the JSON of the Job document for statusDetails.self_test and check if its marked "ready" or something along those lines. The reason I recommend this as an alternative because we would want to differentiate a self-test job doc from a normal job doc when sending out the OTA started event, instead of just sending out an OTA stopped when in self-test. The first solution may be fine, but my initial worry is there may be a time the device is in self-test and receives a job doc for another job.

jasonpcarroll, 2022-06-07:
Honestly, I wouldn't worry too much about solution 2. I will look more into this and update here with a PR.

jasonpcarroll, 2022-06-07:
Could you tell me the specific esp32 dev kit you are using?

aidiaz, 2022-06-08:
Actually I'm not using any of the espressif's boards just an nodemcu esp32 board.

aidiaz, 2022-06-08:
Just tested it on a custom esp-pico-d4 board and seems to work just fine too

closed 2022-06-29 by jasonpcarroll