FreeRTOS/iot-reference-esp32 #11: [General] How to change the target

borch84, 2022-09-07:
**Briefly summarize the issue being raised**
Hello, I am adapting this iot reference for running it in ESP32, when I run idf.py set-target esp32 it does not change the target. 

**Describe the desired outcome**
I should be able to build the binary with esp32 target

**System information**
- Hardware board: esp32 devkit
- IDE used: 4.4.2
- Operating System: Linux
- Code version: v202204.00-7-gd4cd4f7
- Project/Demo: iot-reference

**References**
```
$idf.py set-target esp32
$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/borch/Documents/Robotica/esp/iot-reference-esp32c3/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Component directory /home/borch/Documents/Robotica/esp/iot-reference-esp32c3/components/esp-aws-iot/libraries/common does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/borch/Documents/Robotica/esp/iot-reference-esp32c3/components/esp-aws-iot does not contain a CMakeLists.txt file. No component will be added
-- Building ESP-IDF components for target esp32c3
Processing 1 dependencies:
[1/1] idf (4.4.2)
-- Project sdkconfig file /home/borch/Documents/Robotica/esp/iot-reference-esp32c3/sdkconfig
Loading defaults file /home/borch/Documents/Robotica/esp/iot-reference-esp32c3/sdkconfig.defaults...
-- App "FeaturedFreeRTOSIoTIntegration" version: v202204.00-7-gd4cd4f7-dirty
-- Adding linker script /home/borch/Documents/Robotica/esp/iot-reference-esp32c3/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/borch/Documents/Robotica/esp/esp-idf/components/esp_system/ld/esp32c3/sections.ld.in
-- Adding linker script /home/borch/Documents/Robotica/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.ld
-- Adding linker script /home/borch/Documents/Robotica/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld
-- Adding linker script /home/borch/Documents/Robotica/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld
-- Adding linker script /home/borch/Documents/Robotica/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib.ld
-- Adding linker script /home/borch/Documents/Robotica/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.version.ld
-- Adding linker script /home/borch/Documents/Robotica/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib-time.ld
-- Adding linker script /home/borch/Documents/Robotica/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.eco3.ld
-- Adding linker script /home/borch/Documents/Robotica/esp/esp-idf/components/soc/esp32c3/ld/esp32c3.peripherals.ld
-- Components: Device-Defender-for-AWS-IoT-embedded-sdk Device-Shadow-for-AWS-IoT-embedded-sdk Jobs-for-AWS-IoT-embedded-sdk app_trace app_update asio backoffAlgorithm bootloader bootloader_support bt cbor cmock coap console coreHTTP coreJSON coreMQTT coreMQTT-Agent corePKCS11 cxx driver efuse esp-tls esp32c3 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_secure_cert_mgr esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread ota-for-aws-iot-embedded-sdk partition_table protobuf-c protocomm pthread qrcode riscv sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb unity usb vfs wear_levelling wifi_provisioning wpa_supplicant ws2812_led
-- Component paths: /home/borch/Documents/Robotica/esp/iot-reference-esp32c3/components/esp-aws-iot/libraries/Device-Defender-for-AWS-IoT-embedded-sdk /home/borch/Documents/Robotica/esp/iot-reference-esp32c3/components/esp-aws-iot/libraries/Device-Shadow-for-AWS-IoT-embedded-sdk /home/borch/Documents/Robotica/esp/iot-reference-esp32c3/components/esp-aws-iot/libraries/Jobs-for-AWS-IoT-embedded-sdk /home/borch/Documents/Robotica/esp/esp-idf/components/app_trace /home/borch/Documents/Robotica/esp/esp-idf/components/app_update /home/borch/Documents/Robotica/esp/esp-idf/components/asio /home/borch/Documents/Robotica/esp/iot-reference-esp32c3/components/esp-aws-iot/libraries/backoffAlgorithm /home/borch/Documents/Robotica/esp/esp-idf/components/bootloader /home/borch/Documents/Robotica/esp/esp-idf/components/bootloader_support /home/borch/Documents/Robotica/esp/esp-idf/components/bt /home/borch/Documents/Robotica/esp/esp-idf/components/cbor /home/borch/Documents/Robotica/esp/esp-idf/components/cmock /home/borch/Documents/Robotica/esp/esp-idf/components/coap /home/borch/Documents/Robotica/esp/esp-idf/components/console /home/borch/Documents/Robotica/esp/iot-reference-esp32c3/components/esp-aws-iot/libraries/coreHTTP /home/borch/Documents/Robotica/esp/iot-reference-esp32c3/components/esp-aws-iot/libraries/coreJSON /home/borch/Documents/Robotica/esp/iot-reference-esp32c3/components/esp-aws-iot/libraries/coreMQTT /home/borch/Documents/Robotica/esp/iot-reference-esp32c3/components/esp-aws-iot/libraries/coreMQTT-Agent /home/borch/Documents/Robotica/esp/iot-reference-esp32c3/components/esp-aws-iot/libraries/corePKCS11 /home/borch/Documents/Robotica/esp/esp-idf/components/cxx /home/borch/Documents/Robotica/esp/esp-idf/components/driver /home/borch/Documents/Robotica/esp/esp-idf/components/efuse /home/borch/Documents/Robotica/esp/esp-idf/components/esp-tls /home/borch/Documents/Robotica/esp/esp-idf/components/esp32c3 /home/borch/Documents/Robotica/esp/esp-idf/components/esp_adc_cal /home/borch/Documents/Robotica/esp/esp-idf/components/esp_common /home/borch/Documents/Robotica/esp/esp-idf/components/esp_eth /home/borch/Documents/Robotica/esp/esp-idf/components/esp_event /home/borch/Documents/Robotica/esp/esp-idf/components/esp_gdbstub /home/borch/Documents/Robotica/esp/esp-idf/components/esp_hid /home/borch/Documents/Robotica/esp/esp-idf/components/esp_http_client /home/borch/Documents/Robotica/esp/esp-idf/components/esp_http_server /home/borch/Documents/Robotica/esp/esp-idf/components/esp_https_ota /home/borch/Documents/Robotica/esp/esp-idf/components/esp_https_server /home/borch/Documents/Robotica/esp/esp-idf/components/esp_hw_support /home/borch/Documents/Robotica/esp/esp-idf/components/esp_ipc /home/borch/Documents/Robotica/esp/esp-idf/components/esp_lcd /home/borch/Documents/Robotica/esp/esp-idf/components/esp_local_ctrl /home/borch/Documents/Robotica/esp/esp-idf/components/esp_netif /home/borch/Documents/Robotica/esp/esp-idf/components/esp_phy /home/borch/Documents/Robotica/esp/esp-idf/components/esp_pm /home/borch/Documents/Robotica/esp/esp-idf/components/esp_ringbuf /home/borch/Documents/Robotica/esp/esp-idf/components/esp_rom /home/borch/Documents/Robotica/esp/iot-reference-esp32c3/components/esp_secure_cert_mgr /home/borch/Documents/Robotica/esp/esp-idf/components/esp_serial_slave_link /home/borch/Documents/Robotica/esp/esp-idf/components/esp_system /home/borch/Documents/Robotica/esp/esp-idf/components/esp_timer /home/borch/Documents/Robotica/esp/esp-idf/components/esp_websocket_client /home/borch/Documents/Robotica/esp/esp-idf/components/esp_wifi /home/borch/Documents/Robotica/esp/esp-idf/components/espcoredump /home/borch/Documents/Robotica/esp/esp-idf/components/esptool_py /home/borch/Documents/Robotica/esp/esp-idf/components/expat /home/borch/Documents/Robotica/esp/esp-idf/components/fatfs /home/borch/Documents/Robotica/esp/esp-idf/components/freemodbus /home/borch/Documents/Robotica/esp/esp-idf/components/freertos /home/borch/Documents/Robotica/esp/esp-idf/components/hal /home/borch/Documents/Robotica/esp/esp-idf/components/heap /home/borch/Documents/Robotica/esp/esp-idf/components/idf_test /home/borch/Documents/Robotica/esp/esp-idf/components/ieee802154 /home/borch/Documents/Robotica/esp/esp-idf/components/jsmn /home/borch/Documents/Robotica/esp/esp-idf/components/json /home/borch/Documents/Robotica/esp/esp-idf/components/libsodium /home/borch/Documents/Robotica/esp/esp-idf/components/log /home/borch/Documents/Robotica/esp/esp-idf/components/lwip /home/borch/Documents/Robotica/esp/iot-reference-esp32c3/main /home/borch/Documents/Robotica/esp/esp-idf/components/mbedtls /home/borch/Documents/Robotica/esp/esp-idf/components/mdns /home/borch/Documents/Robotica/esp/esp-idf/components/mqtt /home/borch/Documents/Robotica/esp/esp-idf/components/newlib /home/borch/Documents/Robotica/esp/esp-idf/components/nghttp /home/borch [... cut ...]
I don't see how to change the target when I run idf.py menuconfig. 
There is an option in SDKCONFIG that does not change the target:

```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_RISCV=y
CONFIG_IDF_TARGET="esp32c3"
CONFIG_IDF_TARGET_ESP32C3=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0005
```

I use the Espressif IDE plugin in Visual Code and it is already selected the target esp32

[image]

gitwallit, 2022-09-21:
Thank you for reaching out, and do excuse the late reply here.
You will have to make several changes to get the C3 reference integration to work with esp32; you can refer to [this previous issue ](https://github.com/FreeRTOS/iot-reference-esp32c3/issues/6) on the topic. But to get past the current error, you could check to see if the IDF_TARGET environment variable is set to a different target. If so, try setting IDF_TARGET to esp32.

borch84, 2022-09-26:
Thanks for the information @gitwallit, I set IDF_TARGET=esp32 and executed idf.py set-target esp32 and now it is using the esp32 target.

I've seem several errors related the same issue:

```
In file included from /home/borch/Documents/Robotica/esp/iot-reference-esp32c3/main/networking/mqtt/core_mqtt_agent_manager.c:43:
/home/borch/Documents/Robotica/esp/iot-reference-esp32c3/main/networking/mqtt/core_mqtt_agent_manager.c: In function 'prvBackoffForRetry':
/home/borch/Documents/Robotica/esp/esp-idf/components/log/include/esp_log.h:276:27: error: format '%lu' expects argument of type 'long unsigned int', but argument 6 has type 'uint32_t' {aka 'unsigned int'} [-Werror=format=]
 #define LOG_COLOR(COLOR)  "\033[0;" COLOR "m"
                           ^~~~~~~~~
/home/borch/Documents/Robotica/esp/esp-idf/components/log/include/esp_log.h:279:27: note: in expansion of macro 'LOG_COLOR'
 #define LOG_COLOR_E       LOG_COLOR(LOG_COLOR_RED)
```

I see several error messages, attached the build log as reference.
[build.log](https://github.com/FreeRTOS/iot-reference-esp32c3/files/9642614/build.log)

closed 2022-09-26 by borch84