FreeRTOS/iot-reference-esp32 #13: [BUG] Compiler warnings/errors - can't build no matter what i try

mjdietzx, 2022-10-08:
I am unable to build this. I've been very careful, tried a bunch of things, no luck. I am following instructions very carefully. On Ubuntu, I install deps first:
```
apt-get install -y git wget flex bison gperf python3 python3-venv cmake ninja-build ccache libffi-dev libssl-dev dfu-util libusb-1.0-0 python3-pip
```

then I install esp-idf:
```
cd ~/esp
git clone -b v4.4.2 --recursive https://github.com/espressif/esp-idf.git
esp-idf/install.sh esp32
```

Now from this directory I do
```
. $HOME/esp/esp-idf/export.sh
idf.py set-target esp32
idf.py build
```

I get pretty far into the build, but then it fails, here are some snippets of the errors I am seeing. I can't find any solutions searching around:
```
/home/md/Documents/bird-buddy-v2-esp/main/networking/wifi/app_wifi.c:218:79: warning: passing argument 3 of 'esp_event_handler_register' from incompatible pointer type [-Wincompatible-pointer-types]
     ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &event_handler, NULL));
                                                                               ^~~~~~~~~~~~~~
/home/md/esp/esp-idf/components/esp_common/include/esp_err.h:116:30: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t err_rc_ = (x);                                        \
                              ^
In file included from /home/md/esp/esp-idf/components/esp_wifi/include/esp_wifi.h:56,
                 from /home/md/Documents/bird-buddy-v2-esp/main/networking/wifi/app_wifi.c:12:
/home/md/esp/esp-idf/components/esp_event/include/esp_event.h:153:58: note: expected 'esp_event_handler_t' {aka 'void (*)(void *, const char *, int,  void *)'} but argument is of type 'void (*)(void *, const char *, long int,  void *)'
                                      esp_event_handler_t event_handler,
                                      ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
[1179/1194] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/demo_tasks/temp_sub_pub_and_led_control_demo/temp_sub_pub_and_led_control_demo.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/demo_tasks/temp_sub_pub_and_led_control_demo/temp_sub_pub_and_led_control_demo.c.obj 

...
/home/md/Documents/bird-buddy-v2-esp/main/demo_tasks/temp_sub_pub_and_led_control_demo/hardware_drivers/app_driver.h:22:10: fatal error: driver/temp_sensor.h: No such file or directory
 #include "driver/temp_sensor.h"
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[1180/1194] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/networking/mqtt/core_mqtt_agent_manager.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/networking/mqtt/core_mqtt_agent_manager.c.obj 
/home/md/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/
...

cc1: some warnings being treated as errors
[1181/1194] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/demo_tasks/sub_pub_unsub_demo/sub_pub_unsub_demo.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/demo_tasks/sub_pub_unsub_demo/sub_pub_unsub_demo.c.obj 

...

/home/md/Documents/bird-buddy-v2-esp/main/demo_tasks/sub_pub_unsub_demo/sub_pub_unsub_demo.c:764:13: note: in expansion of macro 'ESP_LOGI'
             ESP_LOGI( TAG,
             ^~~~~~~~
cc1: some warnings being treated as errors
[1186/1194] Building C object esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_core/ed25519/ref10/ed25519_ref10.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

Any ideas/ways for me to fix? Thanks

ActoryOu, 2022-10-11:
Hi @mjdietzx,
Thanks for your feedback!
Unfortunately I can't reproduce it on my environment.
I have one suggestion for you to try.

This repository is created for esp32c3, not esp32. So, could you help to install esp32c3 tool chain and build it again?
Steps:
1. esp-idf/install.sh
2. . $HOME/esp/esp-idf/export.sh
3. idf.py fullclean
4. idf.py set-target esp32c3
5. idf.py build

Please let me know the result after you try.
Thanks.

SolidStateLEDLighting, 2022-10-13:
Built for me.  I'm working on Windows with VS Code and the ESP plug-in.   I just cloned the repository recursively, opened that directory, and launched the tool.

SolidStateLEDLighting, 2022-10-13:
Hey --  I know what your problem is.    Here, I just changed target to the exp32s3 and found that a number of features are unique to the c3 device.   Above, you have your target set to esp32,  and now I see you have the undefined reference to the temp sensor.  Evidently, that c3 device has that temp sensor but the esp32 does not (at least in its library).    Just comment out the functions which are not supported, and you'll build.   The temp sensor area won't work for you (is that is a hardware unique device to the c3??).   You may also see errors related to %ld formatting in logging messages.   Change those to %d (for 32 bit integers).

jasonpcarroll, 2022-10-21:
Hi all - @SolidStateLEDLighting is spot on. Some of the functionality of this repository is exclusive to the C3 - its why esp32c3 is in the name of the repo :). I believe the ESP32-S2 should also work if you change the target as it has both the DS peripheral and temp sensor as well. I imagine you could get this to work on an ESP32 if you disable the temp and led demo, and **do not** provision using the DS peripheral - which is a config for this demo (and make sure the right configs are set for the esp_secure_crt_manager component to **not** use the DS peripheral although if I remember correctly it should disable that by default if the target doesn't support it). Let me know if this issue can be closed @mjdietzx.

closed 2022-10-21 by mjdietzx