FreeRTOS/iot-reference-esp32 #16: [Feature Request] Add support for latest ESP-AWS-IoT, ESP-IDF v5.0, and IDF Eclipse Plugin v2.7

JBHunterI, 2022-12-16:
**Is your feature request related to an issue you've encountered? If so, please elaborate.**
The 202212.00 Getting Started Guide's section 1.2 "Software Requirements" says this reference code was written for "ESP-IDF 4.4.3 or higher" but there's been several API changes introduced in [ESP-IDF](https://github.com/espressif/esp-idf/releases) v5.0, so I'd like a confirmation that this reference code fully supports the [released ESP-IDF v5.0](https://github.com/espressif/esp-idf/releases/tag/v5.0), and ideally the [IDF Eclipse Plugin v2.7](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.7.0), along with the [ESP-AWS-IoT](https://github.com/espressif/esp-aws-iot) soon to be tagged & released as 202210.01-LTS but also immediately available on its master branch (see [Request for information on support with esp-idf "release/v5.0" (CA-241) #138](https://github.com/espressif/esp-aws-iot/issues/138)).

**Describe the new feature request**
Update this reference code so it fully supports the upcoming ESP-AWS-IoT LTS release, the released ESP-IDF v5.0, and the released IDF Eclipse Plugin v2.7.

**Describe other alternatives you've considered**
Revert to using ESP-IDF v4.4.3, but that would likely end-up being something we'd soon regret in the long-term.

**Additional context**
Still in an early phase of a project, so trying to future-proof it against the many changes introduced in the latest ESP-IDF release v5.0.

shubnil, 2022-12-19:
We are looking into the same and will get back soon.

gitwallit, 2022-12-20:
Thanks for reaching out.

Regarding esp-aws-iot, the main branch of this reference implementation, and its [recent release v202212.00](https://github.com/FreeRTOS/iot-reference-esp32c3/releases/tag/v202212.00), supports 202210.01-LTS and references the commit in esp-aws-iot in which 202210.01-LTS was merged. 

esp-idf v5.0 is a major release with breaking changes across the build system, networking, peripherals, the freertos kernel, etc. and as such will require that this reference implementation be potentially updated and extensively tested to be compatible with esp-idf v5.0.

gitwallit, 2022-12-22:
There is a plan to update to esp-idf v5.0, tentatively before, or by Jan end next year.

closed 2023-01-04 by gitwallit