FreeRTOS/iot-reference-esp32 #17: [Feature Request] implement vStopOTACodeSigningDemo function

idea--list, 2022-12-19:
I have a data logging application on an ESP32S3 based board and am trying to integrate the OTA feature of this repo into my project. However i am facing memory issues after integrating this repo as the OTA service is:
- needing 2x as much memory as expected based on the settings in component config ---> AWS OTA section in menuconfig
- always sitting on a lot of memory in the background

1. First i tried to investigate how this repo impacts memory usage of my project, so i configured component config ---> AWS OTA ---> size of file data block **1**, maximum data blocks requested **1**, data buffers reserved **1** inside menuconfig.

After doing so the major memory consumers of my project are:
`app_wifi_start()`  takes about 49K of memory (can get only 41K back when i call `esp_wifi_deinit()` at the end of my program)
`app_wifi_start()` takes another 51K of memory (`esp_wifi_stop()` frees up only 2K of that, the rest is lost until i restart the board)
`xRingBufferCreate()` needs 22K needed to store logged values until those are saved to the SD card (`vRingBufferDelete()` frees that up completely after logging is finished)
My `init_sd_card()` needs 23K (`deinit_sd_card()` frees that up completely)
During data acquisition sometimes i also login to an FTP server which needs another 23K until it finishes file transfers.

So i end up with around 70K free heap space when data acquisition starts. Despite using unpractically low settings for AWS OTA, but this way i can asses at least how much free heap remains while the main part is running

2. After having the above information, i go back to AWS OTA section in menuconfig and try to dial in ca. 60KB usage for OTA by setting size of file data block **12**, maximum data blocks requested **8**, data buffers reserved **14**. No luck with these settings as the system runs out of memory as soon as it tries to create any files on the SD card...

3. After some trial & error with AWS OTA settings it seems to me that the amount of memory needed in reality is about twice as much as one has dialed in inside menuconfig. So having 70K heap free, i try to dial in 24K in AWS OTA section in menuconfig (etting size of file data block **12**, maximum data blocks requested **6**, data buffers reserved **6**). By now i am expecting it will need 48K, leaving some 20K memory free.

4. With those settings the board can create the files on the SD cards, sensor is running fine, data is written to the SD card as expected, but then the TLS component begins to run out of memory occasionally and within some minutes i begin to have connectivity issues.

5. At this point i realized there is no sense further reducing OTA memory settings as that way the OTA transfer would become unpractically slow. So what if i just would call `OTA_Shutdown(0,1)` inside my user code when data logging starts and call `vStartOTACodeSigningDemo()` again when data logging is finished? It sounded good first, but did not release any memory back to pool. Taking a look on `shutdownHandler` (gets triggered by OTA_Shutdown) it seems to me it just uses memset to clean any garbage in otaAgent but does not free up that memory back to pool. In top of that i can not find a way in my user code to free up the memory used by the eventBuffer, which takes a whopping 33K of memory with above AWS OTA settings all the time, despite IMHO this much amount of memory should not even be allocated until an OTA transfer begins.

**Describe the new feature request**
Please implement a `vStopOTACodeSigningDemo()` function inside ota_over_mqtt_demo.c that would allow freeing up all the memory used by OTA back to the pool. This would be really useful as the OTA service is sitting on a LOT of memory even if the OTA service does not need to run all the time. Currently i can not find a nice way to integrate this repo in projects that need a fair amount of memory to run reliably.

**Describe other alternatives you've considered**
I considered to alter ota.c by adding `vPortFree(&otaAgent);` at the end of `shutdownHandler()`, but would like to avoid changing the library files of the esp-aws-iot component unless absolutely necessary.

**Additional context**
Maybe i just missed something that is already implemented in the OTA library files, so any feedback is appreciated.

shubnil, 2022-12-20:
Hi, We will look into the feasibility of the same and get back.

idea--list, 2022-12-21:
@shubnil 
Ok, after reading through all the code of this repo and carefully analyzing heap usage i could identify the problem and managed to bypass it.

core_mqtt_agent_manager is closing the TLS connection and then coreMQTT establishes a new TLS connection. All this is happening every 2-3 seconds. As soon as the FTP transfer starts it takes around 23K from heap until FTP finishes leaving only a few K free of heap memory. The problem manifests if the program tries to reestablish the TLS connection when FTP still is running and there is no enough free heap.

So to bypass the problem i just placed an OTA_Suspend/OTA_Resume wrapper around my FTP logic and also implemented an extern global bool that signalizes FTP logic is active to prevent any mqtt activities in core_mqtt_agent_manager.c and in sub_pub_unsub_demo.c. Does the job and is easy to do AFTER getting familiar with all the details of the repo.

However i find my feature request is still valid as there is no project where OTA is the main feature, so OTA should run only before and after the main feature but not during that. As the repo is quite heavy on the memory, only simple projects can run the main feature and OTA in parallel.

closed 2022-12-21 by idea--list