FreeRTOS/iot-reference-esp32 #19: [Feature Request] Fleet Provisioning with Claim Example

law-ko, 2022-12-29:
Hello,

Currently the device is first provisioned by creating a thing on the AWS IoT. Is there any example or guidelines on how to do fleet provisioning with claim with this example? 

For example, should the fleet provisioning claim certificate still be compiled into a bin file like this example and create another NVS partition to store the provisioned certificate for all future connections to AWS IoT?

Thank you.

johnrhen, 2022-12-29:
Hello,

We have a Fleet Provisioning demo available which you might find useful - you can find the instructions for it here: [AWS IoT Fleet Provisioning Demo](https://freertos.org/iot-fleet-provisioning/demo.html).

Let me know if you have any additional questions.

law-ko, 2022-12-29:
Hello @johnrhen ,

This seems to be highly dependent on the Windows environment, instead of this, is there an embedded version we can follow? Previously we are following the AWS provided example [project](https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/main/demos/fleet_provisioning/fleet_provisioning_with_csr)

johnrhen, 2022-12-30:
The WinSim Fleet Provisioning demo is the only one I'm aware of which uses Fleet Provisioning for FreeRTOS. There isn't a version of that demo which runs directly on an embedded device. The WinSim version also does not demonstrate Fleet Provisioning securely, as it stores its private key in a .dat file, outside of any secure keystore.

Your original idea for storing the claim certificate/provisioned certificate in NVS would be acceptable, as long as your private key is stored securely, such as in an HSM.

law-ko, 2022-12-31:
Hello @johnrhen ,

Thanks for the response. Upon further looking into this WinSim Fleet Provisioning demo, it is following more of the style of the AWS provided example, which is less recommended for production use. In the demo examples, `xDisconnectMqttSession( &xMqttContext, &xNetworkContext )` is called to disconnect the MQTT session after fleet provisioning is complete. However, in this example, I don't find any related MQTT disconnect callbacks, or is a simple `xTlsDisconnect` is suitable?

SolidStateLEDLighting, 2023-01-01:
> Currently the device is first provisioned by creating a thing on the AWS IoT. Is there any example or guidelines on how to do fleet provisioning with claim with this example?
> 
The Thing is created when a Fleet Provisioning Template is activated by your device logging in with a generic certificate that has a provisioning role attached to it.   This low-access generic cert/key is compiled into your binary.  Once that provisioning is triggered, the system sends you the certs/keys and a certID. 

> For example, should the fleet provisioning claim certificate still be compiled into a bin file like this example and create another NVS partition to store the provisioned certificate for all future connections to AWS IoT?
> 
You store the new certs/keys in NVS.   You must log out of the system with the Generic cert/key and log back in again with the provisioned certs/key and then you may connect as a Thing to the Shadow or do what you need as a Thing.

Everything in AWS Core boils down to subscribing, publishing, handling the publish-out response, and then unsubscribing.   You need a couple of third-party tools to digest the JSON and unencrypt the OTA binary download.

law-ko, 2023-01-04:
Hello @SolidStateLEDLighting ,

Thank you for the detailed response. Yes, I understand this will be the procedure to follow for fleet provisioning. However, in this example, they are using a different connection method then the typical one seen in other examples. 

Maybe it's unclean to disconnect only using `xTlsDisconnect`?

SolidStateLEDLighting, 2023-01-04:
different connection method?      Perhaps this example looks unnecessarily overly complex?       

I only see one basic connection method for any MQTT driven AWS IOT service.

law-ko, 2023-01-13:
Using `xTlsDisconnect` will do the job.

closed 2023-01-13 by law-ko