FreeRTOS/iot-reference-esp32 #21: [BUG] Large static DRAM used for OTA_Init

law-ko, 2023-01-13:
Hello,

The `OTA_Init` function used around 60000 bytes of static DRAM, which is around 60% of the DRAM for ESP32. Is there any way to reduce this?

```
 Object File DRAM .data .rtc.data DRAM .bss IRAM0 .text & 0.vectors ram_st_total Flash .text & .rodata & .rodata_noload & .appdesc flash_total
ota_over_mqtt_demo.c.o         56         0     61240           0           0        61296        2104      2352                0          0        4512
```

Thank you

archigup, 2023-01-19:
I'm not able to reproduce this. Could you share the commands used to get that information?

When I build the project and then, in the build dir, run `make size-components`, I get the following:
```
Total sizes:
Used stat D/IRAM:  192772 bytes ( 134908 remain, 58.8% used)
      .data size:   13576 bytes
      .bss  size:   98152 bytes
      .text size:   81044 bytes
Used Flash size : 1092186 bytes
      .text     :  903016 bytes
      .rodata   :  188914 bytes
Total image size: 1186806 bytes (.bin may be padded larger)
Per-archive contributions to ELF file:
            Archive File DRAM .data .rtc.data DRAM .bss IRAM0 .text ram_st_total Flash .text & .rodata & .rodata_noload & .appdesc flash_total
               liblwip.a         21         0      3838           0         3859      115216     15294                0          0      130531
           libnet80211.a        906         0      6059        1454         8419      104414     10969             7515          0      125258
         libmbedcrypto.a        136         0        84          78          298       84908     12541                0          0       97663
                 libbt.a        137         0      5278         680         6095       83148      7565                0          0       91530
                  libc.a          8         0         4           0           12       73460      7714                0          0       81182
           libbtdm_app.a        282         0       390        8902         9574       49310      6750                0          0       65244
     libwpa_supplicant.a         12         0      1016           0         1028       52232      1290                0          0       53534
                 libpp.a       2456         0      1083        3538         7077       35312      2810              100          0       44216
libota-for-aws-iot-embed        320         0      2159          16         2495       24104     17750                0          0       42190
            libmbedtls.a         60         0       200           0          260       31460      4884                0          0       36404
                libphy.a       1582         0       281        8108         9971       23264         0                0          0       32954
  libwifi_provisioning.a          4         0        20           0           24       17930     12610                0          0       30544
               libmain.a         61         0     72352           0        72413       15958     11028                0          0       27047
             libdriver.a        132         0        79        2062         2273       18022      4518                0          0       24734
          libprotocomm.a          0         0       247           0          247       11500      9620                0          0       21120
           libfreertos.a          8         0      2532       15812        18352         194      4610                0          0       20624
          libspi_flash.a       1978         0       160       10030        12168        2900      2081                0          0       16989
           libcoreMQTT.a          0         0         0           0            0       13924      2981                0          0       16905
     libesp_hw_support.a        171        16        35        5134         5356        9398      1562                0          0       16281
          libnvs_flash.a          0         0        24           0           24       15536       480                0          0       16016
         libprotobuf-c.a         12         0         0           0           12       13102       563                0          0       13677
         libesp_system.a        189         0       125        1698         2012        6866      3593                0          0       12346
                libvfs.a        304         0        64           0          368       11016       930                0          0       12250
               libheap.a       1892         0         8        6518         8418        2008      1295                0          0       11713
           libmbedx509.a          0         0         0           0            0        9818      1367                0          0       11185
             libqrcode.a          0         0         0           0            0        7614      2385                0          0        9999
                libhal.a         85         0         0        5850         5935        3644       112                0          0        9691
            libesp-tls.a          0         0         4           0            4        5898      3318                0          0        9216
          libesp_netif.a         12         0        25           0           37        6830      2033                0          0        8875
         libesp_common.a          0         0         0           0            0          60      7463                0          0        7523
     libcoreMQTT-Agent.a          0         0       529           0          529        5202      1710                0          0        6912
        libesp_ringbuf.a          0         0         0        4444         4444           0      2185                0          0        6629
           libesp_wifi.a        528         0        30         434          992        4104      1510                0          0        6576
[... cut ...]
      libtcpip_adapter.a          0         0        17           0           17         250         0                0          0         250
               libmesh.a          0         0         4           0            4          64         0                0          0          64
             libesp_pm.a          0         0         0           0            0          28         0                0          0          28
            libesp_rom.a          0         0         0          24           24           0         0                0          0          24
Built target size-components
```

law-ko, 2023-01-19:
Hi @archigup ,

I have used `idf.py size` and `idf.py size-files` are the two commands that I used.

This is the output of only enabling pubsub demo on ESP32C3
```
Total sizes:
Used stat D/IRAM:  110918 bytes ( 216762 remain, 33.8% used)
      .data size:   13400 bytes
      .bss  size:   33648 bytes
      .text size:   63870 bytes
Used Flash size :  984512 bytes
      .text     :  828670 bytes
      .rodata   :  155586 bytes
Total image size: 1061782 bytes (.bin may be padded larger)
```

This is the output for enabling OTA on ESP32C3:
```
Used stat D/IRAM:  177390 bytes ( 150290 remain, 54.1% used)
      .data size:   13792 bytes
      .bss  size:   97344 bytes
      .text size:   66254 bytes
Used Flash size : 1048556 bytes
      .text     :  869306 bytes
      .rodata   :  178994 bytes
Total image size: 1128602 bytes (.bin may be padded larger)
```
Which additional 66,472 bytes of D/IRAM are consumed. 

I am working on ESP32 directly and it shows a large increase in DRAM partition.

When OTA not enabled on ESP32:
`Used static DRAM:   74748 bytes (  49832 remain, 60.0% used)`

When OTA enabled on ESP32:
`Used static DRAM:  138492 bytes ( -13912 remain, 111.2% used) Overflow detected! You can run idf.py size-files for more information.`

The ESP32 result is based on my current project, however, 63,744 bytes of DRAM are being consumed after OTA is enabled. 

So regardless of whether its on ESP32 or ESP32C3, enabling the OTA will consume lots of DRAM space.

archigup, 2023-01-20:
Hey, seems the majority of that 60k (about 56k) is the OTA eventBuffer in `ota_over_mqtt_demo.c`.
It stores a number of OTA data buffers equal to the number of parallel OTA block requests.

You can reduce the size by reducing CONFIG_MAX_NUM_OTA_DATA_BUFFERS (the number of parallel requests; 1 or 2 might be a good value), or reducing CONFIG_LOG2_FILE_BLOCK_SIZE (The log base 2 of the data block size).

closed 2023-02-27 by law-ko