FreeRTOS/iot-reference-esp32 #22: AWS FreeRTOS support ESP32-DevKitC

Yoimer, 2023-01-22:
Hello everybody, hope you're doing great!!

Currently I only see for the [AWS Quick Connect Demos](https://freertos.org/quickconnect/esp32c3-quickconnect/) support for `ESP32-C3-DevKitC-02`. 

Even the official AWS FreeRTOS GitHub is deprecated [amazon-freertos](https://github.com/aws/amazon-freertos)

In 2021 I was able to create a project on `ESP32-DevKitC` which had the 202007 version. I was able to run MQTT, WiFi Provisioning Over BLE and OTA flawlessly. I was pretty active on GitHub asking for support and helping a bit. I was able to install some sensors and even a GPS on the chip.

So my question is: will there be support for AWS FreeRTOS `ESP32-DevKitC`? 

How could I start testing these new features if I currently do not have any `ESP32-C3-DevKitC-02`?

What if I want to start something like that with the new version of AWS FreeRTOS without making any migration for `ESP32-C3-DevKitC-02`?

I would really appreciate any comments. Looking forward to response.

Thanks in advance!!

gitwallit, 2023-01-22:
The AWS FreeRTOS components in this reference integration should be independent of the type of ESP32 chip. 
With a few changes, you should be able to build the projects for the ESP32-DevKitC. Some of the changes that I can think of are:
1/The toolchain/target (eg. $idf.py set-target esp32. You may also have to set the env variable IDF_TARGET=esp32)
2/Disable/remove the certificate manager, DS peripheral
3/Disable the temperature sensor

jasonpcarroll, 2023-01-23:
Hi @Yoimer,

This comes up a lot, so will dedicate sometime this week to creating a short doc that shows how to use this repo with the ESP32. The main reason its not immediately compatible is because this example uses some hardware features that exist on the ESP32-C3 and ESP32-S2, but do not exist on the ESP32. These features would be the Digital Signature peripheral, the temperature sensor, the LED control, etc, so as pointed out earlier, you just need to disable these things / remove their use from the code in order for it to work on the ESP32. When I worked on this, I made most of these feature uses configurable from the menuconfig - there are configurations to disable DS peripheral use, and you can disable the temperate / LED demo.

I should also mention that espressif also maintains https://github.com/espressif/esp-aws-iot which supports the ESP32.

Best,

Jason Carroll

Yoimer, 2023-01-23:
Hello @jasonpcarroll thanks a lot for responding and explaining, I really appreciate it.

Looking forward to reading that coming doc.

Thanks in advance!

Yoimer, 2023-01-26:
@jasonpcarroll Hi again :)

I think I am almost there, I mean from changing the code from `ESP32-C3` to `ESP`. I will explain what I did in a way that is easy to understand, hopefully.

Basically I forked this very same repo, including the submodules. It's public here [iot-reference-esp32c3-fork](https://github.com/Yoimer/iot-reference-esp32c3-fork)

The ESP-IDF I am using comes from here [esp-idf-v4.4.3](https://dl.espressif.com/github_assets/espressif/esp-idf/releases/download/v4.4.3/esp-idf-v4.4.3.zip). It has a folder that includes the `.git` file and submodules needed for the compilation and building

I forced esp32 target, removed the DS peripheral, and the temperate / LED demo  as seen here [sdkconfig](https://github.com/Yoimer/iot-reference-esp32c3-fork/blob/28251dcc115a05efeee020164be40ddb24f15e10/sdkconfig#L6) issuing the `idf.py set-target esp32` command

When issuing `idf.py build `, the error is this [build-esp-32-error](https://github.com/Yoimer/iot-reference-esp32c3-fork/blob/28251dcc115a05efeee020164be40ddb24f15e10/esp-32-error.txt).

Would you be kind enough to tell me what it's missing, please?

Thanks a lot for supporting us!.

Looking forward to response.

gitwallit, 2023-01-26:
For now, you can move ahead and get around this by modifying your code so that the format specifiers in the log function match the type being passed.
Eg. change the format type from %ld to %u

Yoimer, 2023-01-26:
@gitwallit Thanks for responding. In fact I trivially did what you suggested on` sub_pub_unsub_demo.c`  as seen here [sub_pub_unsub_demo.c](https://github.com/Yoimer/iot-reference-esp32c3-fork/blob/db85240790f91bcdf9730ecae734d1b9f79a3406/main/demo_tasks/sub_pub_unsub_demo/sub_pub_unsub_demo.c)

I changed  the format type from `%ld` to `%u` and no success. 

The error after building can be see here [from %ld to %u error](https://github.com/Yoimer/iot-reference-esp32c3-fork/blob/db85240790f91bcdf9730ecae734d1b9f79a3406/esp-32-error.txt).

Why the compiler shows this logs errors by just changing the target from `ESP32-C3` to `ESP32`?

I was reading an issue similar like this here [ESP_LOGI](https://github.com/espressif/esp-protocols/commit/71401a0f2ffb5a0b374a692e50e4fa7d0e12397b), but still it is not clear for me.

Should I just wait for @jasonpcarroll document or is there anything else I could try on my own?

Thanks a lot for helping and supporting.

Looking forward to your response.

Thanks in advance!

Yoimer, 2023-01-29:
Hello @jasonpcarroll @gitwallit . Hope you're doing great. I figured it out on my own. Here it is the code in case someone else needs it.

[ESP32 Formating](https://github.com/Yoimer/iot-reference-esp32c3-fork/commit/093057fed3fd4c83cdfa519132f7fa2fc77868ce). Basically the changes that have to be made, using `ESP-IDF v4.4.3` are here [sub_pub_unsub_demo.c](https://github.com/Yoimer/iot-reference-esp32c3-fork/blob/093057fed3fd4c83cdfa519132f7fa2fc77868ce/main/demo_tasks/sub_pub_unsub_demo/sub_pub_unsub_demo.c) and here [core_mqtt_agent_manager.c](https://github.com/Yoimer/iot-reference-esp32c3-fork/blob/093057fed3fd4c83cdfa519132f7fa2fc77868ce/main/networking/mqtt/core_mqtt_agent_manager.c#L678)

closed 2023-01-29 by Yoimer