FreeRTOS/iot-reference-esp32 #23: [BUG] couldn't get hostname for AWS endpointipconfig print dns addd

Lance-Kraakman, 2023-01-22:
**Describe the bug**
Please provide a clear and concise description explaining the bug.

**System information**
- Hardware board: esp32c3
- IDE used: 4.4
- Operating System: Windows
- Code version: (run ``git describe --tags`` to find it)
- Project/Demo: [ mqtt_demo_mutual_auth | shadow_demo | custom application | etc... ]
- If your project is a custom application, please add the relevant code snippet(s) in the section titled `"Code to reproduce bug"`.

**Expected behavior**
A clear description of the expected behavior.

**Screenshots or console output**
I (112490) core_mqtt_agent_manager: WiFi connected.
I (112490) app_wifi: Connected with IP Address:192.168.1.91
I (112490) esp_netif_handlers: sta ip: 192.168.1.91, mask: 255.255.255.0, gw: 192.168.1.254
I (112500) wifi_prov_mgr: STA Got IP
I (112500) app_wifi: Provisioning successful
I (112510) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
E (112540) esp-tls: couldn't get hostname for :arn:aws:iot:ap-southeast-2:****:thing/thing2: getaddrinfo() returns 202, addrinfo=0x0
E (112540) esp-tls: Failed to open new connection
I (112860) core_mqtt_agent_manager: Retry attempt 1.
E (112860) esp-tls: couldn't get hostname for :arn:aws:iot:ap-southeast-2:*****:thing/thing2: getaddrinfo() returns 202, addrinfo=0x0
E (112860) esp-tls: Failed to open new connection
I (113320) core_mqtt_agent_manager: Retry attempt 2.

**Steps to reproduce bug**
    Example:
      1. "I am using project [ ... ], and have configured with [ ... ]"
      2. "When run on [ ... ], I observed that [ ... ]"

**Code to reproduce bug**
The code should be wrapped in the ``cpp`` tag in order to be displayed clearly. For example:

```cpp
printf("Hello World")
```

**Additional context**
Include any additional context relevant to the issue here.

archigup, 2023-01-23:
Hey, the MQTT_ ENDPOINT has the wrong value. That should be the AWS IoT Core device data endpoint.

You can get the endpoint by going to https://us-west-2.console.aws.amazon.com/iot/home?region=us-west-2#/settings (and change to correct region). On the IoT Core settings page, the first section has the endpoint URL.

Lance-Kraakman, 2023-01-23:
Thanks for the quick reply and help, sorry about that.

closed 2023-01-23 by Lance-Kraakman