FreeRTOS/iot-reference-esp32 #24: Problems building due to cbor

cloisterbell42, 2023-01-25:
Hi,

I'm very new to AWS & Free RTOS. I've done quite a bit of work with ESP32 on Arduino but none with native compiler etc. I came across this demo when researching how to get into AWS IoT Core and found the instructions excellent thank you. However I have hit a brick wall in terms of my understanding and was wondering if someone more knowledgable than me could help out please?

Followed instructions as far as [https://github.com/FreeRTOS/iot-reference-esp32c3/blob/main/GettingStartedGuide.md#3-build-and-flash-the-demo-project](url)

Initially had a problem building due to the cbor component not being found. Followed instructions here to try and resolve this.

[https://components.espressif.com/components/espressif/cbor](url) by running:

```bash
idf.py add-dependency "espressif/cbor^0.6.0"
```
The above fixed that problem but now build seems to have multiple issues and the build output ends with:

```text
_HINT: The issue is better to resolve by replacing format specifiers to 'PRI'-family macros (include <inttypes.h> header file).
ninja failed with exit code 1, output of the command is in the /home/jason/esp/iot-reference-esp32c3/build/log/idf_py_stderr_output_26098 and /home/jason/esp/iot-reference-esp32c3/build/log/idf_py_stdout_output_26098_
```
The output of the build is in build.txt attached below.

**System information**
- Hardware board: ESPRESSIF ESP32-C3-DevKitC-02
- IDE used: Command Line
- Operating System: Ubuntu 20.04.5 LTS
- Code version: v202212.00-2-ged74af2
- Project/Demo: iot-reference-esp32c3

**Steps to reproduce bug**
idf.py -p /dev/ttyUSB0 flash monitor > build.txt

[build.txt](https://github.com/FreeRTOS/iot-reference-esp32c3/files/10500924/build.txt)

Many thanks...

Yoimer, 2023-01-25:
@cloisterbell42 Hello. Try using another ESP-IDF version. In my case it was fixed by using `ESP-IDF v4.4.3`
When I used the latest `5.1-dev-3025-g49551cc48c` I was struggling with the same issue you mention.

jasonpcarroll, 2023-01-25:
Hi @cloisterbell42,

As @Yoimer pointed out, it's most likely an issue with your ESP-IDF version. This repo has been mainly tested on 4.4.x. That said, what version of ESP-IDF are you using that is not working? @Yoimer, did you try with the release version of 5.0 or just the dev version?

Best,

Jason Carroll

jasonpcarroll, 2023-01-25:
Thank you @Yoimer :)

cloisterbell42, 2023-01-26:
Hi @Yoimer & @jasonpcarroll 

Thank you so much for your advice. You were both right, I was using the dev version of ESP-IDF. I hadn't realised that the master branch was the latest dev version. 

Have switched to tags/v5.0 and it seems to be working now.

Thanks again,
Jason

closed 2023-01-26 by cloisterbell42