FreeRTOS/iot-reference-esp32 #25: [Question] Is this a right supported_key_size for ESP32?

Yoimer, 2023-01-29:
Hello guys, hope you're doing quite well.

As you guys may know because of the previous issue [AWS FreeRTOS support ESP32-DevKitC](https://github.com/FreeRTOS/iot-reference-esp32c3/issues/22), I wanted to port the ESP32-C3 to ESP32 and I did it, because of your guys guidance.

About the certificates, I updated this [Old configure_esp_secure_cert.py](https://github.com/espressif/esp_secure_cert_mgr/blob/78ec22786064eb8c957087bc435ba5bb83758233/tools/configure_esp_secure_cert.py) as it is here [New configure_esp_secure_cert.py](https://github.com/Yoimer/iot-reference-esp32c3-fork/blob/d29c5270ef4942e34eacf7ba3904e3f679506adf/configure_esp_secure_cert.py)

This modification let me issue this command: `python components/esp_secure_cert_mgr/tools/configure_esp_secure_cert.py -p PORT --keep_ds_data_on_host --ca-cert CA_CERT_FILEPATH --device-cert DEVICE_CERT_FILEPATH --private-key PRIVATE_KEY_FILEPATH --target_chip esp32 --secure_cert_type cust_flash` and it worked ok.

After building, flashing and provisioning I was able to send some data to AWS as seen here: [esp32-sending.log](https://github.com/FreeRTOS/iot-reference-esp32c3/files/10529155/esp32-sending.log).

My question is:  Is this [New configure_esp_secure_cert.py](https://github.com/Yoimer/iot-reference-esp32c3-fork/blob/d29c5270ef4942e34eacf7ba3904e3f679506adf/configure_esp_secure_cert.py) a right supported_key_size for ESP32?

I just copied and pasted the  values of `esp32s2` for testing and worked pretty good.

In the past I used to copy and paste the IoT Thing certificates as referenced on the official documentation , [freertos-prereqs](https://docs.aws.amazon.com/freertos/latest/userguide/freertos-prereqs.html)

```
Open demos/include/aws_clientcredential.h.

Specify values for the following constants:

#define clientcredentialMQTT_BROKER_ENDPOINT "Your AWS IoT endpoint";

#define clientcredentialIOT_THING_NAME "The AWS IoT thing name of your board"

```
Looking forward to response.

Thanks in advance!

jasonpcarroll, 2023-01-30:
Hi @Yoimer,

Yoimer, 2023-01-30:
Hi, @jasonpcarroll. How is it going?

jasonpcarroll, 2023-01-30:
Sorry I hit enter for a linebreak haha. But I editted my comment above.

Yoimer, 2023-01-30:
I see @jasonpcarroll . Thanks for responding. Is it too much of a security concern having those keys in plain text, or I can just move on and not worry about them as you just commented out?

What if I untick the `Output provisioned certificates and keys` option from `idf.py menuconfig`? I guess the keys would still be plain text but not showed on the console.

Another question, how would you encrypt them just in case you do want to have those super secure?

Looking forward to response.

Thanks for supporting us!

jasonpcarroll, 2023-01-31:
When it comes to security for those keys, you need to evaluate what's the worst thing that could happen if someone got access to them. For example, if the certificate allowed someone to access sensitive data, then you would probably want those as secure as possible. You could always get a secure element such ass https://www.microchip.com/en-us/product/atecc608a if you really want to keep those keys private. The output provisioned certificates and keys option was more of a debugging tool than anything else - that said you probably don't want that enabled in released devices :P. 
The ESP32 supports [flash encryption](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html) so at the very least you could store them in encrypted flash rather than plaintext flash. But I should mention it has known vulnerabilities (e.g. https://raelize.com/blog/espressif-systems-esp32-bypassing-flash-encryption/) You can also use flash encryption https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html

closed 2023-01-31 by Yoimer