FreeRTOS/iot-reference-esp32 #28: [BUG] Received an unsolicited publish  during OTA .../update/accepted

txf-, 2023-02-17:
Describe the bug
When running OTA accepted messages, from AWS don't seem to be handled. I'm getting lots of warnings.

```
W (779115) core_mqtt_agent_manager: WARN:  Received an unsolicited publish from topic $aws/things/mything/jobs/AFR_OTA-tt3/update/accepted
I (779185) ota_over_mqtt: Sent PUBLISH packet to broker $aws/things/mything/jobs/AFR_OTA-tt3/update to broker.
I (779185) ota_over_mqtt: OTA Event processing completed. Freeing the event buffer to pool.
I (779195) ota_over_mqtt: Sent PUBLISH packet to broker $aws/things/mything/streams/AFR_OTA-9ca58abd-ec6d-4d6d-8e9f-cd605cda9f1e/get/cbor to broker.
```

System information

Hardware board: - Esp32 WROVER (v3) custom board.
IDE used: VSCode with IDF extension
Operating System: [ Windows 11 WSL2 with Docker image ]
Code version: 202212.00 release, IDF 4.4.3
Project/Demo: ota_over_mqtt_demo
Expected behavior
is there supposed to be an accepted handler 

Steps to reproduce bug
Run OTA, I changed TLS connection to be non-blocking and also had to set otaconfigMAX_NUM_BLOCKS_REQUEST  = 1 (OTA seems to kill the TLS connection if I leave it at the default value of 8)

Code to reproduce bug
add `.non_block = true` to `xEspTlsConfig` in components/esp-aws-iot/libraries/coreMQTT/port/network_transport/network_transport.c

set `#define otaconfigOTA_UPDATE_STATUS_FREQUENCY    1U`  in components/esp-aws-iot/libraries/ota-for-aws-iot-embedded-sdk/config/ota_config.h

Change  `CONFIG_MAX_NUM_BLOCKS_REQUEST` to 1 in sdkconfig

AniruddhaKanhere, 2023-02-19:
Hello @txf-,

Thank you for reaching out to us. We are looking into this and someone from our team will reach out to you soon.

Thanks,
Aniruddha

ActoryOu, 2023-02-22:
Hi @txf-,
Thanks for your feedback. Although it doesn't hurt the flow for now, it truely makes users confused.

I left the reason of these wanrnings here:
> From section **"Report job execution status"** in [Jobs workflow documentation](https://docs.aws.amazon.com/iot/latest/developerguide/jobs-workflow-jobs-online.html), topic "$aws/things/mything/jobs/AFR_OTA-tt3/update/accepted" means the AWS cloud get the update from device and it works successfully currently. And device doesn't subscribe the topic because it doesn't need to handle this topic.

I'll try to patch recently. Again, thanks for your contribution!

ActoryOu, 2023-03-08:
Closing due to patch merged.

closed 2023-03-08 by ActoryOu