FreeRTOS/iot-reference-esp32 #30: [BUG] Instant update on shadow documents

law-ko, 2023-02-27:
When attempting to provide instant thing shadow document updates when the device state is changed, it seems like the `MQTTAgent_CommandLoop( &xGlobalMqttAgentContext )` will crash due to LoadProhibited.

The example we made is based on the PubSubUnSub example.

The current attempt to provide instant updates to AWS IoT thing shadow when device state is changed:
```
while (1)
	{
		if (xQueueReceive(xQueue, &updateType, portMAX_DELAY))
		{
			/* Publish to ShadowUpdate topics with device status. */
			sprintf( (char *) pcPayloadBuffer, "{\"state\":{\"reported\":{\"status\":\"%d\", \"led\":{\"power\":%d}}}}", 1, 0);
			prvPublishToTopic( xQoS, pcPubTopicBuffer, (char *) pcPayloadBuffer );
		}
    }
```

Is this the good approach on providing instant status updates? Adding a delay of `vTaskDelay( pdMS_TO_TICKS( 100 ) );` at the end of the while loop would solve the problem, but would cause large delays on updates.

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400916d0  PS      : 0x00060730  A0      : 0x801d4120  A1      : 0x3ffea240  
0x400916d0: memchr at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/string/memchr.c:120

A2      : 0x00000006  A3      : 0x00000000  A4      : 0x00003de2  A5      : 0x3f43b944  
A6      : 0x00000008  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffea210  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffea494  A13     : 0x00000000  
A14     : 0x3ffb6f78  A15     : 0x00000008  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000006  LBEG    : 0x40092111  LEND    : 0x40092121  LCOUNT  : 0xfffffffd  
0x40092111: strlen at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:84

0x40092121: strlen at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:96

Backtrace:0x400916cd:0x3ffea2400x401d411d:0x3ffea260 0x401d5933:0x3ffea570 0x401e9205:0x3ffea5a0 0x40097753:0x3ffea5d0 0x400f3954:0x3ffea620 0x400f3441:0x3ffea650 0x400f36b0:0x3ffea6a0 0x400e156d:0x3ffea6d0 0x40093dd9:0x3ffea6f0 
0x400916cd: memchr at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/string/memchr.c:118

0x401d411d: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1521

0x401d5933: vprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vprintf.c:34 (discriminator 5)

0x401e9205: esp_log_writev at /home/espuser/esp/esp-idf/components/log/log.c:200

0x40097753: esp_log_write at /home/espuser/esp/esp-idf/components/log/log.c:210

0x400f3954: MQTTAgentCommand_Publish at /home/espuser/esp-git/esp-homekit-sdk-mfi-3/libraries/esp-aws-iot/libraries/coreMQTT-Agent/coreMQTT-Agent/source/core_mqtt_agent_command_functions.c:79 (discriminator 13)

0x400f3441: processCommand at /home/espuser/esp-git/esp-homekit-sdk-mfi-3/libraries/esp-aws-iot/libraries/coreMQTT-Agent/coreMQTT-Agent/source/core_mqtt_agent.c:584

0x400f36b0: MQTTAgent_CommandLoop at /home/espuser/esp-git/esp-homekit-sdk-mfi-3/libraries/esp-aws-iot/libraries/coreMQTT-Agent/coreMQTT-Agent/source/core_mqtt_agent.c:1059

0x400e156d: prvMQTTAgentTask at /home/espuser/esp-git/esp-homekit-sdk-mfi-3/examples/im/outletctrl_impm_aws/build/../main/aws_iot/networking_mqtt/core_mqtt_agent_manager.c:516

0x40093dd9: vPortTaskWrapper at /home/espuser/esp/esp-idf/components/freertos/port/xtensa/port.c:131
```

dachalco, 2023-02-28:
Hi @law-ko 

Waiting on a kernel blocking call is best for fast consumption. In this case something may be bottlenecked/effected. To verify, adding a delay fixed the issue?. Looking at the stack trace it appears the MQTT task is attempting to process the publish, but strlen/memchr seem to be giving issues, as though there's an unterminated string or incorrectly reported length.

Are you able to provide a reference of the full modified demo, or outline steps so I can exactly reproduce?

kar-rahul-aws, 2023-05-17:
Hi @law-ko 
Could you please provide the full demo or the steps to reproduce as suggested by @dachalco .
Thanks

cookpate, 2023-06-09:
Closing due to lack of information.

closed 2023-06-09 by cookpate