FreeRTOS/iot-reference-esp32 #32: [BUG] default timeout is too short for reestablishing TLS connection

idea--list, 2023-03-01:
**Describe the bug**
I am trying to integrate OTA feature of this repo into my project. Have more difficulties doing so than expected. Suggestions inside #28 address some of the unexpected things. My project sends log files to an FTP server and it seems that FTP connection and TLS connection are not the best friends...

So i decided to call `xTlsDisconnect` in my user code that triggers closing the TLS connection. `prvCoreMqttAgentConnectionTask` will detect the disconnection and will try to reestablish TLS connection. However it almost always times out before being successful.
The default timeout value `configMQTT_AGENT_CONNACK_RECV_TIMEOUT_MS` used by `MQTT_Connect` at line 629 in core_mqtt_agent_manager.c seems to be too short for reestablishing TLS connection in real-life situations. When i increase that value to 10000 TLS connection can always be reestablished before timing out. However setting such a large timeout for any other ACKs may slow things down. I have the impression adding a new timeout setting in menuconfig that is used exclusively while TLS connection is being reestablished would be a good solution.

**System information**
Adafruit Feather ESP32-S3
Visual Studio Code
ESP-IDF v4.4.3
Windows 10

**Expected behavior**
I do not expect timing outs prevent reestablishing TLS connections in average real-life conditions (common flat with 4 rooms)

dachalco, 2023-03-01:
The timeout value propagates [prvCoreMqttAgentConnect](https://github.com/FreeRTOS/iot-reference-esp32c3/blob/main/main/networking/mqtt/core_mqtt_agent_manager.c#L626) -> [MQTT_Connect](https://github.com/FreeRTOS/coreMQTT/blob/main/source/core_mqtt.c#L2670) -> [receiveConnack](https://github.com/FreeRTOS/coreMQTT/blob/main/source/core_mqtt.c#L2254). In `receiveConnack` there's [a check](https://github.com/FreeRTOS/coreMQTT/blob/main/source/core_mqtt.c#L2305) to exit the, otherwise timed, loop if any MQTT packet is received. So it should just raise timing tolerance for ACKs. ACKs that made it with the shorter timeout should still exit with similar timing, as their received CONNACK will exit the timed loop as mentioned.

If CONNACK fails to come in time, the timing of that thread's other functionality will be delayed by a full `configMQTT_AGENT_CONNACK_RECV_TIMEOUT_MS` duration. The demo is using exponential backoff though, so if an endpoint was consistently failing, the stall shouldn't happen again until an exponentially increasing delay - for every failure.

It can be handy to use different timeout settings, especially in your case where you're aware of a particularly slow endpoint. However it is specific to this demo code, so I'm not sure what the preferred action is here.

@ActoryOu Can you chime in?

ActoryOu, 2023-03-02:
As @dachalco mentioned, `configMQTT_AGENT_CONNACK_RECV_TIMEOUT_MS` is only used for tolerance for MQTT ACKs. So it's not for TLS connection. I'm not sure if I get your point at all. Could you share your log/change on TLS connection? Did you change `configMQTT_AGENT_CONNACK_RECV_TIMEOUT_MS` only to establish the connection?

Thanks!

idea--list, 2023-03-07:
@ActoryOu 
In the past few days i double checked everything. This repo works perfectly well in itself with the default settings. In that regard everything is as expected. However IMHO everyone will try to integrate this repo after some experimenting.

I try to integrate the OTA demo into 2 different data logging projects, i always find that the TLS connection can not be reestablished with the default settings. If i increase `configMQTT_AGENT_CONNACK_RECV_TIMEOUT_MS` either in menuconfig or at the above mentioned line of code, then it always succeeds. If i revert back to the default setting then the issue is back again. Have no idea why this happens if that setting should not affect TLS connection. I could observe this even if that is the only setting of the repo i was modifying.

On the sidenote i think it would be helpful to implement a function called `vStopOTACodeSigningDemo` as a counterpart of `vStartOTACodeSigningDemo`. Here is why: i find that this repo is too heavy on the memory, restricting quite much what else can the user do while this repo is running. For sure OTA must start at system boot time to verify the new FW image, however  it would be useful to have a nice implementation to stop it while heavy user code is running and then let OTA run again after the user session is done. Using several successive calls of OTA_Suspend() i managed to make FTP transfers work after which a single OTA_Resume() continues to run OTA, however this seems to work only for short periods on my side and it also does not free up the resources OTA is taking away.

closed 2023-03-07 by idea--list

ActoryOu, 2023-03-08:
@idea--list,
Thanks for getting these back and suggestion. We do have [OTA_Shutdown()](https://github.com/aws/ota-for-aws-iot-embedded-sdk/blob/main/source/include/ota.h#L488) in the OTA library to stop whole OTA service. But it's not common to shut it down after initialization. It would stop listening OTA topics and miss the next OTA update if it doesn't get init again. Could you provide your scenario for us as example?

Thanks again.

idea--list, 2023-03-08:
@ActoryOu 
One of the project i was integrating OTA is monitoring voltage with a samplerate of 1000Hz. It logs sensor data every 30 ms into a ringbuffer and then to a file. Periodically the files get transferred to an FTP server, while the logging is still going on. So it is basically like an online signal analyzer. The job is quite monotonic but it seems to put some load on the MCU and i can only use 2x4 KB buffer for the OTA, which makes updates really slow. Otherwise the project just runs out of memory and crashes. I am sure there are even more complicated projects than mine.

Until now i tried `OTA_Shutdown`, `xCoreMqttAgentManagerPost`, `OTA_SignalEvent`, `xTlsDisconnect`. All these did half of the job, but while trying to continue OTA i had different issues with all these (either there were errors reestablishing TLS or my FTP files arrived only partially or sometimes not at all). All these made wish to completely turn off OTA when the users starts to measure the voltages as it is not practical to interrupt their measurement with a firmware update that they did not even trigger.
That way i could even get rid of the severe restricitions due to high memory usage when running OTA and the user code in parallel. So IMHO there are many use cases where we need OTA to run at bootup, but really do not need that to run all the time. Until now i could not manage to shut OTA down releasing its resources for long periods (maybe even for hours) and then restart it in a bullet-proof way.

Maybe i just miss something, however things would be easy if the counterpart of `vStartOTACodeSigningDemo` would be implemented out of the box. Did not investigate the 2 other demos yet in this regard.