FreeRTOS/iot-reference-esp32 #33: [BUG] Featured FreeRTOS IoT Integration can't be debugged through USB Serial/JTAG interface of the ESP32-C3

JBHunterI, 2023-03-08:
**Describe the bug**
The ESP32-C3 running an ESP-IDF v4.4.4 build of the [FreeRTOS/iot-reference-esp32c3](https://github.com/FreeRTOS/iot-reference-esp32c3) cannot be debugged though the device's USB Serial/JTAG interface.  The exact same setup can debug the IDF blink example but using the same tools on the IoT Reference code does not stop at the temporary breakpoint automatically inserted at app_main(), nor does it stop at manually added breakpoints.  **I'd be interested to know how you were able too debug this code.**  Was it using the [ESP-Prog board](https://docs.espressif.com/projects/espressif-esp-iot-solution/en/latest/hw-reference/ESP-Prog_guide.html) or some other external JTAG adapter?

**System information**
- Hardware board: [ ESP32-C3-DevKitM-1 or presumably even ESP32-C3-DevKitC-02]
- IDE used: [ IDF tools in a Windows CMD prompt (idf.py, GDB, etc), Eclipse 2022-09]
- Operating System: [ Windows 10 ]
- Code version: ( recently updated to v4.4.4-148-g4c2afac355 but started with v4.4.4 )
- Project/Demo: [ temp_sub_pub_and_led_control_demo ]
- If your project is a custom application, please add the relevant code snippet(s) in the section titled `"Code to reproduce bug"`. [ Can reproduce problem with unmodified IoT Reference demo app with only temp_sub_pub_and_led_control_demo enabled & configured, or maybe even without any demo enabled ]

**Expected behavior**
Successfully connect GDB through ESP32-C3's USB Serial/JTAG interface, have it stop at app_main(), and continue just as when simply running from an unattended power-up, and when debugging the IDF blink example.

**Screenshots or console output**
See the "Additional context" section below. 

**Steps to reproduce bug**
Can be reproduced by simply using GDB/OpenOCD from the IDF CMD shell command line with the following gdbinit file:
```
file build/FeaturedFreeRTOSIoTIntegration.elf
target extended-remote :3333
monitor [target current] configure -event gdb-detach {shutdown}
set remote hardware-watchpoint-limit 2
mon reset halt
maintenance flush register-cache
thb app_main
c
```
or simply use the [ESP-IDF Eclipse Plugin](https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md).

**Code to reproduce bug**
Modify an ESP32-C3-DevKitM-1 to use its USB Serial/JTAG interface (obtain schematic & PCB docs from [ESP32-C3-DevKitM-1 User Guide](https://docs.espressif.com/projects/esp-idf/en/v4.4.4/esp32c3/hw-reference/esp32c3/user-guide-devkitm-1.html#related-documents) and follow of my 2023/March/7 comment in the ESP-IDF issue listed under "Additional context" section below). Run "idf.py menuconfig" to configure the "Component config" -> "ESP System Settings" -> "Channel for console output" = "USB Serial/JTAG Controller", the "Featured FreeRTOS IoT Integration" -> "Endpoint for MQTT Broker to use" and "Thing name" settings, build, flash, run, then try to attach to unmodified [FreeRTOS/iot-reference-esp32c3](https://github.com/FreeRTOS/iot-reference-esp32c3)

**Additional context**
More history in "ESP-IDF repo "Still can't debug an app more complicated than the blink example (IDFGH-9526)" issue #10880](https://github.com/espressif/esp-idf/issues/10880).

paulbartell, 2023-03-09:
@JBHunterI Espressif would probably be best equipped to assist you with this issue. Have you look through the esp-idf documentation?

The error message in your linked github issue for esp-idf would indicate that the watchdog timer is not getting properly disabled. The simplest way to check this is by disabling the watchdog timer in menuconfig:

Run ```idf.py menuconfig``` and de-select ESP_INT_WDT, ESP_TASK_WDT, BOOTLOADER_WDT_ENABLE.

If that doesn't address the issue, I can get an esp32c3 with an esp-prog set up tomorrow to fully validate the configuration.

JBHunterI, 2023-03-09:
Under Component config -> ESP-System Settings, I disabled "Interrupt watchdog" (ESP_INT_WDT) and "Initialized Task Watchdog Timer on startup" (ESP_TASK_WDT), and in Bootloader config I disabled "Use RTC watchdog in start code" (BOOTLOADER_WDT_ENABLE), deleted my build directory, and rebuilt. It made no difference: I still get the exact same incorrect behavior.

I don't know why you're not getting a ESP32-C3-DevKitM-1 with the appropriate mods to use the built-in USB Serial/JTAG interface. You understand that's what I'm trying to get to work, right?  Are you saying this IoT Reference code was developed with ESP-Prog, and you're just making sure that'll (still) allow GDB to work before trying the built-in USB Serial/JTAG interface?

I'm trying the opposite: to get another ESP32-C3-DevKitM-1 to work with ESP-Prog but I'm having problems getting past an OpenOCD "Error: no device found" blowing its DIS_USB_JTAG eFuse so it can use a ESP-Prog, as described in [Configure eFuses](https://docs.espressif.com/projects/esp-idf/en/v4.4.4/esp32c3/api-guides/jtag-debugging/configure-other-jtag.html#configure-efuses) (if I understand that correctly!?!?):
`
C:\Users\JeanBoucher\ESP-IDF_ws\hello_world> espefuse.py -p COM15 -c esp32c3 summary
espefuse.py v3.3.2
Connecting......................................
A fatal error occurred: Failed to connect to ESP32-C3: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Please make sure that you have specified the right port with the --port argument
`

It's really not my preferred choice to dedicate 6 pins (with 4-wire JTAG + 2-wire UART) to debug our code when only 2 pins (USB D+/-) should suffice, but I've already spent way too much time trying to get these tools to work, instead of actually developing our code. It really shouldn't be this hard.

paulbartell, 2023-03-09:
@JBHunterI I understand that you are frustrated. I am only trying to help.

An external jtag probe is often more reliable. If designing a production application, I would strongly suggest breaking out the jtag pins on your board. Your Mileage May Vary and the esp32c3 built in usb-jtag does work in most cases.

I was able to debug this application without any trouble, so i'll share the exact steps I followed to get it set up.

When I run espfuse.py on my system, I get the following output. I did not modify any of the fuses from their default values.
```
iot-reference-esp32c3 % espefuse.py -c esp32c3 summary -p /dev/tty.usbmodem1444401 
espefuse.py v3.3.2
Connecting...

=== Run "summary" command ===
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Config fuses:
DIS_ICACHE (BLOCK0)                                Disables ICache                                    = False R/W (0b0)
DIS_DOWNLOAD_ICACHE (BLOCK0)                       Disables Icache when SoC is in Download mode       = False R/W (0b0)
DIS_FORCE_DOWNLOAD (BLOCK0)                        Disables forcing chip into Download mode           = False R/W (0b0)
DIS_CAN (BLOCK0)                                   Disables the TWAI Controller hardware              = False R/W (0b0)
VDD_SPI_AS_GPIO (BLOCK0)                           Set this bit to vdd spi pin function as gpio       = False R/W (0b0)
BTLC_GPIO_ENABLE (BLOCK0)                          Enable btlc gpio                                   = 0 R/W (0b00)
POWERGLITCH_EN (BLOCK0)                            Set this bit to enable power glitch function       = False R/W (0b0)
POWER_GLITCH_DSENSE (BLOCK0)                       Sample delay configuration of power glitch         = 0 R/W (0b00)
DIS_DIRECT_BOOT (BLOCK0)                           Disables direct boot mode                          = False R/W (0b0)
DIS_USB_SERIAL_JTAG_ROM_PRINT (BLOCK0)             Disables USB-Serial-JTAG ROM printing              = False R/W (0b0)
UART_PRINT_CONTROL (BLOCK0)                        Sets the default UART boot message output mode     = Enabled R/W (0b00)
FORCE_SEND_RESUME (BLOCK0)                         Force ROM code to send a resume command during SPI = False R/W (0b0)
                                                    bootduring SPI boot                              
ERR_RST_ENABLE (BLOCK0)                            Use BLOCK0 to check error record registers         = without check R/W (0b0)
BLOCK_USR_DATA (BLOCK3)                            User data                                         
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 

Efuse fuses:
WR_DIS (BLOCK0)                                    Disables programming of individual eFuses          = 0 R/W (0x00000000)
RD_DIS (BLOCK0)                                    Disables software reading from BLOCK4-10           = 0 R/W (0b0000000)

Flash Config fuses:
FLASH_TPUW (BLOCK0)                                Configures flash startup delay after SoC power-up, = 0 R/W (0x0)
                                                    unit is (ms/2). When the value is 15, delay is 7.
                                                   5 ms                                              

Identity fuses:
SECURE_VERSION (BLOCK0)                            Secure version (used by ESP-IDF anti-rollback feat = 0 R/W (0x0000)
                                                   ure)                                              
MAC (BLOCK1)                                       Factory MAC Address                               
   = xx:xx:xx:xx:xx:xx (OK) R/W 
WAFER_VERSION (BLOCK1)                             WAFER version                                      = 3 R/W (0b011)
PKG_VERSION (BLOCK1)                               Package version                                    = ESP32-C3 R/W (0b000)
BLOCK1_VERSION (BLOCK1)                            BLOCK1 efuse version                               = 0 R/W (0b000)
OPTIONAL_UNIQUE_ID (BLOCK2)                        Optional unique 128-bit ID                        
   = 20 05 13 4d ae a2 62 84 29 19 32 03 97 ac 85 73 R/W 
BLOCK2_VERSION (BLOCK2)                            Version of BLOCK2                                  = 3 R/W (0b011)
CUSTOM_MAC (BLOCK3)                                Custom MAC Address                                
   = 00:00:00:00:00:00 (OK) R/W 

Jtag Config fuses:
SOFT_DIS_JTAG (BLOCK0)                             Software disables JTAG. When software disabled, JT = 0 R/W (0b000)
                                                   AG can be activated temporarily by HMAC peripheral
DIS_PAD_JTAG (BLOCK0)                              Permanently disable JTAG access via pads. USB JTAG = False R/W (0b0)
                                                    is controlled separately.                        

Security fuses:
DIS_DOWNLOAD_MANUAL_ENCRYPT (BLOCK0)               Disables flash encryption when in download boot mo = False R/W (0b0)
                                                   des                                               
SPI_BOOT_CRYPT_CNT (BLOCK0)                        Enables encryption and decryption, when an SPI boo = Disable R/W (0b000)
                                                   t mode is set. Enabled when 1 or 3 bits are set,di
[... cut ...]
Breakpoint 2 at 0x42007cc2: file /path/to/iot-reference-esp32c3/main/main.c, line 117.
Note: automatically using hardware breakpoints for read-only addresses.
=> continue

Thread 2 "main" hit Breakpoint 2, prvInitializeNetworkContext () at /path/to/iot-reference-esp32c3/main/main.c:117
117	{
=> step
148	    xNetworkContext.pcHostname = CONFIG_GRI_MQTT_ENDPOINT;
=> step
149	    xNetworkContext.xPort = CONFIG_GRI_MQTT_PORT;
```

Please try to provide as much detail as possible so I can assist you in debugging the issue.

Thanks much,
Paul

JBHunterI, 2023-03-09:
Paul, I don't mean to sound like I'm taking it out on you, and I apologize if I do. On the contrary I appreciate your help. 

So, your setup consists of ESP32-C3-DevKitC-02 with ESP-IDF v4.4.4 on Linux/MacOS, which should but may not be the same tool functionality, but definitely different binaries from the same IDF version on Windows 10 running on my ESP32-C3-DevKitM-1. Since you didn't modify any of the fuses from their default values, (apparently) you can't be using the 4-wire JTAG interface on IO4/5/6/7. 

The big difference I see in your picture is that you're using **_both_** the UART0 Tx/RX (CP2102's /dev/tty.usbserial-144430 from the black microUSB cable coming from the right of the picture) **_and_** the built-in USB Serial/JTAG (ESP32-C3's /dev/tty.usbmodem1444401 from the PCB-mounted USB connector at the top of the picture). So you drive different signals based on the port you specify by using 3 devices: 2 debug console interfaces (CP2102 & internal USB), and 1 (USB) JTAG interface.

Problem is, I need the UART0 Tx/RX to connect to another device's UART without any debug console traffic on that link. I wonder if try your espefuse.py command after disconnecting the PCB-mounted USB cable and changing the port from tty.usbmodem1444401 to tty.usbserial-144430? Are you setting Component config -> ESP-System Settings -> Channel for console output = "USB Serial/JTAG Controller", **_or_** leaving it to "Default: UART0" **_and_** setting Channel for console secondary  output = "USB Serial/JTAG Controller"? Since you have all 3 devices, I guess the built-in USB JTAG interface is available regardless? I was just setting my port with "set ESPPORT=COM8" environment variable in %IDF_TOOLS_PATH%\idf_cmd_init.bat, along with IDF_TARGET=esp32c3, ESPBAUD=115200, and OPENOCD_COMMANDS=-s share/openocd/scripts -f board/esp32c3-builtin.cfg, to avoid having to specifying those for every command, which won't work in this setup, and probably not in Eclipse either.

I'll give your setup a try on an unmodified ESP32-C3-DevKitM-1 with no config changes & let you know.

paulbartell, 2023-03-10:
>So, your setup consists of ESP32-C3-DevKitC-02 with ESP-IDF v4.4.4 on Linux/MacOS, which should but may not be the same tool functionality, but definitely different binaries from the same IDF version on Windows 10 running on my ESP32-C3-DevKitM-1. 

Yes. Can you check that the version of the tools you are using are the same though? The version strings of openocd and riscv32-esp-elf-gdb-no-python should be the same regardless of the platform.

> Since you didn't modify any of the fuses from their default values, (apparently) you can't be using the 4-wire JTAG interface on IO4/5/6/7.

Correct. I'm using the built in usb pins for debugging as you requested and running openocd with the ```-f board/esp32c3-builtin.cfg``` argument to use the onboard usb jtag peripheral.

> The big difference I see in your picture is that you're using both the UART0 Tx/RX (CP2102's /dev/tty.usbserial-144430 from the black microUSB cable coming from the right of the picture) and the [...]

Yesterday I encountered some very odd behavior when attempting to access the internal usb from idf monitor and openocd simultaneously, so reverted back to using the CP2102 for logging access (with CONFIG_ESP_CONSOLE_UART_DEFAULT=y in the config).

I tried this again today with CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y and I'm able to debug with gdb/openocd and follow the log with idf monitor simultaneously.

You might consider [logging via the application tracing library](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/app_trace.html#logging-to-host) if you run into a similar issue.

Thanks,
Paul

JBHunterI, 2023-03-10:
I can confirm your OpenOCD and GDB versions for the MacOS are identical to my Windows ones.

> Yesterday I encountered some very odd behavior when attempting to access the internal usb from idf monitor and openocd simultaneously, so reverted back to using the CP2102 for logging access (with CONFIG_ESP_CONSOLE_UART_DEFAULT=y in the config).

Sounds like you see problems using **_only_** the built-in USB Serial port for all console I/O, as I do, **_iff_** CONFIG_ESP_CONSOLE_UART_DEFAULT=y **_and_** CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG is not set. 
Conversely, you're saying doing the same thing but with CONFIG_ESP_CONSOLE_UART_DEFAULT not set **_and_** CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y, you see no problem debugging **_and_** logging through the built-in USB Serial port. I'm assuming you also have CONFIG_ESP_CONSOLE_UART always follow the same value as  CONFIG_ESP_CONSOLE_UART_DEFAULT. Also, my CONFIG_ESP_CONSOLE_UART_NUM=0 always.

paulbartell, 2023-03-13:
@JBHunterI I use idf.py menuconfig to ensure that the configuration is valid rather than changing it by hand. This means that 
only one of CONFIG_ESP_CONSOLE_UART_DEFAULT or CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG will ever be set to y at any given time..

>Sounds like you see problems using only the built-in USB Serial port for all console I/O, as I do, iff CONFIG_ESP_CONSOLE_UART_DEFAULT=y and CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG is not set.

I ran into issues using idf monitor and openocd concurrently with CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y. 

> Conversely, you're saying doing the same thing but with CONFIG_ESP_CONSOLE_UART_DEFAULT not set and CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y, you see no problem debugging and logging through the [...]

I was not able to reproduce the problem the next day with the same config(CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y). Perhaps this has to do with the order in which the commands were run. Either way, it will always be a bit less reliable to use the same usb device to provide both the serial and jtag functions simultaneously.

Regardless, I currently have no issues with either of these two configurations:
1. Using the CP2102 microUSB port with CONFIG_ESP_CONSOLE_UART=y for logging with idf monitor and the builtin usb-jtag for debugging with openocd
2. Using the built in usb-uart/jtag with CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y for logging idf monitor and debugging with openocd.

paulbartell, 2023-03-15:
@JBHunterI Were you able to debug the project successfully?

JBHunterI, 2023-03-15:
No, even a brand new unmodified board with a second USB cable connecting only D+/- to IO19/IO18, I get this after successfully flashing my image:
`
> idf.py monitor
Executing action: monitor
Running idf_monitor in directory C:\Users\...\test\ESPUSBSerialJTAG\myTest
Executing "C:\App\ESP\IDF\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\App\ESP\IDF\esp-idf-v4.4.4\tools/idf_monitor.py -p COM8 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 --decode-panic backtrace C:\Users\...\test\ESPUSBSerialJTAG\myTest\build\myTest.elf -m 'C:\App\ESP\IDF\python_env\idf4.4_py3.8_env\Scripts\python.exe' 'C:\App\ESP\IDF\esp-idf-v4.4.4\tools\idf.py'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM8 instead...
--- idf_monitor on \\.\COM8 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (308) esp_image: segment 4: paddr=00124b84 vaddr=40385cESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x420ac604
0x420ac604: cpu_ll_waiti at C:/App/ESP/IDF/esp-idf-v4.4.4/components/hal/esp32c3/include/hal/cpu_ll.h:204
 (inlined by) esp_pm_impl_waiti at C:/App/ESP/IDF/esp-idf-v4.4.4/components/esp_pm/pm_impl.c:830
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x16d0
load:0x403cc710,len:0x8dc
load:0x403ce710,len:0x2a4c
entry 0x403cc710
I (24) boot: ESP-IDF v4.4.4-148-g4c2afac355-dirty 2nd stage bootloader
I (24) boot: compile time 04:06:05
I (25) boot: chip revision: v0.3
I (28) boot.esp32c3: SPI Speed      : 80MHz
I (33) boot.esp32c3: SPI Mode       : DIO
I (38) boot.esp32c3: SPI Flash Size : 4MB
I (42) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00006000
I (66) boot:  1 nvs              WiFi data        01 02 00013000 00006000
I (74) boot:  2 otadata          OTA data         01 00 00019000 00002000
I (81) boot:  3 phy_init         RF data          01 01 0001b000 00001000
I (89) boot:  4 ota_0            OTA app          00 10 00020000 00190000
I (96) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
I (103) boot:  6 storage          WiFi data        01 02 00340000 00010000
I (111) boot:  7 nvs_key          NVS keys         01 04 00350000 00001000
I (119) boot: End of partition table
I (123) esp_image: segment 0: paddr=00020020 vaddr=3c0e0020 size=277a0h (161696) map
I (156) esp_image: segment 1: paddr=000477c8 vaddr=3fc94200 size=02c48h ( 11336) load
I (158) esp_image: segment 2: paddr=0004a418 vaddr=40380000 size=05c00h ( 23552) load
I (166) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=d4b5ch (871260) map
I (301) esp_image: segment 4: paddr=00124b84 vaddr=40385c00 size=0e484h ( 58500) load
I (317) boot: Loaded app from partition at offset 0x20000
I (318) boot: Disabling RNG early entropy source...
I (318) cpu_start: Pro cpu up.
I (333) cpu_start: Pro cpu start user code
I (333) cpu_start: cpu freq: 160000000
I (333) cpu_start: Application information:
I (336) cpu_start: Project name:     myTest
I (341) cpu_start: App version:      0.0.1
I (346) cpu_start: Compile time:     Mar  9 2023 04:05:22
I (352) cpu_start: ELF file SHA256:  342ecf377278f8e6...
I (358) cpu_start: ESP-IDF:          v4.4.4-148-g4c2afac355-dirty
I (365) heap_init: Initializing. RAM available for dynamic allocation:
I (373) heap_init: At 3FC9F960 len 0003CDB0 (243 KiB): DRAM
I (379) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (385) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (392) spi_flash: detected chip: generic
I (393) spi_flash: flash io: dio
I (393) sleep: Configure to isolate all GPIO pins in sleep state
I (398) sleep: Enable automatic switching of GPIO sleep configuration
I (406) coexist: coexist rom version 9387209
I (406) cpu_start: Starting scheduler.                                                 <---- green up to here
E (407) esp_secure_cert: Metadata magic word does not match        <----  all RED starting from here
E (407) esp_secure_cert: Error in reading the metadata
E (407) ./main/main.c: Error in getting device certificate. Error: ESP_FAIL
E (407) esp_secure_cert: Metadata magic word does not match
E (407) esp_secure_cert: Error in reading the metadata
E (407) ./main/main.c: Error in getting CA certificate. Error: ESP_FAIL
E (407) esp_secure_cert: Metadata magic word does not match
E (407) esp_secure_cert: Error in reading the metadata
E (407) ./main/main.c: Error in getting private key. Error: ESP_FAIL
E (407) ./main/main.c: Failed to initialize global network context.         <--- nothing after this
`

Never seen this, & have no idea why that's appearing. That's after running the blink example on it, so I'm trying to erase the entire Flash so I start from a known state. I suspect I may have to start in Download Mode from my other COM port connected to the CP2102 (COM8 is my USB-attached one), but that will just confirm that this thing can be recovered once we build our custom board.

paulbartell, 2023-03-15:
@JBHunterI flashing from openocd by default only updates the app partition. 

``idf.py flash`` should update all relevant partitions.

Have you followed the [instructions for provisioning](https://github.com/FreeRTOS/iot-reference-esp32c3/blob/main/GettingStartedGuide.md#23-provision-the-esp32-c3-with-the-private-key-device-certificate-and-ca-certificate-in-development-mode) the key / certificate onto the device?

JBHunterI, 2023-03-15:
I always use `idf.py flash` to update my app image, unless I'm in Eclipse, which I haven't been for some time now since I'm trying to get this to work at the most basic level, i.e in the IDF CMD shell (Windows cmd.exe + export.bat) command-line utilities (mostly idf.py & openocd/gdb).

Trying to erase Flash by following the Erasing the ESP32 Flash from [this](https://randomnerdtutorials.com/esp32-erase-flash-memory/) keeps giving me an error, whether I use my USB Serial COM8 or my CP2102 COM5:
C:\Users\...\ESP-IDF_ws\blink> idf.py erase-flash -p COM8
Executing action: erase-flash
Running esptool.py in directory C:\Users\...\ESP-IDF_ws\blink\build
Executing "C:\App\ESP\IDF\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\App\ESP\IDF\esp-idf-v4.4.4\components/esptool_py/esptool/esptool.py -p COM8 -b 115200 --before default_reset --after hard_reset --chip esp32c3 --no-stub erase_flash"...
esptool.py v3.3.2
Serial port COM8
Connecting...
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 84:f7:03:08:36:88
Enabling default SPI flash mode...
Erasing flash (this may take a while)...

A fatal error occurred: ESP32-C3 ROM does not support function erase_flash.
esptool.py failed with exit code 2

C:\Users\...\ESP-IDF_ws\blink> idf.py erase-flash -p COM5
Option "port" provided for "erase-flash" is already defined to a different value. This option cannot be set in command line if the ESPPORT environment variable is set to a different value.

C:\Users\...\ESP-IDF_ws\blink> set ESPPORT
ESPPORT=COM8

C:\Users\...\ESP-IDF_ws\blink> set ESPPORT=COM5

C:\Users\...\ESP-IDF_ws\blink> idf.py erase-flash
Executing action: erase-flash
Running esptool.py in directory C:\Users\...\ESP-IDF_ws\blink\build
Executing "C:\App\ESP\IDF\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\App\ESP\IDF\esp-idf-v4.4.4\components/esptool_py/esptool/esptool.py -p COM5 -b 115200 --before default_reset --after hard_reset --chip esp32c3 --no-stub erase_flash"...
esptool.py v3.3.2
Serial port COM5
Connecting....
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 84:f7:03:08:36:88
Enabling default SPI flash mode...
Erasing flash (this may take a while)...

A fatal error occurred: ESP32-C3 ROM does not support function erase_flash.
esptool.py failed with exit code 2

C:\Users\...\ESP-IDF_ws\blink> set ESPPORT=COM8

C:\Users\...\ESP-IDF_ws\blink> idf.py monitor
Executing action: monitor
Running idf_monitor in directory C:\Users\...\ESP-IDF_ws\blink
Executing "C:\App\ESP\IDF\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\App\ESP\IDF\esp-idf-v4.4.4\tools/idf_monitor.py -p COM8 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 --decode-panic backtrace C:\Users\JeanBoucher\ESP-IDF_ws\blink\build\blink.elf -m 'C:\App\ESP\IDF\python_env\idf4.4_py3.8_env\Scripts\python.exe' 'C:\App\ESP\IDF\esp-idf-v4.4.4\tools\idf.py'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM8 instead...
--- idf_monitor on \\.\COM8 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x5 (DOWNLOAD(USB/UART0/1))
Saved PC:0x4004c72e
waiting for download
'
Shouldn't I always be able to at least fall back to recover this chip by talking to the Stage 1 ROM Bootloader???

JBHunterI, 2023-03-15:
Yes, I did followed the [instructions for provisioning](https://github.com/FreeRTOS/iot-reference-esp32c3/blob/main/GettingStartedGuide.md#23-provision-the-esp32-c3-with-the-private-key-device-certificate-and-ca-certificate-in-development-mode) to program the key/certificate on my **_orignal modified board_** with the CP2102 suspended & diconnected & routing the board's USB connector D+/- to IO19/18, but not on this **_new unmodified board_** with the 2nd USB connection directly between D+/- and IO19/18. Let me do that now...

JBHunterI, 2023-03-16:
OK, did that & now it seems to be running. Still seeing the same problem though: unable to debug the project successfully.
Compare the (attached) 2 log files using the same **_unmodified board with a 2nd USB connection directly between D+/- and IO19/18 just like yours_**, once with the blink example, & the other the IoT Reference. I `start idf.py openocd` that opens a new window then I run `idf.py gdb` in this original window.  
[blink_log.txt](https://github.com/FreeRTOS/iot-reference-esp32c3/files/10985803/blink_log.txt)
[ESPUSBSerialJTAG_log.txt](https://github.com/FreeRTOS/iot-reference-esp32c3/files/10985805/ESPUSBSerialJTAG_log.txt)

You can see the OpenOCD window for the blink example seems clean, but for the IoT Reference ends with:
```
...
Error: Abstract command ended in error 'not supported' (abstractcs=0x10001202)
Error: Timed out after 5s waiting for busy to go low (abstractcs=0x10001202). Increase the timeout with riscv set_command_timeout_sec.
Error: Abstract command ended in error 'not supported' (abstractcs=0x10001202)
Error: Timed out after 5s waiting for busy to go low (abstractcs=0x10001202). Increase the timeout with riscv set_command_timeout_sec.
Info : Disabling abstract command reads from CSRs.
Error: Abstract command ended in error 'not supported' (abstractcs=0x10001202)
Error: Timed out after 5s waiting for busy to go low (abstractcs=0x10001202). Increase the timeout with riscv set_command_timeout_sec.
Error: Fatal: Failed to read MISA from hart 0.
Warn : target esp32c3 examination failed
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Error: Target not examined yet
Error executing event gdb-attach on target esp32c3:

Warn : No symbols for FreeRTOS!
Error: Target not halted
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected

```

I have no idea why I'm getting different results from yours with :
```
...
# CONFIG_ESP_CONSOLE_UART_DEFAULT is not set
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y
CONFIG_ESP_CONSOLE_SECONDARY_NONE=y
# CONFIG_ESP_CONSOLE_UART is not set
...
# CONFIG_GRI_ENABLE_SUB_PUB_UNSUB_DEMO is not set
CONFIG_GRI_ENABLE_TEMPERATURE_PUB_SUB_AND_LED_CONTROL_DEMO=y

#
# Temperature pub sub and LED control demo configurations
#
CONFIG_GRI_TEMPERATURE_PUB_SUB_AND_LED_CONTROL_DEMO_STRING_BUFFER_LENGTH=150
CONFIG_GRI_TEMPERATURE_PUB_SUB_AND_LED_CONTROL_DEMO_DELAY_BETWEEN_PUBLISH_OPERATIONS_MS=3000
CONFIG_GRI_TEMPERATURE_PUB_SUB_AND_LED_CONTROL_DEMO_MAX_COMMAND_SEND_BLOCK_TIME_MS=500
CONFIG_GRI_TEMPERATURE_PUB_SUB_AND_LED_CONTROL_DEMO_QOS_LEVEL=1
CONFIG_GRI_TEMPERATURE_PUB_SUB_AND_LED_CONTROL_DEMO_TASK_PRIORITY=1
CONFIG_GRI_TEMPERATURE_PUB_SUB_AND_LED_CONTROL_DEMO_TASK_STACK_SIZE=3072
# end of Temperature pub sub and LED control demo configurations

# CONFIG_GRI_ENABLE_OTA_DEMO is not set
```

Yoimer, 2023-03-16:
@JBHunterI Good morning. Hope you're doing great over there. Other than the external USB connector you show on the picture, have you tried the official `ESP-PROG`?. Like this official one. [ESP-Prog](https://www.aliexpress.us/item/3256804639864635.html?)

I am pretty interested on debugging these 3 examples from AWS, and I may invest some time on it.

Currently I have the ESP32C3 and ESP-Prog at my place. As a good point, I am able to debug the ESP32 using the ESP-PROG on Windows 11 through WLS2 flawlessly. I haven't tried the ESP32C3 yet, but I might.

JBHunterI, 2023-03-16:
Good point @Yoimer, here's a pic of my unmodified ESP32-C3-DevKitM-1's connection to dual USB cables, with the main USB cable used to power the kit & access its CP2102 UART/USB converter, while the extra USB cable connects only its D+/- directly to the ESP32-C3's internal USB PHY D+/- via IO19/IO18. The 10K resistor is simply a pull-up for IO2.
[image]

BTW, I used `start idf.py openocd` and `idf.py gdb` to debug  with my environment containing:
```
OPENOCD_COMMANDS=-s share/openocd/scripts -f board/esp32c3-builtin.cfg
OPENOCD_SCRIPTS=C:\App\ESP\IDF\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\share\openocd\scripts
ESPBAUD=115200
ESPPORT=COM8  <-- USB CDC ACM serial port
IDF_TARGET=esp32c3
```
as I ignored my COM5 (CP2102 USB/UART port). Line 9 of previous log files shows the USB JTAG was being used by OpenOCD.

@paulbartell could you attach your sdkconfig file so I can compare it with mine? That's the last thing I can think of.

I've got no choice but to switch to using a dedicated 4-wire JTAG on IO4/5/6/7 today, on yet another ESP32-C3-DevKitM-1 with the DIS_USB_JTAG eFuse burnt to permanently connect the USB_SERIAL_JTAG and the JTAG port of the CPU, to see if that will allow me to debug the FreeRTOS IoT Reference demo any more reliably.

JBHunterI, 2023-03-17:
OK, before I gave-up completely on it, I gave the USB Serial/JTAG interface one more try, and on both my **_original modified kit_** as well as on my **_unmodified kit_** (pictured above & ignoring its CP2102), through it I can debug & monitor the IoT Reference iff I **_don't use_** `idf.py openocd gdb` but **_rather use_** `start openocd %OPENOCD_COMMANDS% && riscv32-esp-elf-gdb -x gdbinit`, where that GDB says in purple "GNU gdb (esp-gdb) _11.2_20220823" and comes from:
`C:\App\ESP\IDF\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin\riscv32-esp-elf-gdb.exe`.

This is the same full path that's in my Eclipse Debug config's "Debugger" tab -> "Start GDB session" -> "Actual Executable" and, as the idf.py command (which ever of the several GDB executables in my PATH it) uses to start OpenOCD & GDB, --- neither work. I can't explain why but this is clearly good news for me, even if it means using GDB in text mode to debug my code. So, we can now move forward with our custom hardware designed to rely the USB Serial/JTAG interface instead of dedicating 6 more pins (TMS/TCK/TDO/TDI/DTR/RTS) just for debugging.  Hopefully along the way I'll figure-out or someone will enlighten me on what's wrong with both the idf.py OpenOCD/GDB command and the Eclipse Plugin that keeps them both from debugging more than a trivial blink example.

JBHunterI, 2023-03-17:
Thanks for your help, Paul.

closed 2023-03-17 by JBHunterI