FreeRTOS/iot-reference-esp32 #35: [DOC] idf.py menuconfig breaks when cbor not added as dependency to the project

padaszewski, 2023-04-10:
**Describe the issue**
**ESP-IDF 5+**: ```cbor``` was moved from esp-idf to the component registry [(docs)](https://docs.espressif.com/projects/esp-idf/en/v5.0.1/esp32/migration-guides/release-5.x/removed-components.html), therefore when trying to run ```idf.py menuconfig``` I got the error, that component ```cbor``` could not be resolved. Please update the DOCs for the further users.
What needs to be done before ```idf.py menuconfig```:
run ```idf.py add-dependency "espressif/cbor^0.6.0~1"``` in the project directory.
Component registry dependency cbor can be found [here](https://components.espressif.com/components/espressif/cbor).

**Reference**
Look [here](https://github.com/FreeRTOS/iot-reference-esp32c3/blob/main/GettingStartedGuide.md#22-configure-the-project-with-the-aws-iot-thing-name-and-aws-device-endpoint). Nothing about ```cbor```.

**Screenshot**
Before adding cbor:
[image]

After adding cbor:
[image]

+got menuconfig working

**Browser**
N/A

rawalexe, 2023-04-10:
Thank you for information, we will look into it and get back to you soon

archigup, 2023-04-27:
Hi, the versions of some of the libraries used in this repo are not compatible with IDF 5.0. For now, I would recommend using IDF 4.4. 

If IDF 5.0 is needed at the moment, you can try updating the OTA library to the main branch.

kar-rahul-aws, 2023-05-16:
Hi @padaszewski 
Any updates on the suggestions provided by @archigup .
Thanks

padaszewski, 2023-05-17:
Hi @kar-rahul-aws,
I've just used an older library and example from espressiff's repository. I wanted to proceed with IDF 5.0.
It may work with 4.4, however it would be good to update this to 5.0 and also add the examples for other boards.
Feel free to close the issue if You know You won't extend this.

idea--list, 2023-05-17:
@padaszewski 
Pullrequest #20 is sitting there for a while, you can take a look at it. On my side the commit of from 3rd january works almost as expected with ESP-IDF v5.0.1. The only thing i noticed is that the sub_pub_unsub demo task tends to hang randomly within the first hour. I am still investigating why.

I can not get later commits of that PR work on my side, apparently something changed about how things work with the certificates. So either that is broken or i miss what has been changed there.

Skptak, 2023-05-23:
Hey @padaszewski, the [Pull Request](https://github.com/FreeRTOS/iot-reference-esp32c3/pull/20) that @idea--list mentioned has been merged into mainline. I ran the `idf.py menuconfig` command you mentioned and am not seeing the cbor error you were referring to. I was wondering if you'd be able to try the `idf.py menuconfig` command out again to see if this fixes your issue.  If it doesn't I'd greatly appreciate it if you could provide the error your receive so we can help you debug your issue.

padaszewski, 2023-05-23:
Hi there @Skptak, sure, will do, however I'm currently busy on other topics so it can take me one or even two weeks to come back to this.

Skptak, 2023-08-18:
@padaszewski marking this as closed for now. If you run into any further issues feel free to re-open this issue or to create a new one!

closed 2023-08-18 by Skptak

padaszewski, 2023-08-18:
Hi @Skptak 
Sorry, I had no chance to try this. Doing currently other things. Maybe I can delegate this.
Closing is okay for me, hope this works now!
Thx for the information!