FreeRTOS/iot-reference-esp32 #36: [BUG] <ESP32 start Faild>

gavin-hy, 2023-04-17:
**My ESP32 boot faild**
When I refer to the steps in the [page](https://github.com/FreeRTOS/iot-reference-esp32c3/blob/main/UseSecurityFeatures.md) ,my esp32c3 stopped running.

**System information**
- Hardware board:  ESP32-C3
- IDE used: ESP-IDF v4.4.4
- Operating System: Windows 
- Code version: The latest code of the main branch
- Project/Demo: iot-reference-esp32c3

**Screenshots or console output**
Here is the ouput content.
```cpp
invalid header: 0xe92d5614
invalid header: 0xe92d5614
invalid header: 0xe92d5614
invalid header: 0xe92d5614
invalid header: 0xe92d5614
invalid header: 0xe92d5614
invalid header: 0xe92d5614
ClearCommError failed (PermissionError(13, '???????????????????????????', None, 22))
Waiting for the device to reconnect
```
Now it doesn't work no matter what program I download.

**Detailed description**
    Step:
      1. I am using project "iot-reference-esp32c3", and have connected with AWS.
      2. I referred to the page and the previous steps were fine,but when i reached 5.1,I did this 
 ```cpp
idf.py bootloader
```
then output information similar to the web page.

```cpp
Bootloader built. Secure boot enabled, so bootloader not flashed automatically.
To sign the bootloader with additional private keys.
        d:/software/Espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe D:/software/Espressif/frameworks/esp-idf-v4.4.4/components/esptool_py/esptool/espsecure.py sign_data -k secure_boot_signing_key2.pem -v 2 --append_signatures -o signed_bootloader.bin build/bootloader/bootloader.bin
Secure boot enabled, so bootloader not flashed automatically.
        d:/software/Espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe  D:/software/Espressif/frameworks/esp-idf-v4.4.4/components/esptool_py/esptool/esptool.py --chip esp32c3 --port=(PORT) --baud=(BAUD) --before=default_reset --after=no_reset --no-stub write_flash 
--flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 E:/ESP-Project/iot-reference-esp32c3/build/bootloader/bootloader.bin
```

      3. When run on
     
  ```cpp
      esptool.py --chip esp32c3 --port=COM10 --baud=460800 --before=default_reset --after=no_reset --no-stub write_flash 
--flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 E:/ESP-Project/iot-reference-esp32c3/build/bootloader/bootloader.bin
```
     4.last, run the 

```cpp
        idf.py -p COM10 flash monitor
```
the out is 
```cpp
Executing action: flash
Running ninja in directory e:\esp-project\iot-reference-esp32c3\build
Executing "ninja flash"...
[1/5] cmd.exe /C "cd /D E:\ESP-Project\iot-reference-esp32...eference-esp32c3/build/FeaturedFreeRTOSIoTIntegration.bin"
FeaturedFreeRTOSIoTIntegration.bin binary size 0x131000 bytes. Smallest app partition is 0x190000 bytes. 0x5f000 bytes (24%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D E:\ESP-Project\iot-reference-esp32c3\build\bootloader\esp-idf\esptool_py && d:\software\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe D:/software/Espressif/frameworks/esp-idf-v4.4.4/components/partition_table/check_sizes.py --offset 0xb000 bootloader 0x0 E:/ESP-Project/iot-reference-esp32c3/build/bootloader/bootloader.bin"
Bootloader binary size 0xa000 bytes. 0x1000 bytes (9%) free.
[2/3] cmd.exe /C "cd /D D:\software\Espressif\frameworks\e...sp-idf-v4.4.4/components/esptool_py/run_serial_tool.cmake"
esptool.py esp32c3 -p COM10 -b 460800 --before=default_reset --after=no_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x20000 FeaturedFreeRTOSIoTIntegration.bin 0xb000 partition_table/partition-table.bin 0x19000 ota_data_initial.bin
esptool.py v3.3.2
Serial port COM10
Connecting...
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 60:55:f9:7b:a1:00
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00020000 to 0x00150fff...
Flash will be erased from 0x0000b000 to 0x0000bfff...
Flash will be erased from 0x00019000 to 0x0001afff...
Erasing flash...
Took 1.31s to erase flash block
Writing at 0x00020000... (0 %)
Writing at 0x00020400... (0 %)
Writing at 0x00022800... (0 %)
Writing at 0x00022c00... (0 %)
Writing at 0x00023000... (1 %)
Writing at 0x00023400... (1 %)
Writing at 0x00023800... (1 %)
.
.
.
Writing at 0x0014f800... (99 %)
Writing at 0x0014fc00... (99 %)
Writing at 0x00150000... (99 %)
Writing at 0x00150400... (99 %)
Writing at 0x00150800... (99 %)
Writing at 0x00150c00... (100 %)
Wrote 1249280 bytes at 0x00020000 in 27.8 seconds (359.0 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.03s to erase flash block
Writing at 0x0000b000... (33 %)
Writing at 0x0000b400... (66 %)
Writing at 0x0000b800... (100 %)
Wrote 3072 bytes at 0x0000b000 in 0.0 seconds (522.9 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.04s to erase flash block
Writing at 0x00019000... (12 %)
Writing at 0x00019400... (25 %)
Writing at 0x00019800... (37 %)
Writing at 0x0001a400... (75 %)
Writing at 0x0001a800... (87 %)
Writing at 0x0001ac00... (100 %)
Wrote 8192 bytes at 0x00019000 in 0.1 seconds (516.0 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
Executing action: monitor
Running idf_monitor in directory e:\esp-project\iot-reference-esp32c3
Executing "D:\software\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe D:\software\Espressif\frameworks\esp-idf-v4.4.4\tools/idf_monitor.py -p COM10 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 --decode-panic backtrace e:\esp-project\iot-reference-esp32c3\build\FeaturedFreeRTOSIoTIntegration.elf -m 'D:\software\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe' 'D:\software\Espressif\frameworks\esp-idf-v4.4.4\tools\idf.py' '-p' 'COM10'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM10 instead...
--- idf_monitor on \\.\COM10 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004c97a
invalid header: 0xe92d5614
invalid header: 0xe92d5614
invalid header: 0xe92d5614
invalid header: 0xe92d5614
```

**Other  problem**
When cloning the project in China, there would be an error and I could not download the complete project. I delayed for three days and finally found that it was the problem of the project.

paulbartell, 2023-04-24:
@gavin-hy as noted in step [5.1.2](https://github.com/FreeRTOS/iot-reference-esp32c3/blob/main/UseSecurityFeatures.md#51-build-and-flash-the-secure-boot-enabled-bootloader) of the tutorial, you must manually flash the bootloader using the command output during the ```idf.py bootloader``` step. 

This command should be something like:
```
esptool.py --chip esp32c3 --port= COM10 --baud= 460800 --before=default_reset --after=no_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 build/bootloader/bootloader.bin
```

Can you try this and let me know if you can boot the device successfully?

paulbartell, 2023-04-24:
@gavin-hy The output you posted does not correspond to flashing the bootloader. Can you share output of that step?

gavin-hy, 2023-04-28:
@paulbartell  thanks for your response.  I folloowed the stp you mentioned, but it still outputs the same error. I think there's problem with the safety boot.

closed 2023-05-12 by gavin-hy