FreeRTOS/iot-reference-esp32 #37: [DOC] Certs not found

memo26167, 2023-05-29:
**Describe the issue**
Certificates not found.

Certificates are in `main/certs/` as described in [https://github.com/FreeRTOS/iot-reference-esp32c3/blob/main/GettingStartedGuide.md#23-provision-the-esp32-c3-with-the-private-key-device-certificate-and-ca-certificate-in-development-mode](url)

Had to move `esp_secure_cert_mgr` from `managed_components` to `components` otherwise `idf.py` complains, and then I ran:
`python components/esp_secure_cert_mgr/tools/configure_esp_secure_cert.py -p /dev/ttyUSB0 --keep_ds_data_on_host --ca-cert main/certs/aws-root-ca.pem --device-cert main/certs/client.crt --private-key main/certs/client.key --target_chip esp32 --secure_cert_type cust_flash`

The guide said the partition folder would be `esp_ds_data`, but `esp_secure_cert_data` was created instead.
Then I ran:
`esptool.py --no-stub --port /dev/ttyUSB0 write_flash 0xD000 esp_secure_cert_data/esp_secure_cert.bin`

And finally build and flashed the project.
**It seems the project fails because certs are not found.**

The versions that I'm using are:
- ESP-IDF v5.0.2
- esptool.py v4.5.1
and I'm using a ESP32-D0WDQ6.

I get the following log:
```
entry 0x4008064c
I (27) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (27) boot: compile time 10:39:41
I (27) boot: chip revision: v1.0
I (31) boot.esp32: SPI Speed      : 40MHz
I (35) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 4MB
I (44) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (68) boot:  1 nvs              WiFi data        01 02 00013000 00006000
I (75) boot:  2 otadata          OTA data         01 00 00019000 00002000
I (83) boot:  3 phy_init         RF data          01 01 0001b000 00001000
I (90) boot:  4 ota_0            OTA app          00 10 00020000 00190000
I (98) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
I (105) boot:  6 storage          WiFi data        01 02 00340000 00010000
I (113) boot:  7 nvs_key          NVS keys         01 04 00350000 00001000
I (121) boot: End of partition table
I (125) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=3806ch (229484) map
I (216) esp_image: segment 1: paddr=00058094 vaddr=3ffbdb60 size=04e80h ( 20096) load
I (224) esp_image: segment 2: paddr=0005cf1c vaddr=40080000 size=030fch ( 12540) load
I (230) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=b8af4h (756468) map
I (504) esp_image: segment 4: paddr=00118b1c vaddr=400830fc size=13850h ( 79952) load
I (548) boot: Loaded app from partition at offset 0x20000
I (549) boot: Disabling RNG early entropy source...
I (560) cpu_start: Pro cpu up.
I (561) cpu_start: Starting app cpu, entry point is 0x40081384
0x40081384: call_start_cpu1 at /home/memo/esp/esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (577) cpu_start: Pro cpu start user code
I (577) cpu_start: cpu freq: 160000000 Hz
I (577) cpu_start: Application information:
I (581) cpu_start: Project name:     FeaturedFreeRTOSIoTIntegration
I (588) cpu_start: App version:      v202212.00-20-g6bec3de-dirty
I (595) cpu_start: Compile time:     May 29 2023 10:39:29
I (601) cpu_start: ELF file SHA256:  606c94b871df3a86...
I (607) cpu_start: ESP-IDF:          v5.0.2-dirty
I (613) cpu_start: Min chip rev:     v0.0
I (617) cpu_start: Max chip rev:     v3.99 
I (622) cpu_start: Chip rev:         v1.0
I (627) heap_init: Initializing. RAM available for dynamic allocation:
I (634) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
I (640) heap_init: At 3FFD9368 len 00006C98 (27 KiB): DRAM
I (646) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (653) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (659) heap_init: At 4009694C len 000096B4 (37 KiB): IRAM
I (667) spi_flash: detected chip: generic
I (670) spi_flash: flash io: dio
I (675) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (686) esp_secure_cert_tlv: Could not find the tlv of type 1
E (696) esp_secure_cert_tlv: Could not find header for TLV type 1
E (696) main: Error in getting device certificate. Error: ESP_FAIL
E (706) main: Error in getting CA certificate. Error: ESP_FAIL
E (716) esp_secure_cert_tlv: Could not find the tlv of type 2
E (716) esp_secure_cert_tlv: Could not find header for TLV type 2
E (726) main: Error in getting private key. Error: ESP_FAIL
E (736) main: Failed to initialize global network context.
```

**Reference**
[Here](https://github.com/FreeRTOS/iot-reference-esp32c3/blob/main/GettingStartedGuide.md#23-provision-the-esp32-c3-with-the-private-key-device-certificate-and-ca-certificate-in-development-mode).

**Screenshot**
[image]

**Browser**
N/A

emaiben, 2023-05-29:
have the same problem somewhere have something?

seorospa, 2023-05-29:
I solved my problem by enabling the following option in the menuconfig.

`Component config > ESP Secure Cert Manager -> Enable support for legacy formats`

rawalexe, 2023-10-01:
WE apologize for the confusion, the documentation has now been updated and reflects appropriate necessary directions. Please, let us know so that we can close this issue, if you are satisfied.

AniruddhaKanhere, 2024-01-15:
I shall be closing this ticket as the documentation has been updated. Please feel free to open this issue or open a new one in case you are not satisfied with the changes.

closed 2024-01-15 by AniruddhaKanhere