FreeRTOS/iot-reference-esp32 #38: Using encrypted flag for the NVS partition called "storage" is probably forbidden

idea--list, 2023-06-04:
**Describe the bug**
After enabling flash and NVS encryption, i wanted to use the NVS partition called "storage" in the partition table to store custom factory settings for each of my devices. Instead of using the default NVS partition called "nvs".
Modified all the relevant API calls in code (from nvs_flash_init() to nvs_flash_init_partition() and nvs_open() to nvs_open_from_partition()) accordingly.

Flashed an NVS image containing predefined values to the address of "storage" nvs partition.
When my ESP32S3 is starting up, it failed by reporting ESP_ERR_NVS_WRONG_ENCRYPTION.
Found [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#c.ESP_ERR_NVS_WRONG_ENCRYPTION) relevant info in the documentation.
So not sure if marking "storage" nvs partition with encrypted flag is right or not.

**System information**
- Hardware board: ESP32S3
- IDE used: VSCode
- Operating System: Windows
- Code version: IDF 5.0.2

**Expected behavior**
I expect both my variables initialized properly and WiFi credentials loaded fine even when flash and NVS encryption is enabled.

**Steps to reproduce bug**
    Example:
      1. enable flash and nvs encrytion in menuconfig
      2. flash your NVS binary image containing predefined values to the address 0x340000
      3. use "storage" partition instead of the default one called "nvs" to retrieve values, change nvs_flash_init() to nvs_flash_init_partition() and nvs_open() to nvs_open_from_partition() in code accordingly
      4. observe variables initializing with proper values after removing "encrypted" flag from "storage" partition inside partitions.csv

Skptak, 2023-08-18:
Hey @idea--list sorry for the long delay without a response. 
I wanted to check if this is something you still believe is a bug or if you have found a root cause as to why this was happening.

idea--list, 2023-08-18:
@Skptak 
This bug pretty much still exists. Which forces me to set the nvs partition as unencrypted.

Skptak, 2023-08-19:
Hey @idea--list thanks for letting me know,
I've talked to some people on the team about this. Unfortunately, as we don't include encryption on our references this is something we'd need to try out ourselves as well. I don't have an ESP32S3 on me yet, but I placed an order for one where once that gets delivered I'll be taking a deeper look into this. Where if you haven't asked the [ESP-Forums](https://www.esp32.com/viewforum.php?f=23) it might be worth posting there in the meantime?

In the meantime, is your project something that you'd feel comfortable sharing by any chance? Where we could then see the exact error you're seeing? Partly asking since you mention some of the particular APIs you're using but didn't mention how you set up the [NVS Key Partition](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_encryption.html#nvs-key-partition) mentioned on ESP's website. Or the usage of [nvs_flash_secure_init_partition()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#_CPPv431nvs_flash_secure_init_partitionPKcP13nvs_sec_cfg_t). 

Thanks again for your understanding and patience, I'll let you know when I get a board and have a chance to try this out

closed 2024-01-05 by idea--list