FreeRTOS/iot-reference-esp32 #39: Unable to build with container IDF

beewoolie, 2023-06-06:
We have followed the instructions to clone this repo and then mount the source directory in the IDF containers for v4.4.3, v4.4.4, v5.0.2, and latest.  None of these IDF containers will build this repo.  What more do we need to do to build this sample?

ActoryOu, 2023-07-11:
Hi @beewoolie,
Sorry for late response. Could you help to paste the compile log here?
I guess there are some submodules are not cloned correctly.
Please try below command in your local repo.
`git submodule update --init --recursive`

Thanks.

Skptak, 2023-08-18:
Closing this as there has been no response in a few months, if you experience any other issues please do not hesitate to re-open this issue or create a new one!

closed 2023-08-18 by Skptak