FreeRTOS/iot-reference-esp32 #41: Correct way to deinit

zafeer-birde, 2023-06-24:
**Is your feature request related to an issue you've encountered? If so, please elaborate.**
As part of the project we need to be able to disconnect and reconnect MQTT .

I am struggling to deinit the demo eg shadow_pub_sub correctly. Reversing the sequence does not really help. I have tried 

1. MQTT_Disconnect
2. MQTTAgent_CancelAll
3. xCoreMqttAgentManagerDeregisterHandler
4. MQTTAgent_Terminate
5. Task deletions but 

in sequence with no luck. When I try to reinitialize , it has references of the first run and crashes everytime.

Is there any sample code available for correct way to deinit as per the full process ?

Skptak, 2023-08-18:
Hey @zafeer-birde, I apologize for the long delay in getting a response.
I wanted to check if you are still facing issues with deinitializing in the demo you linked, or if you figured out how to do this.

Skptak, 2023-08-22:
Hey again, on the chance you, or somebody else, comes back to this in the future 
When running `idf.py menuconfig` By going to Component Config ->  ESP-MQTT Configurations you can enable "Skip publish if disconnected", which might assist in correctly de-initing the demo.

Skptak, 2023-12-14:
Closing issue due to no response, please re-open this if you are still seeing this issue!

closed 2023-12-14 by Skptak