FreeRTOS/iot-reference-esp32 #42: [BUG] Build Issue for ESP32

djalpiza, 2023-07-12:
**Describe the bug**
I am trying to build the example to run in the ESP32-D0WDQ6.

**System information**
- Hardware board: ESP32-D0WDQ6
- IDE used: ESP-IDF v5.1
- Operating System: Windows
- Code version: main on 12/07/2023
- Project/Demo: Final goal is to do mqtt OTA - right now I am having issues with GettingStartedGuide, on Step 3: Build and flash the demo project
- If your project is a custom application, please add the relevant code snippet(s) in the section titled `"Code to reproduce bug"`.

**Expected behavior**
Build Pass

**Screenshots or console output**
Executing action: all (aliases: build)
Running ninja in directory C:\iot-reference-esp32c3\build
Executing "ninja all"...
[1/1] cmd.exe /C "cd /D C:\iot-reference-esp32c3\build\bootloader\esp-idf\esptool_py && C:\Espressif\python.../check_sizes.py --offset 0xb000 bootloader 0x1000 C:/iot-reference-esp32c3/build/bootloader/bootloader.bin 
Bootloader binary size 0x6850 bytes. 0x37b0 bytes (35%) free.
[2/4] Linking CXX executable FeaturedFreeRTOSIoTIntegration.elfFAILED: FeaturedFreeRTOSIoTIntegration.elf
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--Map=C:/iot-reference-esp32c3/build/FeaturedFreeRTOSIoTIntegration.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T memory.ld -T sections.ld @CMakeFiles\FeaturedFreeRTOSIoTIntegration.elf.rsp -o FeaturedFreeRTOSIoTIntegration.elf  && cd ."      
c:/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: FeaturedFreeRTOSIoTIntegration.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
c:/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
c:/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
c:/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 568 bytes
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\iot-reference-esp32c3\build\log\idf_py_stderr_output_24016 and C:\iot-reference-esp32c3\build\log\idf_py_stdout_output_24016

**Steps to reproduce bug**
    Example:
      1. " am using the full cloned repo as available in VSCode trying to build the demo for ESP32
      2. "When run on "idf.py build", I observed the attacht console error.

**Code to reproduce bug**
```cpp
idf.py build`
```

**Additional context**
Include any additional context relevant to the issue here.

paulbartell, 2023-07-13:
@djalpiza It looks like you have run out of RAM while building your project. 

Please refer to the [esp-idf documentation on this topic](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/performance/ram-usage.html).

Skptak, 2023-08-22:
Hey @djalpiza just wanted to see if you were still looking for further assistance with this, or if you manged to figure it out yourself?

AniruddhaKanhere, 2023-09-13:
I am closing this issue as we have not heard back from you @djalpiza. If you feel that your issue is not resolved or if you have more questions, please feel free to reopen this issue or create a new one.

Sincerely,
Aniruddha

closed 2023-09-13 by AniruddhaKanhere