FreeRTOS/iot-reference-esp32 #47: Core MQTT doesn't reconnect to broker after disconnecting

Obaidiaa, 2023-08-19:
I faced a problem with core MQTT it doesn't reconnect after disconnecting I think the problem in this line of code 

https://github.com/FreeRTOS/iot-reference-esp32c3/blob/d25036b33828e4c1af96574d8401fae3cc1c0a2a/main/networking/mqtt/core_mqtt_agent_manager.c#L778

### It should look like this :
` while( xEventGroupWaitBits( xNetworkEventGroup, CORE_MQTT_AGENT_DISCONNECTED_BIT, pdFALSE, pdFALSE, 0 ) == CORE_MQTT_AGENT_DISCONNECTED_BIT ) `

It starts working well after this change.

Skptak, 2023-08-22:
Hey @Obaidiaa 
Thanks for raising this issue, looking at the API for [xEventGroupWaitBits](https://www.freertos.org/xEventGroupWaitBits.html) it should return the value of the event group. So it seems that by swapping it from  `xEventGroupWaitBits() != CORE_MQTT_AGENT_DISCONNECTED_BIT` to `xEventGroupWaitBits() ==  CORE_MQTT_AGENT_DISCONNECTED_BIT` I'm worried that you're actually skipping over the the code that is supposed to cause the disconnect?
Would you mind trying the policy change I mentioned in #46 [here](https://github.com/FreeRTOS/iot-reference-esp32c3/issues/46#issuecomment-1687351129) and seeing if this works for you?

Obaidiaa, 2023-08-22:
Hey @Skptak ,
I don't think it related to policy I did try before with open policy and same issue. 
It stuck in process loop and didn't initiate the reconnect code.

Skptak, 2023-09-12:
Hey @Obaidiaa, it looks like this issue might be caused by a timeout in the TLS connection. Us on the team are having difficulties replicating these issues, but @ActoryOu proposed a potential fix for this in #46
> It seems like 1 second timeout is not enough for the device to finish TLS flow.
> Could you help to set Featured FreeRTOS IoT Integration -> TLS Transport Send / Receive timeout in milliseconds to 10000 by idf.py menuconfig and retest?

If you're still working on this, would you mind trying this out and seeing if this solves your issue?

Obaidiaa, 2023-09-22:
The change I made above fix this issue for me

rawalexe, 2023-10-01:
We are closing this thread as the issue seems to be resolved.

closed 2023-10-01 by rawalexe