FreeRTOS/iot-reference-esp32 #49: [BUG] Build error FAILED: FeaturedFreeRTOSIoTIntegration.elf

Youhana-Hana, 2023-09-18:
**Describe the bug**
Please provide a clear and concise description explaining the bug.
Error building the project with error FAILED: FeaturedFreeRTOSIoTIntegration.elf 

Bootloader binary size 0x6860 bytes. 0x37a0 bytes (35%) free.
[4/6] Linking CXX executable FeaturedFreeRTOSIoTIntegration.elf
FAILED: FeaturedFreeRTOSIoTIntegration.elf 
: && /Users/ymaher/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -mlongcalls -Wno-frame-address -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--Map=/Users/ymaher/projects/iot-reference-esp32c3/build/FeaturedFreeRTOSIoTIntegration.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T memory.ld -T sections.ld CMakeFiles/FeaturedFreeRTOSIoTIntegration.elf.dir/project_elf_src_esp32.c.obj -o FeaturedFreeRTOSIoTIntegration.elf -L/Users/ymaher/esp/esp-idf/components/soc/esp32/ld   -L/Users/ymaher/esp/esp-idf/components/esp_rom/esp32/ld   -L/Users/ymaher/projects/iot-reference-esp32c3/build/esp-idf/esp_system/ld   -L/Users/ymaher/esp/esp-idf/components/esp_phy/lib/esp32   -L/Users/ymaher/esp/esp-idf/components/esp_coex/lib/esp32   -L/Users/ymaher/esp/esp-idf/components/esp_wifi/lib/esp32   -L/Users/ymaher/esp/esp-idf/components/bt/controller/lib_esp32/esp32 esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/bt/libbt.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/fatfs/libfatfs.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/qrcode/libqrcode.a  esp-idf/coreMQTT/libcoreMQTT.a  esp-idf/coreMQTT-Agent/libcoreMQTT-Agent.a  esp-idf/corePKCS11/libcorePKCS11.a  esp-idf/coreJSON/libcoreJSON.a  esp-idf/backoffAlgorithm/libbackoffAlgorithm.a  esp-idf/espressif__esp_secure_cert_mgr/libespressif__esp_secure_cert_mgr.a  esp-idf/espressif__cbor/libespressif__cbor.a  esp-idf/ota-for-aws-iot-embedded-sdk/libota-for-aws-iot-embedded-sdk.a  esp-idf/coreHTTP/libcoreHTTP.a  esp-idf/FreeRTOS-Libraries-Integration-Tests/libFreeRTOS-Libraries-Integration-Tests.a  esp-idf/espressif__led_strip/libespressif__led_strip.a  esp-idf/main/libmain.a  esp-idf/Device-Defender-for-AWS-IoT-embedded-sdk/libDevice-Defender-for-AWS-IoT-embedded-sdk.a  esp-idf/Device-Shadow-for-AWS-IoT-embedded-sdk/libDevice-Shadow-for-AWS-IoT-embedded-sdk.a  esp-idf/Fleet-Provisioning-for-AWS-IoT-embedded-sdk/libFleet-Provisioning-for-AWS-IoT-embedded-sdk.a  esp-idf/Jobs-for-AWS-IoT-embedded-sdk/libJobs-for-AWS-IoT-embedded-sdk.a  esp-idf/app_trace/libapp_trace.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/bt/libbt.a  -lbtdm_app  -u ld_include_hli_vectors_bt  esp-idf/console/libconsole.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/json/libjson.a  esp-idf/qrcode/libqrcode.a  esp-idf/coreMQTT-Agent/libcoreMQTT-Agent.a  esp-idf/backoffAlgorithm/libbackoffAlgorithm.a  esp-idf/espressif__esp_secure_cert_mgr/libespressif__esp_secure_cert_mgr.a  esp-idf/FreeRTOS-Libraries-Integration-Tests/libFreeRTOS-Libraries-Integration-Tests.a  esp-idf/unity/libunity.a  esp-idf/coreMQTT/libcoreMQTT.a  esp-idf/ota-for-aws-iot-embedded-sdk/libota-for-aws-iot-embedded-sdk.a  esp-idf/corePKCS11/libcorePKCS11.a  esp-idf/espressif__cbor/libespressif__cbor.a  esp-idf/coreHTTP/libcoreHTTP.a  esp-idf/espressif__led_strip/libespressif__led_strip.a  esp-idf/coreJSON/libcoreJSON.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/mbedtls/mbedtls/library/libmbedtls.a  esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a  esp-idf/mbedtls/mbedtls/library/libmbedx509.a  /Users/ymaher/esp/esp-idf/components/esp_coex/lib/esp32/libcoexist.a  /Users/ymaher/esp/esp-idf/components/esp_wifi/lib/esp32/libcore.a  /Users/ymaher/esp/esp-idf/components/esp_wifi/lib/esp32/libespnow.a  /Users/ymaher/esp/esp-idf/components/esp_wifi/lib/esp32/libmesh.a  /Users/ymaher/esp/esp-idf/components/esp_wifi/lib/esp32/libnet80211.a  /Users/ymaher/esp/esp-idf/components/esp_wifi/lib/esp32/libpp.a  /Users/ymaher/esp/esp-idf/components/esp_wifi/lib/esp32/libsmartconfig.a  /Users/ymaher/esp/esp-idf/components/esp_wifi/lib/esp32/libwapi.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a [... cut ...]
If appropriate, please paste the console output/error log explaining the issue. If possible, include the call stack.

**Steps to reproduce bug**
    Example:
      1. "I am using project [ ... ], and have configured with [ ... ]"
      2. "When run on [ ... ], I observed that [ ... ]"

**Code to reproduce bug**
The code should be wrapped in the ``cpp`` tag in order to be displayed clearly. For example:

```cpp
printf("Hello World")
```

**Additional context**
Include any additional context relevant to the issue here.

kstribrnAmzn, 2023-09-18:
```
/Users/ymaher/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: FeaturedFreeRTOSIoTIntegration.elf section .dram0.bss' will not fit in region dram0_0_seg'
```

```
/Users/ymaher/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: region `dram0_0_seg' overflowed by 640 bytes
```

Are suggesting that you've run out of RAM when building your application. The [ESP-IDF documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/performance/ram-usage.html) provides suggested ways around this

puneet-arora15, 2023-09-26:
I'm having the same issue as @Youhana-Hana in ESP5.1 and wanted to see if someone can help solve the IRAM issue with the repo code. As I'm getting RAM issues with the repo code without any changes. 

@Skptak I have seen that you offered help to one of the previous PR and wanted to see if you could help here in the same request.

kstribrnAmzn, 2023-09-26:
@puneet-arora15 that's not good to hear. Our demos should build out of the box with no issues. Can you tell me which demo you are building? I'd like to try and duplicate it on my end.

puneet-arora15, 2023-09-26:
Yeah sure @kstribrnAmzn, I'm trying to setup the ota_over_mqtt_demo and sub_pub_unsub_demo together. It works perfectly fine on the v5.0 ESP IDF although the same code doesn't compile and errors on overflow on RAM. I'm new to ESP world, so having a bit of difficulties with resolving the issue. 

I tried following the ESP doc on it, but didn't have much luck with it. Hopefully you can help with it

rawalexe, 2023-10-01:
Hello @puneet-arora15 and @Youhana-Hana 
For esp32 chip you can try reducing the `coreMQTT-Agent network buffer size` that is under idf.py menuconfig ->Featured FreeRTOS IoT Integration -> coreMQTT-Agent Manager Configurations. Some of the task size configuration under Featured FreeRTOS IoT Integration are more than what's needed you may try reducing them where ever needed.

I have only noticed this memory problem with few of the latest esp-idf and if you do not want to change the buffer size you may try to use older version of esp-idf.

Hopefully this helps out

Best Regards.
AR

ran202, 2023-12-30:
I also see the same compilation error.  I also followed the suggestion above to reduce "coreMQTT-Agent network buffer size" but that didnot help (reduced to 1000 from 10000).

My setup-
- Hardware board: ESP32 WROOM Dev board
- IDE used: ESP IDF v5.1.2
- Operating System: Windows
- Code version: v202212.00-29-g19888ff
- Project/Demo: OTA over mqtt and LED on/off

Will highly appreciate your help on this.  
Regards.

gamename, 2024-01-09:
@kstribrnAmzn

I'm getting the same error as well.  Will provide any diagnostics you like from my environment.

Hardware: esp32-wroom-32s (DevKitC v4)
IDE: vscode 1.85.1 
ESP-IDF: v5.3-dev-796-g30870c819f
OS: MacOS 14.2.1 (23C71)

Log:
[flash-and-monitor.log](https://github.com/FreeRTOS/iot-reference-esp32c3/files/13878533/flash-and-monitor.log)

kstribrnAmzn, 2024-01-23:
Both of you are having `dram0_0_seg` overflowing. From what I'm seeing online this looks common when a static variable is too large. I'm going to try to build the OTA over MQTT demo and will post my map file details.

Skptak, 2024-01-26:
Hey @gamename, I've started working on the issues you're hitting in #68.

I don't have a ESP32-C2 to verify they work on hardware, but am going to be trying the various demos in this repo on the C3 and S3 when I can.

If you'd be able to try checking out my branch and running them yourself that'd be extremely helpful though!

Thanks!

gamename, 2024-01-26:
@Skptak 
Thanks for looking at this.  No problem.  Will be happy to try your code.  Let me know when its ready for testing and I'll pull a copy.

gamename, 2024-01-26:
@Skptak 

When trying to compile without any modification, I get the following error (see attached)

[compile.log](https://github.com/FreeRTOS/iot-reference-esp32c3/files/14068976/compile.log)

Skptak, 2024-01-26:
Hey @gamename, thanks for trying out my branch! I really appreciate you taking the time to do that! That error was actually the exact reason I made [this commit](https://github.com/FreeRTOS/iot-reference-esp32c3/pull/68/commits/5bfd1f4f50c83750aa4da209e3e2b0d823e20b54) to remove the v4.4.3 build from the CI-CD

From what I can tell this file doesn't exist for the ESP32 default device target. A link to a GitHub action log showing this error is [here](https://github.com/FreeRTOS/iot-reference-esp32c3/actions/runs/7671736148/job/20910760048#step:4:2772)

I updated the [GettingStartedGuide.md](https://github.com/FreeRTOS/iot-reference-esp32c3/pull/68/commits/89abab0b90a23372811f2c9831330979615b994d#diff-7eb4ec8845a00e67a78ed3ef84df2e03815d85424965865b0bb1c95d9b23bb2eR54) to mark that the 4.4.3 version of esp-idf is out of support now, and marked that this repository now only will support v5+ as well.

Would you be willing to try updating your `idf.py` to version 5 or above and trying again?

I was just able to validate this demo on my esp32-c3 using this branch. Where was going to start looking into #45, #44 and the related [coreMQTT-Agent issue #104](https://github.com/FreeRTOS/coreMQTT-Agent/issues/104)

gamename, 2024-01-26:
> Would you be willing to try updating your `idf.py` to version 5 or above and trying again?

Sure. Np. Will do that and get back to you.

gamename, 2024-01-27:
@Skptak 

Got another failure. Here are the details
version:
```
idf.py --version
ESP-IDF v5.1.2-dirty
```

[build.log](https://github.com/FreeRTOS/iot-reference-esp32c3/files/14070522/build.log)

gamename, 2024-01-27:
@Skptak 

Actually, I'm already on ESP-IDF version `v5.3-dev-796-g30870c819f`

Skptak, 2024-01-29:
Hey @gamename,

> Actually, I'm already on ESP-IDF version v5.3-dev-796-g30870c819f

I believe v5.3 is still in the development phase and as such does not have an official tag to use for it. As such I don't know if we can make any promises to support that version. I've added `v5.2`, as the latest officially released version of ESP-IDF to the CI-CD Builds in this [commit](https://github.com/FreeRTOS/iot-reference-esp32c3/pull/68/commits/b18a81246ce2a13564a3a3ffb5082a7b86b88b1b)

> PS. My hardware is esp32-wroom-32d. Still want me to try it?

I also know that this repo maintains that it only officially supports the ESP32-C3, ESP32-S3, and ESP32-S2. I'm not sure if that chip is close enough to the ESP32 default target that is supplied in the [espressif/esp-idf-ci-action](https://github.com/espressif/esp-idf-ci-action)

gamename, 2024-01-30:
Hi @Skptak 

Understood.  I'll give it a try anyway and get back to you with any results.  No worries if it isn't really supported.  Will let you know what happens just for the sake of interest. :)

Skptak, 2024-01-30:
Thanks @gamename, we've just merged #68, so will wait to hear back from you how it goes

paulbartell, 2024-02-01:
I believe this issue has been addressed. Feel free to reopen this issue if build failures persist.

gamename, 2024-02-09:
@Skptak 

I got an esp32-s3-wroom-1 board and tried it again. FYI

```
Users/tennis/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: FeaturedFreeRTOSIoTIntegration.elf section `.iram0.text' will not fit in region `iram0_0_seg'
/Users/tennis/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: FeaturedFreeRTOSIoTIntegration.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/Users/tennis/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: DRAM segment data does not fit.
/Users/tennis/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: IRAM0 segment data does not fit.
/Users/tennis/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: DRAM segment data does not fit.
/Users/tennis/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: region `iram0_0_seg' overflowed by 760 bytes
/Users/tennis/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: region `dram0_0_seg' overflowed by 2384 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
HINT: The applications static IRAM usage is larger than the available IRAM size.
For more information on how to reduze IRAM usage run 'idf.py docs -sp api-guides/performance/ram-usage.html#optimizing-iram-usage' 
ninja failed with exit code 1, output of the command is in the /Users/tennis/src/iot-reference-esp32c3/build/log/idf_py_stderr_output_29238 and /Users/tennis/src/iot-reference-esp32c3/build/log/idf_py_stdout_output_29238
╭─tennis@tennismbp2019.local ~/src/iot-reference-esp32c3  ‹main*› 
╰─➤  idf --version                                                                                                                                                                                         2 ↵
ESP-IDF v5.3-dev-1353-gb3f7e2c8a4
```

kstribrnAmzn, 2024-02-13:
I was able to reproduce your issue with an ESP32 C3 board once in an afternoon of trying. It appears my ESP-IDF setup is really in a bad state OR the plugin doesn't like me. I don't think this issue is resolved and so I'll reopen it.

I will have to do a bit further of a dive. I suspect part of this will be fixing my ESP-IDF installation.

kstribrnAmzn, 2024-02-15:
My ESP-IDF installation was certainly in a weird state. I was able to flash both an ESP32-C3 and ESP32-S3 today without either showing a memory sizing issue. I used ESP-IDF 5.1.2.

Would you be able to try with an officially release ESP-IDF version?

gamename, 2024-02-15:
@kstribrnAmzn 
OK.  I'll give it a try and get back to you.

gamename, 2024-02-17:
@kstribrnAmzn 
It worked! I tried it with v5.2 and it compiled and loaded on my esp32-s3-wroom-1 (ESP32-S3-DevKitC-1-N8R2)

gamename, 2024-02-18:
@kstribrnAmzn 

Update.

I tried an IOT OTA update and got the following error:
```I (108924) AWS_OTA: Received final block of the update.
E (109024) AWS_OTA: Signature verification failed.
I (109704) ota_over_mqtt_demo:  Received: 286   Queued: 286   Processed: 285   Dropped: 0
I (110724) ota_over_mqtt_demo:  Received: 286   Queued: 286   Processed: 285   Dropped: 0
I (111814) ota_over_mqtt_demo:  Received: 286   Queued: 286   Processed: 285   Dropped: 0
I (112914) ota_over_mqtt_demo:  Received: 286   Queued: 286   Processed: 285   Dropped: 0
I (113994) ota_over_mqtt_demo:  Received: 286   Queued: 286   Processed: 285   Dropped: 0
I (115224) ota_over_mqtt_demo:  Received: 286   Queued: 286   Processed: 285   Dropped: 0
I (116274) ota_over_mqtt_demo:  Received: 286   Queued: 286   Processed: 285   Dropped: 0
I (117274) ota_over_mqtt_demo:  Received: 286   Queued: 286   Processed: 285   Dropped: 0
E (117614) AWS_OTA: Failed to close the OTA file: Error=(OtaPalSignatureCheckFailed:0xe3000000)
E (117614) AWS_OTA: Failed to ingest data block, rejecting image: ingestDataBlock returned error: OtaErr_t=-2
I (117624) AWS_OTA: otaPal_SetPlatformImageState, 3
W (117624) AWS_OTA: Set image as invalid!
```

Have you tried an OTA update?

ActoryOu, 2024-03-25:
closing since it's confirmed fixed by new ESP-IDF

closed 2024-03-25 by ActoryOu