FreeRTOS/iot-reference-esp32 #56: [BUG] Cannot open menuconfig

RajatVardam, 2023-10-19:
**Describe the bug**
I simply git cloned this project as described in the readme. opened in VS-Code and straight away went for menuconfig. did not touch a thing after cloning. In VS-Code, we get the menuconfig button on the bottom left. It shows errors.

**System information**
- Hardware board: [ esp32 ]
- IDE used: [ VS-Code with esp-idf extension v5.0.4 ]
- Operating System: [ Windows 10 home latest ]
- Code version: (v202212.00-28-g44d6fe5)
- Project/Demo: [ iot-reference-esp32c3 ]
- If your project is a custom application, please add the relevant code snippet(s) in the section titled `"Code to reproduce bug"`.

**Expected behavior**
menuconfig should open like normal. I have run ota over mqtt code before this from espressif/eso-aws-iot repo. menuconfig does open there like normal. not here.

**Screenshots or console output**
[SDK Configuration Editor]
---------------------------ERROR--------------------------

CMake Error at C:/Users/vfami/esp/esp-idf/tools/cmake/component.cmake:224 (message):
  CMake Warning (dev) at build_properties.temp.cmake:8:

    Syntax Warning in cmake code at column 51

  

    Argument not separated from preceding token by whitespace.

  Call Stack (most recent call first):

    C:/Users/vfami/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:3 (include)

  This warning is for project developers.  Use -Wno-dev to suppress it.

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
---------------------------ERROR--------------------------

  

  CMake Error at
  C:/Users/vfami/.espressif/tools/cmake/3.24.0/share/cmake-3.24/Modules/ExternalProject.cmake:1253
  (define_property):

    define_property command is not scriptable

  Call Stack (most recent call first):

    C:/Users/vfami/esp/esp-idf/tools/cmake/kconfig.cmake:1 (include)
    C:/Users/vfami/esp/esp-idf/tools/cmake/idf.cmake:43 (include)
    C:/Users/vfami/esp/esp-idf/tools/cmake/project.cmake:12 (include)
    C:/Users/vfami/esp/esp-idf/components/iot-reference-esp32c3/CMakeLists.txt:6 (include)
    C:/Users/vfami/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:106 (include)
    C:/Users/vfami/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:124 (__component_get_requirements)

  

  

Call Stack (most recent call first):
  C:/Users/vfami/esp/esp-idf/tools/cmake/build.cmake:553 (__component_get_requirements)
  C:/Users/vfami/esp/esp-idf/tools/cmake/project.cmake:530 (idf_build_process)
  CMakeLists.txt:8 (project)

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
-- Configuring incomplete, errors occurred!
See also "C:/Users/vfami/esp/esp-idf/components/iot-reference-esp32c3/build/CMakeFiles/CMakeOutput.log".

[SDK Configuration Editor]
---------------------------ERROR--------------------------

cmake failed with exit code 1, output of the command is in the C:\Users\vfami\esp\esp-idf\components\iot-reference-esp32c3\build\log\idf_py_stderr_output_4300 and C:\Users\vfami\esp\esp-idf\components\iot-reference-esp32c3\build\log\idf_py_stdout_output_4300

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
---------------------------ERROR--------------------------

SDK Configuration editor confserver process exited with code: 2
-----------------------END OF ERROR-----------------------

**Steps to reproduce bug**
    Example:
      1. open the above mentioned project in VS-Code and straight go for menuconfig.

**Code to reproduce bug**
nothing was changed in the code. No file was modified.

ActoryOu, 2023-10-20:
Hi @RajatVardam,
Thank you for bringing this to our attention!
We will be looking further into this issue as soon as we can, thank you for your patience!

ActoryOu, 2023-10-23:
Hi @RajatVardam,
It seems like a bug in esp-idf, refer to [this discussion](https://stackoverflow.com/questions/72680055/cmake-error-define-property-command-is-not-scriptable).
Could you try on another esp-idf version? ESP-IDF v5.1.1 works fine in my environment.

Thanks.

RajatVardam, 2023-10-23:
The error still persists. I changed it according to the discussion tagged here and also upgraded to esp-idf 5.1.1 main branch. This is the output:

CMake Error at C:/Users/vfami/esp/esp-idf/tools/cmake/component.cmake:224 (message):
  CMake Warning (dev) at build_properties.temp.cmake:8:

    Syntax Warning in cmake code at column 47

  

    Argument not separated from preceding token by whitespace.

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
---------------------------ERROR--------------------------

  Call Stack (most recent call first):

    C:/Users/vfami/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:3 (include)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  

  CMake Error at
  C:/Users/vfami/.espressif/tools/cmake/3.24.0/share/cmake-3.24/Modules/ExternalProject.cmake:1253
  (define_property):

    define_property command is not scriptable

  Call Stack (most recent call first):

    C:/Users/vfami/esp/esp-idf/tools/cmake/kconfig.cmake:1 (include)
    C:/Users/vfami/esp/esp-idf/tools/cmake/idf.cmake:43 (include)
    C:/Users/vfami/esp/esp-idf/tools/cmake/project.cmake:29 (include)
    C:/Users/vfami/esp/esp-idf/components/iot-reference-esp32c3/CMakeLists.txt:6 (include)
    C:/Users/vfami/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:106 (include)
    C:/Users/vfami/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:124 (__component_get_requirements)

  

  

Call Stack (most recent call first):
  C:/Users/vfami/esp/esp-idf/tools/cmake/build.cmake:574 (__component_get_requirements)
  C:/Users/vfami/esp/esp-idf/tools/cmake/project.cmake:547 (idf_build_process)
  CMakeLists.txt:8 (project)

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
-- Configuring incomplete, errors occurred!
See also "C:/Users/vfami/esp/esp-idf/components/iot-reference-esp32c3/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/vfami/esp/esp-idf/components/iot-reference-esp32c3/build/CMakeFiles/CMakeError.log".

[SDK Configuration Editor]
---------------------------ERROR--------------------------

cmake failed with exit code 1, output of the command is in the C:\Users\vfami\esp\esp-idf\components\iot-reference-esp32c3\build\log\idf_py_stderr_output_13944 and C:\Users\vfami\esp\esp-idf\components\iot-reference-esp32c3\build\log\idf_py_stdout_output_13944

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
---------------------------ERROR--------------------------

SDK Configuration editor confserver process exited with code: 2
-----------------------END OF ERROR-----------------------

RajatVardam, 2023-10-23:
Tried it. no change. upgraded to esp-idf v5.1.1 main and also went through the discussion tagged. error is exactly the same as it was before.

ActoryOu, 2023-10-24:
Could you help to provide your `component_properties.temp.cmake` under build folder?
Note that it's deleted automatically.