FreeRTOS/iot-reference-esp32 #65: [BUG] Project fails to link when compiled as C++

txf-, 2024-01-15:
**Describe the bug**
app_wifi, core_mqtt_agent_manager and subscription_manager are not linking in a C++ project.

**Expected behavior**
 The header file is missing the indication to treat these as C linkage.

```
#ifdef __cplusplus
extern "C" {
#endif

/*FUNCTION PROTOTYPES*/

#ifdef __cplusplus
}
#endif

```
I don't know if this is done on purpose or it is a bug.

ericbj29, 2024-01-15:
Hi @txf-, thanks for bringing this to our attention. This is a bug. If you are willing, we would appreciate it if you could open a PR fixing the issue. If not, I'm happy to do so as well. Thanks again!

ericbj29, 2024-01-15:
If you do decide to raise a PR, please double check the other header files. It seems [`core_mqtt_agent_manager_events.h`](https://github.com/FreeRTOS/iot-reference-esp32c3/blob/main/main/networking/mqtt/core_mqtt_agent_manager_events.h) needs to be updated as well.

closed 2024-01-18 by ericbj29