FreeRTOS/iot-reference-esp32 #67: Long filenames a problem

monkeytronics, 2024-01-17:
Hi there. 

I am hoping to use this example code to upgrade my project from v3.1. This issue is a question and a feature request.

Question
----------
Do you have a recommended method to use this code in user projects. 

It occurred to me that adding this repo as a submodule would be a good idea, including referencing the esp-aws-iot repo as a recursive submodule. That way, it would be efficient to reuse lots of the code in your main/mqtt folder and easily update it as you develop this repo. My application could be roughly based on main and demo tasks. And it would ensure that the submodules would be in matching state. Is this how you would recommend? Is there another alternative to doing a horrible cut and paste of the code?

Feature Request
------------------
Due to the very long file paths (Fleet-Provisioning-for-AWS-IoT-embedded-sdk/Fleet-Provisioning-for-AWS-IoT-embedded-sdk), it isn't possible to successfully run submodule update --init --recursive. Would you consider reducing the file names to something more concise like "fleet-prov"? I think the rest is implied. 

Thanks.

jasonpcarroll, 2024-01-23:
Hi @monkeytronics,

I would advise against submoduling this repo for your project as it is intended to be used as a reference project. I would recommend either forking this repository and updating it for your own project or use this project as a template to get started making your own.

Best,

Jason Carroll

monkeytronics, 2024-01-24:
Thanks @jasonpcarroll, Agree. That does sound best. 

I think the API for this library needs a lot of work. It's kind of crazy that you need to write thousands of lines of code (full of callback spaghetti) just to get the hello world examples to run! Say what you like about Arduino - but they know how to abstract a library so that almost anyone can easily use it.

closed 2024-01-31 by paulbartell