FreeRTOS/iot-reference-esp32 #80: [question] I want to perform MQTT initialization connection after WIFI connection

ookamipjq, 2024-04-27:
I am using the code of this warehouse for mass production planning. According to the code execution process in the main function, after the esp32 is powered on, it first obtains the certificate stored inside the module and passes it into the network context. It performs the MQTT kernel before WIFI initialization. Initialization. Now I want to connect to WIFI after powering on for the first time, and get the certificate content from HTTP to connect to AWS. How can I put MQTT initialization on the WIFI connection and do it after I get the certificate from HTTP.

cookpate, 2024-05-06:
Without changing the demo APIs, one could check if the certificate is available in flash before continuing with the demo. If it is not, then initialize WiFi without starting the CoreMQTT Agent in order to perform the HTTP download, store the received certificate in flash, then restart the device.

If restarting a device would take too long, a device could stop the WiFi after retrieving a certificate in order to start coreMQTT-Agent correctly. A clean `app_wifi_stop()` function could stop the WiFi connection (but not de-init), then clear the bits in `wifi_event_group` to re-arm the WiFi connected event.

Take a look instead at how WiFi is initialized in the demo [main/networking/wifi/app_wifi.c](https://github.com/FreeRTOS/iot-reference-esp32c3/blob/76f7059/main/networking/wifi/app_wifi.c#L254-L411)

Also, refer to Espressif documentation on the ESP WIFI API:
https://docs.espressif.com/projects/esp-idf/en/v5.1.3/esp32c3/api-reference/network/esp_wifi.html

ookamipjq, 2024-05-20:
Thanks for your answer, this is what I'm using and I'm looking for a better solution

closed 2024-05-20 by ookamipjq