FreeRTOS/iot-reference-esp32 #89: [BUG] <AWS cert write and read is not the same>

JasonYan324, 2024-07-18:
**Describe the bug**
Please provide a clear and concise description explaining the bug.

**System information**
- Hardware board: [ESP32-C3-DevKit-RUST-1]
- IDE used: [VS code]
- Operating System: [ Windows ]
- Code version: (v202212.00-64-g06f5ce1)
- Project/Demo: [ mqtt_demo_mutual_auth]
- If your project is a custom application, please add the relevant code snippet(s) in the section titled `"Code to reproduce bug"`.

**Expected behavior**
A clear description of the expected behavior.

**Screenshots or console output**
1. Run the following command to create and flash the certificate partition.
python managed_components/espressif__esp_secure_cert_mgr/tools/configure_esp_secure_cert.py -p "COM10" --configure_ds --keep_ds_data_on_host --ca-cert "D:/99ESP/T3/iot-reference-esp32c3/main/certs/aws-root-ca.pem" --device-cert "D:/99ESP/T3/iot-reference-esp32c3/main/certs/certificate.pem.crt" --private-key "D:/99ESP/T3/iot-reference-esp32c3/main/certs/private.pem.key" --target_chip "esp32c3" --secure_cert_type cust_flash --priv_key_algo "RSA" "2048"
2. ouput:
D:\99ESP\T3\iot-reference-esp32c3>python managed_components/espressif__esp_secure_cert_mgr/tools/configure_esp_secure_cert.py -p "COM10" --configure_ds --keep_ds_data_on_host --ca-cert "D:/99ESP/T3/iot-reference-esp32c3/main/certs/aws-root-ca.pem" --device-cert "D:/99ESP/T3/iot-reference-esp32c3/main/certs/certificate.pem.crt" --private-key "D:/99ESP/T3/iot-reference-esp32c3/main/certs/private.pem.key" --target_chip "esp32c3" --secure_cert_type cust_flash --priv_key_algo "RSA" "2048"
Using the same hmac key burned in efuse BLOCK_KEY1
Flashing the esp_secure_cert partition at 0xD000 offset
Note: You can skip this step by providing --skip_flash argument
esptool.py v4.7.0
Serial port COM10
Connecting...
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: 40:4c:ca:89:c1:84
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x0000d000 to 0x00012fff...
Compressed 24576 bytes to 3339...
Writing at 0x0000d000... (100 %)
Wrote 24576 bytes (3339 compressed) at 0x0000d000 in 0.3 seconds (effective 660.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
3. then build and flash the demo project by running idf.py -p "COM10" flash monitor
4. then monitoring the demo, and error output:
E (392) esp_secure_cert: Metadata magic word does not match
E (392) esp_secure_cert: Error in reading the metadata
E (402) main: Error in getting device certificate. Error: ESP_FAIL
E (412) main: Error in getting CA certificate. Error: ESP_FAIL
6. then in the function "esp_secure_cert_read_metadata" block verification just like this:
[image]
     //if (metadata->magic_word != ESP_SECURE_CERT_METADATA_MAGIC_WORD) {
    //    ESP_LOGE(TAG, "Metadata magic word does not match");
    //    return ESP_FAIL;
    //}
8. and the running out change to this(Private Key's Length is wrong, and the value can not be read ):
-----END CERTIFICATE-----
W (622) mmap: paddr block is mapped already, vaddr_start: 0x3c120000, size: 0x10000
W (632) mmap: paddr block is mapped already, vaddr_start: 0x3c120000, size: 0x10000
I (642) main:
Private Key:
Length: 1200
��=*�����Lc_m�t�� �*m���O��5�p3պ0ؤ���1\��ζI#��״�7�n�h�2M��%�Z��Zt��>]K���2S���s$ս�N[�ŨM7�[�eG�7�(��>u ���1{g-4b�Ym��7�:���]^�65?=
��q!���>_-��jx1(��$�p<�+�y�6c��Cآ�3���w�\
,4�D�J�       _�
I (672) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (672) main:
CS Cert:
Length: 0

I (672) temp_sub_pub_and_led_control_demo: Sending subscribe request to agent for topic filter: /filter/TempSubPubLED with id 1
I (682) ota_over_mqtt_demo: OTA over MQTT demo, Application version 0.0.0

....

I (4402) main_task: Returned from app_main()
I (4412) wifi:<ba-add>idx:1 (ifx:0, 3e:06:a7:1b:0a:35), tid:0, ssn:279, winSize:64
E (4702) esp-tls-mbedtls: mbedtls_pk_parse_keyfile returned -0x3D00
E (4702) esp-tls-mbedtls: Failed to set client pki context
E (4702) esp-tls-mbedtls: Failed to set client configurations, returned [0x8019] (ESP_ERR_MBEDTLS_PK_PARSE_KEY_FAILED)
E (4712) esp-tls: create_ssl_handle failed
E (4712) esp-tls: Failed to open new connection
I (4742) ota_over_mqtt_demo:  Received: 0   Queued: 0   Processed: 0   Dropped: 0
I (5032) core_mqtt_agent_manager: Retry attempt 1.
E (5362) esp-tls-mbedtls: mbedtls_pk_parse_keyfile returned -0x3D00
E (5362) esp-tls-mbedtls: Failed to set client pki context
E (5372) esp-tls-mbedtls: Failed to set client configurations, returned [0x8019] (ESP_ERR_MBEDTLS_PK_PARSE_KEY_FAILED)
E (5382) esp-tls: create_ssl_handle failed
E (5382) esp-tls: Failed to open new connection

**Steps to reproduce bug**
    Example:
      1. "I am using project [ iot-reference-esp32c3 ], and have configured with [ target=esp32c3, port=COM10, device Endpoint=...-ats.iot.ap-northeast-1.amazonaws.com, thing=AwsMqttTest ]"
      10. "When run on [ ESP32-C3-DevKit-RUST-1 ], I observed that [ E (392) esp_secure_cert: Metadata magic word does not match
E (392) esp_secure_cert: Error in reading the metadata
E (402) main: Error in getting device certificate. Error: ESP_FAIL
E (412) main: Error in getting CA certificate. Error: ESP_FAIL ]"

**Code to reproduce bug**
The code should be wrapped in the ``cpp`` tag in order to be displayed clearly. For example:

```cpp
printf("Hello World")
```

**Additional context**
Include any additional context relevant to the issue here.

JasonYan324, 2024-07-18:
[image]
[image]

moninom1, 2024-07-19:
HI @JasonYan324 , Thank you for raising the bug.
We will look into the problem and possible solutions internally, meanwhile if you have a quick solution for the issue, You can also contribute by raising a PR.

archigup, 2024-10-24:
Hey since this seems to have been handled in https://github.com/espressif/esp-idf/issues/14300, I'll close this; feel free to reopen if you still have an issue.

closed 2024-10-24 by archigup