FreeRTOS/iot-reference-esp32 #95: Download failed

JasonYan324, 2024-07-26:
using git cmd：git clone https://github.com/FreeRTOS/iot-reference-esp32c3.git --recurse-subm

result：

Receiving objects: 100% (570/570), 168.63 KiB | 728.00 KiB/s, done.
Resolving deltas: 100% (328/328), done.
Submodule path 'components/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/dependency/3rdparty/pkcs11': checked out '478bfc5077901865993e3a193777c1a9f6516cdf'
fatal: Failed to recurse into submodule path 'components/esp-aws-iot'

and can not idf.py menuconfig

CMake Error at C:/JasonSoftware/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/component.cmake:224 (message):
  CMake Warning (dev) at build_properties.temp.cmake:8:

    Syntax Warning in cmake code at column 106

    Argument not separated from preceding token by whitespace.

  Call Stack (most recent call first):

    C:/JasonSoftware/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/scripts/component_get_requirements.cmake:3 (include)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  CMake Error at
  D:/99ESP/T4/iot-reference-esp32c3/components/FreeRTOS-Libraries-Integration-Tests/CMakeLists.txt:11
  (include):

    include could not find requested file:

      D:/99ESP/T4/iot-reference-esp32c3/components/FreeRTOS-Libraries-Integration-Tests/FreeRTOS-Libraries-Integration-Tests/src/ota_pal_test.cmake

  Call Stack (most recent call first):

    C:/JasonSoftware/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/scripts/component_get_requirements.cmake:107 (include)
    C:/JasonSoftware/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/scripts/component_get_requirements.cmake:127 (__component_get_requirements)

Call Stack (most recent call first):
  C:/JasonSoftware/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/build.cmake:578 (__component_get_requirements)
  C:/JasonSoftware/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
See also "D:/99ESP/T4/iot-reference-esp32c3/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

NightSkySK, 2024-07-26:
Try `git submodule update --init --recursive`

ActoryOu, 2024-07-27:
Hi @JasonYan324,
Looks like some submodules are not cloned to your local branch successfully.
Could you try @NightSkySK comment and feedback again with any clone error msg?

Thank you.

ActoryOu, 2024-07-28:
It's possible that the file path is too long to clone.
Could you help move the root folder to somewhere shorter? E.g. C:\esp32c3\...

Thank you.

JasonYan324, 2024-07-28:
> Try `git submodule update --init --recursive`

it does not works

ActoryOu, 2024-07-29:
Hi @JasonYan324,
I'm closing this issue based on your information in #96.
Thank you.

closed 2024-07-29 by ActoryOu