FreeRTOS/iot-reference-esp32 #96: Cannot find ota_pal_test.cmake

abarke, 2024-07-26:
**Describe the bug**
When opening the config in ESP-IDF, cmake fails as FreeRTOS-Libraries-Integration-Tests/src/ota_pal_test.cmake is not found although referenced here: https://github.com/FreeRTOS/iot-reference-esp32c3/blob/main/components/FreeRTOS-Libraries-Integration-Tests/CMakeLists.txt#L11

**System information**
- Hardware board: ESP32-C3
- IDE used: VSCode
- Operating System: Windows
- Code version: main
- Project/Demo: 

**Expected behavior**
No error... although when I removed this line it works: https://github.com/FreeRTOS/iot-reference-esp32c3/blob/main/components/FreeRTOS-Libraries-Integration-Tests/CMakeLists.txt#L11

**Screenshots or console output**
If appropriate, please paste the console output/error log explaining the issue. If possible, include the call stack.

**Steps to reproduce bug**
    Example:
      1. git clone --recursive https://github.com/FreeRTOS/iot-reference-esp32c3.git
      2. Click on `ESP-IDF: SDK Configuration Editor (menuconfig)`

JasonYan324, 2024-07-26:
the same issue.
using git cmd to download the soure code just like this: git clone https://github.com/FreeRTOS/iot-reference-esp32c3.git --recurse-submodules
and the result is(can not found error):
Cloning into 'D:/T4/iot-reference-esp32c3/components/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/dependency/3rdparty/pkcs11'...
remote: Enumerating objects: 570, done.
remote: Counting objects: 100% (127/127), done.
remote: Compressing objects: 100% (88/88), done.
remote: Total 570 (delta 63), reused 94 (delta 34), pack-reused 443
Receiving objects: 100% (570/570), 168.63 KiB | 235.00 KiB/s, done.
Resolving deltas: 100% (328/328), done.
Submodule path 'components/esp-aws-iot/libraries/corePKCS11/corePKCS11/source/dependency/3rdparty/pkcs11': checked out '478bfc5077901865993e3a193777c1a9f6516cdf'

using VS Code bulding(IDF version 5.2.2), result:
 *  Executing task: C:\JasonSoftware\Espressif\tools\cmake\3.24.0\bin\cmake.exe -G=Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B=d:\99ESP\T4\iot-reference-esp32c3\build -S=d:\99ESP\T4\iot-reference-esp32c3 

-- Found Git: C:/JasonSoftware/Espressif/tools/idf-git/2.44.0/cmd/git.exe (found version "2.44.0.windows.1") 
-- Component directory D:/99ESP/T4/iot-reference-esp32c3/components/esp-aws-iot/libraries/common does not contain a CMakeLists.txt file. No component will be added
-- Component directory D:/99ESP/T4/iot-reference-esp32c3/components/esp-aws-iot does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/JasonSoftware/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/JasonSoftware/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/JasonSoftware/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c3
Processing 4 dependencies:
[1/4] espressif/cbor (0.6.0~1)
[2/4] espressif/esp_secure_cert_mgr (2.4.1) from file:///C:/JasonSoftware/Espressif/registry
[3/4] espressif/led_strip (2.5.4) from file:///C:/JasonSoftware/Espressif/registry
[4/4] idf (5.2.2)
CMake Error at C:/JasonSoftware/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/component.cmake:224 (message):
  CMake Warning (dev) at build_properties.temp.cmake:8:

    Syntax Warning in cmake code at column 106

    Argument not separated from preceding token by whitespace.

  Call Stack (most recent call first):

    C:/JasonSoftware/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/scripts/component_get_requirements.cmake:3 (include)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  CMake Error at
  D:/99ESP/T4/iot-reference-esp32c3/components/FreeRTOS-Libraries-Integration-Tests/CMakeLists.txt:11
  (include):

    include could not find requested file:

      D:/99ESP/T4/iot-reference-esp32c3/components/FreeRTOS-Libraries-Integration-Tests/FreeRTOS-Libraries-Integration-Tests/src/ota_pal_test.cmake

  Call Stack (most recent call first):

    C:/JasonSoftware/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/scripts/component_get_requirements.cmake:107 (include)
    C:/JasonSoftware/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/scripts/component_get_requirements.cmake:127 (__component_get_requirements)

Call Stack (most recent call first):
  C:/JasonSoftware/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/build.cmake:578 (__component_get_requirements)
  C:/JasonSoftware/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
[... 45 lines cut ...]
    C:/JasonSoftware/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/scripts/component_get_requirements.cmake:3 (include)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  CMake Error at
  D:/99ESP/T4/iot-reference-esp32c3/components/FreeRTOS-Libraries-Integration-Tests/CMakeLists.txt:11
  (include):

    include could not find requested file:

      D:/99ESP/T4/iot-reference-esp32c3/components/FreeRTOS-Libraries-Integration-Tests/FreeRTOS-Libraries-Integration-Tests/src/ota_pal_test.cmake

  Call Stack (most recent call first):

    C:/JasonSoftware/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/scripts/component_get_requirements.cmake:107 (include)
    C:/JasonSoftware/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/scripts/component_get_requirements.cmake:127 (__component_get_requirements)

Call Stack (most recent call first):
  C:/JasonSoftware/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/build.cmake:578 (__component_get_requirements)
  C:/JasonSoftware/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
See also "D:/99ESP/T4/iot-reference-esp32c3/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

wreyford, 2024-07-26:
In the project folder, go to Components -> FreeRTOS-Libraries-Integration-Tests, and edit the CMakeLists.txt, and comment out the following line
# This gives OTA_PAL_TEST_SOURCES, and OTA_PAL_TEST_INCLUDE_DIRS
#include(${CMAKE_CURRENT_LIST_DIR}/FreeRTOS-Libraries-Integration-Tests/src/ota_pal_test.cmake)

After this MenuConfig worked. It appears ota_pal_test.cmake was dropped 1/12 ago.

JasonYan324, 2024-07-27:
> In the project folder, go to Components -> FreeRTOS-Libraries-Integration-Tests, and edit the CMakeLists.txt, and comment out the following line `# This gives OTA_PAL_TEST_SOURCES, and [...]

after this MenuConfig worked. But can not buid. error as:

D:/99ESP/T4/iot-reference-esp32c3/main/demo_tasks/ota_over_mqtt_demo/ota_over_mqtt_demo.c:232:36: error: 'otaconfigMAX_NUM_OTA_DATA_BUFFERS' undeclared here (not in a function)
  232 | static OtaDataEvent_t dataBuffers[ otaconfigMAX_NUM_OTA_DATA_BUFFERS ] = { 0 };

JasonYan324, 2024-07-27:
can I ask when I make a define for otaconfigMAX_NUM_OTA_DATA_BUFFERS, then build, an error like this:
[1075/1077] Linking CXX executable FeaturedFreeRTOSIoTIntegration.elf
FAILED: FeaturedFreeRTOSIoTIntegration.elf
cmd.exe /C "cd . && C:\JasonSoftware\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imc_zicsr_zifencei -nostartfiles -march=rv32imc_zicsr_zifencei --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C3=0 -Wl,--Map=D:/99ESP/T4/iot-reference-esp32c3/build/FeaturedFreeRTOSIoTIntegration.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32c3.peripherals.ld -T esp32c3.rom.ld -T esp32c3.rom.api.ld -T esp32c3.rom.libgcc.ld -T esp32c3.rom.newlib.ld -T esp32c3.rom.version.ld -T esp32c3.rom.eco3.ld -T memory.ld -T sections.ld @CMakeFiles\FeaturedFreeRTOSIoTIntegration.elf.rsp -o FeaturedFreeRTOSIoTIntegration.elf  && cd ."
C:/JasonSoftware/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: FeaturedFreeRTOSIoTIntegration.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
C:/JasonSoftware/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: DRAM segment data does not fit.
C:/JasonSoftware/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: DRAM segment data does not fit.
C:/JasonSoftware/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: section .iram0.text VMA [40380000,40394fa9] overlaps section .dram0.bss VMA [3fc98380,404a4baf]
C:/JasonSoftware/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 8217760 bytes
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\JasonSoftware\Espressif\tools\ninja\1.11.1\ninja.exe" terminated with exit code: 1.

ActoryOu, 2024-07-27:
Hi @abarke,
Thanks for bringing this to us! I'm looking into this and create PR as soon as I can.

Thank you again!

ActoryOu, 2024-07-28:
Hi @abarke, @JasonYan324,
Could you help re-compile with #99? I fixed the issue locally, but I'd like to have your result with the change.

Thank you.

JasonYan324, 2024-07-28:
> Hi @abarke, @JasonYan324, Could you help re-compile with #99? I fixed the issue locally, but I'd like to have your result with the change.
> 
> Thank you.

sorry, so much informations. i do not know whic should follow. I just check this:

"Test Steps
idf.py menuconfig (Featured FreeRTOS IoT Integration -> Enable OTA demo -> check)
idf.py build"

and it is checked as default, and i do uncheck, check, then save. after then, comment out my define for "otaconfigMAX_NUM_OTA_DATA_BUFFERS", buil result:

D:/99ESP/T4/iot-reference-esp32c3/main/demo_tasks/ota_over_mqtt_demo/ota_over_mqtt_demo.c:234:36: error: 'otaconfigMAX_NUM_OTA_DATA_BUFFERS' undeclared here (not in a function)
  234 | static OtaDataEvent_t dataBuffers[ otaconfigMAX_NUM_OTA_DATA_BUFFERS ] = { 0 };

And i do the test as follow:
1.make a define otaconfigMAX_NUM_OTA_DATA_BUFFERS 1024
2.comment out vStartOTACodeSigningDemo
        #if 0 //CONFIG_GRI_ENABLE_OTA_DEMO
            #if CONFIG_GRI_OUTPUT_CERTS_KEYS
                ESP_LOGI( TAG, "\nCS Cert: \nLength: %zu\n%s",
                          strlen( pcAwsCodeSigningCertPem ),
                          pcAwsCodeSigningCertPem );
            #endif /* CONFIG_GRI_OUTPUT_CERTS_KEYS */
 
            ESP_LOGI( TAG, "Application version number: %u.%u.%u",
                      CONFIG_GRI_OTA_DEMO_APP_VERSION_MAJOR,
                      CONFIG_GRI_OTA_DEMO_APP_VERSION_MINOR,
                      CONFIG_GRI_OTA_DEMO_APP_VERSION_BUILD );
 
            if( otaPal_SetCodeSigningCertificate( pcAwsCodeSigningCertPem ) )
            {
                vStartOTACodeSigningDemo();
            }
            else
            {
                ESP_LOGE( TAG,
                          "Failed to set the code signing certificate for the AWS OTA "
                          "library. OTA demo will not be started." );
            }
        #endif /* CONFIG_GRI_ENABLE_OTA_DEMO */

build result:
[6/8] Linking CXX executable FeaturedFreeRTOSIoTIntegration.elf
FAILED: FeaturedFreeRTOSIoTIntegration.elf
cmd.exe /C "cd . && C:\JasonSoftware\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imc_zicsr_zifencei -nostartfiles -march=rv32imc_zicsr_zifencei --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C3=0 -Wl,--Map=D:/99ESP/T4/iot-reference-esp32c3/build/FeaturedFreeRTOSIoTIntegration.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32c3.peripherals.ld -T esp32c3.rom.ld -T esp32c3.rom.api.ld -T esp32c3.rom.libgcc.ld -T esp32c3.rom.newlib.ld -T esp32c3.rom.version.ld -T esp32c3.rom.eco3.ld -T memory.ld -T sections.ld @CMakeFiles\FeaturedFreeRTOSIoTIntegration.elf.rsp -o FeaturedFreeRTOSIoTIntegration.elf  && cd ."
C:/JasonSoftware/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: FeaturedFreeRTOSIoTIntegration.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
C:/JasonSoftware/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: DRAM segment data does not fit.
C:/JasonSoftware/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: DRAM segment data does not fit.
C:/JasonSoftware/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: section .iram0.text VMA [40380000,40394f03] overlaps section .dram0.bss VMA [3fc98380,404a3ddf]
C:/JasonSoftware/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 8214224 bytes
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

JasonYan324, 2024-07-28:
Actually, before fix the keep alive issue, the code form master branch can be build and running. I use git to download the soure code at 2024.07.16

JasonYan324, 2024-07-28:
> Hi @abarke, @JasonYan324, Could you help re-compile with #99? I fixed the issue locally, but I'd like to have your result with the change.
> 
> Thank you.

Actually, in my iot-reference-esp32c3\main\demo_tasks\ota_over_mqtt_demo\ota_over_mqtt_demo_config.h does not hav a define name "otaconfigMAX_NUM_OTA_DATA_BUFFERS"

JasonYan324, 2024-07-28:
sorry  to make trouble. may be is git download issue. i download againt and make the change as follow, build ok, menuconfig works too, amazing!!!!!!

1.components\FreeRTOS-Libraries-Integration-Tests\CMakeLists.txt
# This gives OTA_PAL_TEST_SOURCES, and OTA_PAL_TEST_INCLUDE_DIRS
#include(${CMAKE_CURRENT_LIST_DIR}/FreeRTOS-Libraries-Integration-Tests/src/ota_pal_test.cmake)

comment out include(${CMAKE_CURRENT_LIST_DIR}/FreeRTOS-Libraries-Integration-Tests/src/ota_pal_test.cmake)

2\main\demo_tasks\ota_over_mqtt_demo\ota_over_mqtt_demo.c
static OtaDataEvent_t dataBuffers[512] = {0};//[ otaconfigMAX_NUM_OTA_DATA_BUFFERS ] = { 0 };

does not use otaconfigMAX_NUM_OTA_DATA_BUFFERS

JasonYan324, 2024-07-28:
> sorry to make trouble. may be is git download issue. i download againt and make the change as follow, build ok, menuconfig works too, amazing!!!!!! [...]

But, when input Broker endpoint in menuconfig, build result: dram0_0_seg' overflowed by 4019360 bytes. i dont understand
[1057/1059] Linking CXX executable FeaturedFreeRTOSIoTIntegration.elf
FAILED: FeaturedFreeRTOSIoTIntegration.elf
cmd.exe /C "cd . && C:\JasonSoftware\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imc_zicsr_zifencei -nostartfiles -march=rv32imc_zicsr_zifencei --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C3=0 -Wl,--Map=D:/0/build/FeaturedFreeRTOSIoTIntegration.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32c3.peripherals.ld -T esp32c3.rom.ld -T esp32c3.rom.api.ld -T esp32c3.rom.libgcc.ld -T esp32c3.rom.newlib.ld -T esp32c3.rom.version.ld -T esp32c3.rom.eco3.ld -T memory.ld -T sections.ld @CMakeFiles\FeaturedFreeRTOSIoTIntegration.elf.rsp -o FeaturedFreeRTOSIoTIntegration.elf  && cd ."
C:/JasonSoftware/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: FeaturedFreeRTOSIoTIntegration.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
C:/JasonSoftware/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: DRAM segment data does not fit.
C:/JasonSoftware/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: DRAM segment data does not fit.
C:/JasonSoftware/Espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 4019360 bytes
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

JasonYan324, 2024-07-29:
> But, when input Broker endpoint in menuconfig, build result: dram0_0_seg' overflowed by 4019360 bytes. i dont understand [1057/1059] Linking CXX executable FeaturedFreeRTOSIoTIntegration.elf FAILED: [...]

I have found the reason. After compare the new and the old SDKconfig, i found then the new SDKconfig miss the follow define（but have in the sdkconfig.defaults）

# AWS OTA
#
CONFIG_LOG2_FILE_BLOCK_SIZE=12
CONFIG_MAX_NUM_BLOCKS_REQUEST=8
CONFIG_MAX_NUM_OTA_DATA_BUFFERS=8
CONFIG_ALLOW_DOWNGRADE=0
CONFIG_OTA_DATA_OVER_MQTT=y
# CONFIG_OTA_DATA_OVER_HTTP is not set
CONFIG_OTA_DATA_OVER_MQTT_PRIMARY=y
CONFIG_OTA_PRIMARY_DATA_PROTOCOL=1

#

and befor my define for otaconfigMAX_NUM_OTA_DATA_BUFFERS is 512, so dram0_0_seg overflowed, after i make as 8, bulide succeed.

so, the big problem is why the update code at 2024.7.26 will miss the OTA define

ActoryOu, 2024-07-29:
Hi @JasonYan324,
Thanks for your feedback! Yes, it needs a clean build after changing SDKconfig file.

JasonYan324, 2024-07-29:
> Hi @JasonYan324, Thanks for your feedback! Yes, it needs a clean build after changing SDKconfig file.

no, my point is the new sdkconfig missing these define
CONFIG_LOG2_FILE_BLOCK_SIZE=12
CONFIG_MAX_NUM_BLOCKS_REQUEST=8
CONFIG_MAX_NUM_OTA_DATA_BUFFERS=8
CONFIG_ALLOW_DOWNGRADE=0
CONFIG_OTA_DATA_OVER_MQTT=y

JasonYan324, 2024-07-29:
> Hi @JasonYan324, Thanks for your feedback! Yes, it needs a clean build after changing SDKconfig file.

and, can you help me to find out what cause this problem? Thanks. Afer i fix the bulid issue, it can not running, reboot and reboot...

I (269) uart: queue free spaces: 10
I (269) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (269) main:
CS Cert:
Length: 0

I (269) main: Application version number: 0.0.0
E (269) temp_sub_pub_and_led_control_demo: Subscribe Topic: /Meshed/Jason/Subscribe
I (269) temp_sub_pub_and_led_control_demo: Sending subscribe request to agent for topic filter: /Meshed/Jason/Subscribe with id 1
I (269) ota_over_mqtt_demo: OTA over MQTT demo, Application version 0.0.0
I (269) pp: pp rom version: 9387209
I (269) net80211: net80211 rom version: 9387209
I (279) wifi:wifi driver task: 3fcc3cfc, prio:23, stack:6656, core=0
I (279) wifi:wifi firmware version: 0caa81945
I (279) wifi:wifi certification version: v7.0
I (279) wifi:config NVS flash: enabled
I (279) wifi:config nano formating: disabled
I (279) wifi:Init data frame dynamic rx buffer num: 32
I (279) wifi:Init static rx mgmt buffer num: 5
I (279) wifi:Init management short buffer num: 32
I (279) wifi:Init dynamic tx buffer num: 32
I (279) wifi:Init static tx FG buffer num: 2
I (279) wifi:Init static rx buffer size: 1600
I (279) wifi:Init static rx buffer num: 10
I (279) wifi:Init dynamic rx buffer num: 32
I (279) wifi_init: rx ba win: 6
I (279) wifi_init: accept mbox: 6
I (279) wifi_init: tcpip mbox: 32
I (279) wifi_init: udp mbox: 6
I (279) wifi_init: tcp mbox: 6
I (279) wifi_init: tcp tx win: 5760
I (279) wifi_init: tcp rx win: 5760
I (279) wifi_init: tcp mss: 1440
I (279) wifi_init: WiFi IRAM OP enabled
I (279) wifi_init: WiFi RX IRAM OP enabled
I (279) wifi_prov_scheme_ble: BT memory released
I (289) app_wifi: Already provisioned, starting Wi-Fi STA
I (289) wifi_prov_scheme_ble: BTDM memory released
I (289) phy_init: phy_version 1180,01f2a49,Jun  4 2024,16:34:25
I (329) wifi:mode : sta (40:4c:ca:89:c1:84)
I (329) wifi:enable tsf
I (2739) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
I (2739) wifi:state: init -> auth (0xb0)
I (2739) wifi:state: auth -> assoc (0x0)
I (2769) wifi:state: assoc -> run (0x10)
I (2789) wifi:connected with Rompa Guest, aid = 23, channel 1, BW20, bssid = f6:2a:7d:1e:d6:b9
I (2789) wifi:security: WPA2-PSK, phy: bgn, rssi: -47
I (2789) wifi:pm start, type: 1

I (2789) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (2789) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
I (2819) wifi:<ba-add>idx:0 (ifx:0, f6:2a:7d:1e:d6:b9), tid:4, ssn:8, winSize:64
I (2829) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (3789) core_mqtt_agent_manager: WiFi connected.
I (3789) app_wifi: Connected with IP Address:192.168.2.158
I (3789) esp_netif_handlers: sta ip: 192.168.2.158, mask: 255.255.255.0, gw: 192.168.2.1
I (3789) main_task: Returned from app_main()
I (3799) wifi:<ba-add>idx:1 (ifx:0, f6:2a:7d:1e:d6:b9), tid:0, ssn:1018, winSize:64
I (5799) core_mqtt_agent_manager: TLS connection established.
I (6539) coreMQTT: MQTT connection established with the broker.
I (6539) core_mqtt_agent_manager: Session present: 0

I (6539) core_mqtt_agent_manager: coreMQTT-Agent connected.
I (6539) sub_pub_unsub_demo: coreMQTT-Agent connected.
I (6539) ota_over_mqtt_demo: coreMQTT-Agent connected. Resuming OTA agent.
I (6539) temp_sub_pub_and_led_control_demo: coreMQTT-Agent connected.
[... 112 lines cut ...]
I (244) app_init: App version:      v202212.00-93-g87cc57a-dirty
I (244) app_init: Compile time:     Jul 29 2024 11:46:27
I (244) app_init: ELF file SHA256:  313fa59c2...
I (245) app_init: ESP-IDF:          v5.3-dirty
I (245) efuse_init: Min chip rev:     v0.3
I (245) efuse_init: Max chip rev:     v1.99
I (245) efuse_init: Chip rev:         v0.4
I (245) heap_init: Initializing. RAM available for dynamic allocation:
I (245) heap_init: At 3FCB5280 len 0000AD80 (43 KiB): RAM
I (245) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM
I (246) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM
I (246) heap_init: At 50000200 len 00001DE8 (7 KiB): RTCRAM
I (247) spi_flash: detected chip: generic
I (247) spi_flash: flash io: dio
I (248) sleep: Configure to isolate all GPIO pins in sleep state
I (248) sleep: Enable automatic switching of GPIO sleep configuration
I (248) coexist: coex firmware version: dab85ae96
I (249) coexist: coexist rom version 9387209
I (249) main_task: Started on CPU0
I (249) main_task: Calling app_main()
I (249) esp_secure_cert: Pre-provisioned partition information:
I (249) esp_secure_cert: partition format: CUST_FLASH, partition name: esp_secure_cert
W (249) mmap: paddr block is mapped already, vaddr_start: 0x3c120000, size: 0x10000
W (249) mmap: paddr block is mapped already, vaddr_start: 0x3c120000, size: 0x10000
I (249) main:

ActoryOu, 2024-07-29:
Hi @JasonYan324,

> CONFIG_LOG2_FILE_BLOCK_SIZE=12
> CONFIG_MAX_NUM_BLOCKS_REQUEST=8
> CONFIG_MAX_NUM_OTA_DATA_BUFFERS=8
> CONFIG_ALLOW_DOWNGRADE=0
> CONFIG_OTA_DATA_OVER_MQTT=y

Those macros are used in old OTA demo. It's replaced with newer one by #88.
Could you help try new OTA?
BTW, this issue is resolved by #99, I'll close this one.
If you have any new finding, please create a new issue then we can discuss in a clear thread.

Thank you.

ActoryOu, 2024-07-29:
This issue is resolved by #99.

abarke, 2024-07-29:
Thanks @ActoryOu I pulled main and now it's working