FreeRTOS/iot-reference-esp32 #97: [BUG] <Multiple CMake errors when trying to run MenuConfig in VsCode Studio>

wreyford, 2024-07-26:
Multiple cMake errors when trying to run MenuConfig of the cloned current repo, in VSCode studio.

ESP32S3
ESP-IDF 5.2.0
Windows 10

> Executing action: confserver
Running cmake in directory C:\iot-reference-esp32c3\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=c:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 C:\iot-reference-esp32c3"...
-- Found Git: C:/Espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 

-- Component directory C:/iot-reference-esp32c3/components/esp-aws-iot/libraries/common does not contain a CMakeLists.txt file. No component will be added

-- Component directory C:/iot-reference-esp32c3/components/esp-aws-iot does not contain a CMakeLists.txt file. No component will be added

-- ccache will be used for faster recompilation

-- The C compiler identification is GNU 13.2.0

-- The CXX compiler identification is GNU 13.2.0

-- The ASM compiler identification is GNU

-- Found assembler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe

-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - done

-- Check for working C compiler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped

-- Detecting C compile features

-- Detecting C compile features - done

-- Detecting CXX compiler ABI info

-- Detecting CXX compiler ABI info - done

-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped

-- Detecting CXX compile features

-- Detecting CXX compile features - done

-- Building ESP-IDF components for target esp32s3

Processing 4 dependencies:

[1/4] espressif/cbor (0.6.0~1)

[2/4] espressif/esp_secure_cert_mgr (2.4.1)

[3/4] espressif/led_strip (2.5.4)

[4/4] idf (5.2.0)

---------------------------ERROR--------------------------

CMake Error at C:/Espressif/v5.2/esp-idf/tools/cmake/component.cmake:224 (message):
  CMake Warning (dev) at build_properties.temp.cmake:8:

    Syntax Warning in cmake code at column 106

-----------------------END OF ERROR-----------------------
-- Configuring incomplete, errors occurred!
See also "C:/iot-reference-esp32c3/build/CMakeFiles/CMakeOutput.log".

---------------------------ERROR--------------------------

  

    Argument not separated from preceding token by whitespace.

  Call Stack (most recent call first):

    C:/Espressif/v5.2/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:3 (include)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  

  CMake Error at
  C:/iot-reference-esp32c3/components/FreeRTOS-Libraries-Integration-Tests/CMakeLists.txt:11
  (include):

    include could not find requested file:

  

      C:/iot-reference-esp32c3/components/FreeRTOS-Libraries-Integration-Tests/FreeRTOS-Libraries-Integration-Tests/src/ota_pal_test.cmake

  Call Stack (most recent call first):

    C:/Espressif/v5.2/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:107 (include)
    C:/Espressif/v5.2/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:127 (__component_get_requirements)

  

  

Call Stack (most recent call first):
  C:/Espressif/v5.2/esp-idf/tools/cmake/build.cmake:578 (__component_get_requirements)
  C:/Espressif/v5.2/esp-idf/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:7 (project)

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

cmake failed with exit code 1, output of the command is in the C:\iot-reference-esp32c3\build\log\idf_py_stderr_output_15388 and C:\iot-reference-esp32c3\build\log\idf_py_stdout_output_15388

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

SDK Configuration editor confserver process exited with code: 2
-----------------------END OF ERROR-----------------------

wreyford, 2024-07-26:
In the project folder, go to Components -> FreeRTOS-Libraries-Integration-Tests, and edit the CMakeLists.txt, and comment out the following line
`# This gives OTA_PAL_TEST_SOURCES, and OTA_PAL_TEST_INCLUDE_DIRS`
`#include(${CMAKE_CURRENT_LIST_DIR}/FreeRTOS-Libraries-Integration-Tests/src/ota_pal_test.cmake)`

After this MenuConfig worked.  It appears ota_pal_test.cmake was dropped 1/12 ago.

ActoryOu, 2024-07-27:
Hi @wreyford,
Thanks for bringing this to us! I'm looking into this and create PR as soon as I can.

Thank you again!

ActoryOu, 2024-07-28:
Hi @wreyford,
Could you help re-compile with #99? I fixed the issue locally, but I'd like to have your result with the change.

Thank you.

ActoryOu, 2024-07-29:
This issue should be resolved by #99. I'm closing it.
Feel free to reopen if it's not resolved with your finding.

Thank you.

closed 2024-07-29 by ActoryOu