FreeRTOS/iot-reference-esp32 #101: [Feature Request] Separate branch for LTS202407.00 release and 202210.01-LTS

NightSkySK, 2024-07-29:
Can we have a separate branch to trace changes applied based on 202210.01 LTS and a separate one for the new LTS202407 release?

Some users (including myself) don't want yet to transfer to the new LTS release and searching through commits which is for the new LTS which is related to the previose one is tricky.

Please consider keeping similar split as [esp-aws-iot repo](https://github.com/espressif/esp-aws-iot) 

[image]

Than if any new commit might be applied to previose  202210.01 LTS easier

kstribrnAmzn, 2024-07-29:
Thanks for the suggestion. For now, we'll keep using our existing tag strategy. If we do end up applying a patch to our older (202210.01) LTS release we will consider using this strategy.

NightSkySK, 2024-07-30:
The last release was published in Dec 2022. 
[image]
Since then hundreds of commits were published which were still applicable to (202210.01) LTS 
[image]

And now since one month you've started to move into LTS202407.00, there is no clear borderline between those events. That's why I wish to have a separate branch.

ActoryOu, 2024-08-08:
Hi @NightSkySK,
IMHO, either stay in old tag or using new tag is fine. It's not recommended to cherry pick commits back to the old tag unless you know what you're doing.
If there is any critical issue using old tag, please let us know.

Thank you.

ActoryOu, 2024-10-18:
Closing the issue as there are no further questions for now.

closed 2024-10-18 by ActoryOu