FreeRTOS/iot-reference-esp32 #102: [Feature Request] <Reboot>

JasonYan324, 2024-07-29:
After merge main branch which fix issue with #99, app can not run, auto reboot.
hardware: ESP32-C3
IDF version: V5.2.2 and V5.3
IDE: VS code

what i did:
1.Write cert int to ESP32 by:
python managed_components/espressif__esp_secure_cert_mgr/tools/configure_esp_secure_cert.py -p "COM19" --configure_ds --keep_ds_data_on_host --ca-cert "D:/0/main/certs/aws-root-ca.pem" --device-cert "D:/0/main/certs/certificate.pem.crt" --private-key "D:/0/main/certs/private.pem.key" --target_chip "esp32c3" --secure_cert_type cust_flash --priv_key_algo "RSA" "2048"
2.write endpoint and thing name by menuconfig
3.bulid and flash
4.running log(always reboot):
PS D:\0> & set IDF_PATH='D:/_00SDK/Espressif/frameworks/esp-idf-v5.3/'
PS D:\0> & 'D:/_00SDK/Espressif/python_env/idf5.3_py3.11_env/Scripts/python.exe' 'D:\_00SDK\Espressif\frameworks\esp-idf-v5.3\tools\idf_monitor.py' -p COM19 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 'd:\0\build\FeaturedFreeRTOSIoTIntegration.elf'
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM19 instead...
--- esp-idf-monitor 1.4.0 on \\.\COM19 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xe (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038903e
0x4038903e: rv_utils_wait_for_intr at D:/_00SDK/Espressif/frameworks/esp-idf-v5.3/components/riscv/include/riscv/rv_utils.h:55
 (inlined by) esp_cpu_wait_for_intr at D:/_00SDK/Espressif/frameworks/esp-idf-v5.3/components/esp_hw_support/cpu.c:62

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1188
load:0x403cc710,len:0x8d4
load:0x403ce710,len:0x2c8c
entry 0x403cc710
I (235) cpu_start: Unicore app
I (244) cpu_start: Pro cpu start user code
I (244) cpu_start: cpu freq: 160000000 Hz
I (244) app_init: Application information:
I (244) app_init: Project name:     FeaturedFreeRTOSIoTIntegration
I (244) app_init: App version:      v202212.00-100-ge45d25b
I (244) app_init: Compile time:     Jul 29 2024 15:42:29
I (245) app_init: ELF file SHA256:  eba7b8572...
I (245) app_init: ESP-IDF:          v5.3-dirty
I (245) efuse_init: Min chip rev:     v0.3
I (245) efuse_init: Max chip rev:     v1.99
I (245) efuse_init: Chip rev:         v0.4
I (245) heap_init: Initializing. RAM available for dynamic allocation:
I (245) heap_init: At 3FCA9250 len 00016DB0 (91 KiB): RAM
I (246) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM
I (246) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM
I (246) heap_init: At 50000200 len 00001DE8 (7 KiB): RTCRAM
I (247) spi_flash: detected chip: generic
I (247) spi_flash: flash io: dio
I (248) sleep: Configure to isolate all GPIO pins in sleep state
I (248) sleep: Enable automatic switching of GPIO sleep configuration
I (248) coexist: coex firmware version: dab85ae96
I (249) coexist: coexist rom version 9387209
I (249) main_task: Started on CPU0
I (249) main_task: Calling app_main()
I (249) esp_secure_cert: Pre-provisioned partition information:
I (249) esp_secure_cert: partition format: CUST_FLASH, partition name: esp_secure_cert        
W (249) mmap: paddr block is mapped already, vaddr_start: 0x3c120000, size: 0x10000
W (249) mmap: paddr block is mapped already, vaddr_start: 0x3c120000, size: 0x10000
I (249) main:
Device Cert:
Length: 1221
-----BEGIN CERTIFICATE-----
...
-----END CERTIFICATE-----

I (249) main:
CA Cert:
Length: 1207
-----BEGIN CERTIFICATE-----
[... 172 lines cut ...]
I (244) app_init: App version:      v202212.00-100-ge45d25b
I (244) app_init: Compile time:     Jul 29 2024 15:42:29
I (244) app_init: ELF file SHA256:  eba7b8572...
I (244) app_init: ESP-IDF:          v5.3-dirty
I (244) efuse_init: Min chip rev:     v0.3
I (244) efuse_init: Max chip rev:     v1.99
I (244) efuse_init: Chip rev:         v0.4
I (245) heap_init: Initializing. RAM available for dynamic allocation:
I (245) heap_init: At 3FCA9250 len 00016DB0 (91 KiB): RAM
I (245) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM
I (245) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM
I (245) heap_init: At 50000200 len 00001DE8 (7 KiB): RTCRAM
I (246) spi_flash: detected chip: generic
I (247) spi_flash: flash io: dio
I (247) sleep: Configure to isolate all GPIO pins in sleep state
I (247) sleep: Enable automatic switching of GPIO sleep configuration
I (248) coexist: coex firmware version: dab85ae96
I (248) coexist: coexist rom version 9387209
I (249) main_task: Started on CPU0
I (249) main_task: Calling app_main()
I (249) esp_secure_cert: Pre-provisioned partition information:
I (249) esp_secure_cert: partition format: CUST_FLASH, partition name: esp_secure_cert        
W (249) mmap: paddr block is mapped already, vaddr_start: 0x3c120000, size: 0x10000
W (249) mmap: paddr block is mapped already, vaddr_start: 0x3c120000, size: 0x10000
5.but if i disable "Enable OTA demo", then rebulid and flash, app can be run no reboot

ActoryOu, 2024-07-29:
Hi @JasonYan324,
Could you elaborate on the steps you did for this?
Is it a clean clone? What configurations have you checked in menuconfig?

Thank you.

JasonYan324, 2024-07-30:
> Hi @JasonYan324, Could you elaborate on the steps you did for this? Is it a clean clone? What configurations have you checked in menuconfig?
> 
> Thank you.

Hi @ActoryOu，
What I did following as below(wifi setup before, no need to setup again)：
1. git clone again by using git clone https://github.com/FreeRTOS/iot-reference-esp32c3.git --recurse-submodules
from the log clone done without any fail
3. idf.py menuconfig to setup MQTT endpoint and thing name, log out change from UART0 to USB
4. python managed_components/espressif__esp_secure_cert_mgr/tools/configure_esp_secure_cert.py ... to write the MQTT certs again
5. vs code open the soure code, then select target esp32c3，com port ，uart to flash
6. then build，and no error
[image]
8. then flash, and done
[image]
10. then start monitor，and got the same result as above, alway detect ota_over_mqtt_demo/ota_over_mqtt_demo then reboot
[image]
11. and idf.py menuconfig to uncheck "Enable OTA demo", rebuild, flash, then can be running.
12. even worse, I compiled with IDF5.2 version and couldn't start it directly, restarting as soon as it started. We won't discuss this for now. If necessary, we can open another issue

JasonYan324, 2024-07-30:
@ActoryOu 
so, what do you suggest I do for this issue. Thanks

JasonYan324, 2024-07-30:
what i did in menuconfig
[image]
[image]

ActoryOu, 2024-07-31:
Hi @JasonYan324,
Thanks for the detail! If it's possible to use UART logging for testing purpose?
We'd find some time try this but it might take a few days.

Thank you.

JasonYan324, 2024-07-31:
> Hi @JasonYan324, Thanks for the detail! Is it possible to chage back to UART logging for testing purpose? We'd find some time try this but it might take a few days.
> 
> Thank you.

Thanks a lot! I have try from a new soure code, olny setup endpoint and thing name in menuconifg, but the result is the same as above.

ActoryOu, 2024-08-01:
Hi @JasonYan324,
We found the root cause and raised #103 to fix this.
Could you help try once?

Thank you.

ActoryOu, 2024-08-02:
This issue is being closed as the proposed fix has been merged into the codebase. However, if the same problem persists, feel free to reopen this issue for further investigation and resolution.

closed 2024-08-02 by ActoryOu

JasonYan324, 2024-08-02:
> This issue is being closed as the proposed fix has been merged into the codebase. However, if the same problem persists, feel free to reopen this issue for further investigation and resolution.

the code has no merge now， can you tell me where and what to change for test?
[image]

JasonYan324, 2024-08-02:
> the code has no merge now， can you tell me where and what to change for test? [...]

Sorry, I foud what you change

ActoryOu, 2024-08-02:
Sorry for confusing. #103 is the patch to fix this issue, and it's yet merged. Please give it a try.

ActoryOu, 2024-08-02:
Note that it's merged.