FreeRTOS/iot-reference-esp32 #106: [DOC]

IotSHCJC, 2024-08-07:
Hello,
       My development environment is Windows, and when I enter this command, I idf.py set-target esp32s3 and I get the following error

`-- Component directory D:/soft/iot-reference-esp32c3/components/esp-aws-iot/libraries/common does not contain a CMakeLists.txt file. No component will be added
-- Component directory D:/soft/iot-reference-esp32c3/components/esp-aws-iot does not contain a CMakeLists.txt file. No component will be added
CMake Error at D:/soft/espressif_1/esp-idf_5.0.3/frameworks/esp-idf-v5.0.3/tools/cmake/targets.cmake:81 (message):
  Toolchain file
  D:/soft/espressif_1/esp-idf_5.0.3/frameworks/esp-idf-v5.0.3/tools/cmake/toolchain-esp32c3

  .cmake not found
Call Stack (most recent call first):
  D:/soft/espressif_1/esp-idf_5.0.3/frameworks/esp-idf-v5.0.3/tools/cmake/project.cmake:413 (__target_set_toolchain)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
CMake Warning:
  Value of IDF_TARGET contained a newline; truncating

cmake failed with exit code 1, output of the command is in the D:\soft\iot-reference-esp32c3\build\log\idf_py_stderr_output_27284 and D:\soft\iot-reference-esp32c3\build\log\idf_py_stdout_output_27284
`

tony-josi-aws, 2024-08-07:
@IotSHCJC 

Could you try running this on a Linux PC

aggarg, 2024-08-26:
You need to follow these instructions to setup ESP-IDF - https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/index.html. 

Assuming that you followed [manual installation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/windows-setup.html), the IDF is installed in `%userprofile%\esp\esp-idf` directory. You need to run the `export.bat`(if you are using cmd.exe) or `export.ps1` (if you are using power shell) from your command line. After that you should be able to use `idf.py`.

tony-josi-aws, 2024-09-10:
This issue is not reproducible at our end on the Windows machine with ESPIDF version v5.0.7, hence closing this. Feel free to reopen if you think otherwise.

closed 2024-09-10 by tony-josi-aws