FreeRTOS/iot-reference-esp32 #107: [Feature Request] <VS  Code Debug ESP32C3>

JasonYan324, 2024-08-08:
Dear All:
Can some body give me some suggest for the debug by uising VS Code? Thansk!

IDF: 5.3
IDE: VS Code 1.92.0
TagetBoard:  ESP32-C3-DevKit-RUST-1
Connection: built-in JATA, OpenOCD

launch.json:
{
    "configurations": [
        {
        "type": "gdbtarget",
        "request": "attach",
        "name": "ESP32C3",
        "program": "${workspaceFolder}/build/${command:espIdf.getProjectName}.elf",
        "initCommands": [
            "set remote hardware-watchpoint-limit 10",
            "mon reset halt",
            "maintenance flush register-cache",
            "thb app_main",
            "c"
        ],
        "gdb": "${command:espIdf.getToolchainGdb}",
        "target": {
            "connectCommands": [
            "set remotetimeout 500",
            "-target-select extended-remote localhost:3333"
            ]
        }
        }
    ]
}

setting.json:
{
    "idf.adapterTargetName": "esp32c3",
    "files.associations": {
        "freertos.h": "c",
        "task.h": "c",
        "array": "c",
        "vector": "c",
        "xstring": "c",
        "xutility": "c",
        "gpio.h": "c",
        "temp_sensor.h": "c",
        "uart.h": "c",
        "esp_log.h": "c",
        "esp_wifi.h": "c",
        "assert.h": "c",
        "subscription_manager.h": "c",
        "stddef.h": "c",
        "string.h": "c",
        "freertosconfig.h": "c"
    },
    "idf.portWin": "COM8",
    "idf.flashType": "JTAG",
    "idf.openOcdConfigs": [
        "board/esp32c3-builtin.cfg"
    ]
}

1.when start debug, it frequently prompt connection failure, just like this:
[image]
[image]

the "DEBUG CONSOLE" output:
GNU gdb (esp-gdb) 14.2_20240403
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-w64-mingw32 --target=riscv32-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
warning: could not convert 'main' from the host encoding (CP1252) to UTF-32.
This normally should not happen, please file a bug report.
set remotetimeout 500

2.Even if it is connected, I find that there cannot be a breakpoint.

a. the "DEBUG CONSOLE" output::
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-w64-mingw32 --target=riscv32-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
warning: could not convert 'main' from the host encoding (CP1252) to UTF-32.
This normally should not happen, please file a bug report.
set remotetimeout 500
0x420db2da in semihosting_call_noerrno (id=536870912, id@entry=259, data=data@entry=0x3fcae1e4) at C:/JasonSoftware/Espressif/frameworks/esp-idf-v5.3/components/riscv/include/riscv/semihosting.h:77
mon reset halt
JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
[esp32c3] Reset cause (3) - (Software core reset)
maintenance flush register-cache
Register cache flushed.
thb app_main
Hardware assisted breakpoint 1 at 0x42006668: file C:/Code/forDebug/main/main.c, line 153.
GDB unhandled notify: breakpoint-created: {"bkpt":{"number":"1","type":"hw breakpoint","disp":"del","enabled":"y","addr":"0x42006668","func":"app_main","file":"C:/Code/forDebug/main/main.c","fullname":"C:\\Code\\forDebug\\main\\main.c","line":"153","thread-groups":["i1"],"times":"0","original-location":"app_main"}}
c
Continuing.
[New Thread 1070259692]
[Remote target exited]
[New Thread 1070261060]
[New Thread 1070254924]
[Switching to Thread 1070259692]

Thread 2 "main" hit Temporary breakpoint 1, app_main () at C:/Code/forDebug/main/main.c:153
153	    xNetworkContext.pcHostname = CONFIG_GRI_MQTT_ENDPOINT;

Thread 2 "main" received signal SIGINT, Interrupt.
0x42005180 in usb_serial_jtag_ll_txfifo_writable () at C:/JasonSoftware/Espressif/frameworks/esp-idf-v5.3/components/hal/esp32c3/include/hal/usb_serial_jtag_ll.h:160
160	    return USB_SERIAL_JTAG.ep1_conf.serial_in_ep_data_free;

b. the "OUTPUT" output:

[OpenOCD]
Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

debug_level: 2

C:\JasonSoftware\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin\openocd.exe

Info : only one transport 
option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000

Info : Listening on port 6666 for tcl connections

Info : Listening on port 4444 for telnet connections

Info : esp_usb_jtag: serial (40:4C:CA:89:67:2C)

Info : esp_usb_jtag: Device found. Ba
se speed 40000KHz, div range 1 to 255

Info : clock speed 40000 kHz

Info : JTAG tap: esp32c3
.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)

C:\JasonSoftware\Espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin\riscv32-esp-elf-gdb.exe

Info : [esp32c3] datacount=2 progbufsize=16

Info : [esp32c3] Examined RISC-V co
re; found 1 harts
Info : [esp32c3]  XLEN=32, misa=0x40101104
Info : [esp32c3] Examination succeed
Info : starting gdb server for esp32c3 on 3333

Info : Listening on port 3333 for gdb connections

Info : accepting 'gdb' connection on tcp/3333

Warn : No symbols for FreeRTOS!

Info : [esp32c3] Found 8 triggers

Info : Flash mapping 0: 0x10020 -> 0x3c020020, 
34 KB

Info : Auto-detected flash bank 'esp32c3.flas
h' size 4096 KB
Info : Using flash bank 'esp32c3.flash' size 4096 KB

Info : Flash mapping 0: 0x10020 -> 0x3c020020, 34 KB
Info 
: Using flash bank 'esp32c3.irom' size 0 KB

Info : Flash mapping 0: 0x10020 -> 0x3c020
020, 34 KB
Info : Using flash bank 'esp32c3.drom' size 36 KB

Info : Detected FreeRTOS version
: (10.5.1)

Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25
 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)

Info : [esp32c3] Reset cause (3) - (Software core reset)

Info : Detected FreeRTOS version: (10.5.1)

Info : The target is not running when halt was requested, stopping GDB.
Info : The target is not running when halt was requested, stopping GDB.

Info : The target is not running when halt was requested, stopping GDB.

Info : The target is not 
running when halt was requested, stopping GDB.

Info : The target is not running when halt was requested, sto
pping GDB.

Info : The target is not running when halt 
was requested, stopping GDB.

Info : The target is not running when halt was requested, sto
pping GDB.

Info : The target is not running when halt was requested, stopping GDB.

Info : The target is not running when
 halt was requested, stopping GDB.

c. But when in debugging "Info : The target is not running when halt was requested, stopping GDB.", i found "idf.py monitor" has log output

JasonYan324, 2024-08-08:
Be clear than, I uncheck watchdog in menuconfig
[image]

tony-josi-aws, 2024-08-10:
I would suggest posting this in the [Espressif forums](https://esp32.com/search.php?search_id=active_topics&sid=42d869ddfff241c12c6ef2802b8e3b04) as they own and have more expertise in the ESP32 tools.

I'm closing this issue for now; please feel free to reopen it if you think otherwise.

closed 2024-08-10 by tony-josi-aws