FreeRTOS/iot-reference-esp32 #112: [BUG] Slow to publish message, why it is mandatory to wait for prvWaitForCommandAcknowledgment

WilliamFrasson, 2024-08-24:
**Describe the bug**
Please provide a clear and concise description explaining the bug.

**System information**
- Hardware board: ESP32-S3
- IDE used: 5.2.2 
- Operating System: Windows 
- Code version:  69d13095924854a86510ffb0a2025589b64eb4b3  (after https://github.com/FreeRTOS/iot-reference-esp32/pull/103)
- Project/Demo: temp_sub_pub_and_led_control_demo
- If your project is a custom application, please add the relevant code snippet(s) in the section titled `"Code to reproduce bug"`.

**Expected behavior**
during MQTT publish if during normal working, the mqtt connection goes down happen that the task block 
in xTaskNotifyWait( 0,  0, &ulNotification,  portMAX_DELAY ).

while i'm expecting that it exit for some timeout.  but The only timeout i can set   is  xCommandParams.blockTimeMs = waitTimeoutMsec;  that seem to be not working if the publish is already queues.

if i try to avoid the notification xTaskNotifyWait( 0,  0, &ulNotification,  portMAX_DELAY ) the application fastly going in crash

[image]

there is a way to allow the Publish wait for a timeout only without crashing?

**Screenshots or console output**
If appropriate, please paste the console output/error log explaining the issue. If possible, include the call stack.

**Steps to reproduce bug**
    Example:
      1. "I am using project [ ... ], and have configured with [ ... ]"
      2. "When run on [ ... ], I observed that [ ... ]"

**Code to reproduce bug**

      xCommandContext.xTaskToNotify = xTaskGetCurrentTaskHandle();
      
      xCommandContext.ulNotificationValue = ulPublishMessageId;
      xCommandParams.blockTimeMs = waitTimeoutMsec;
      xCommandParams.cmdCompleteCallback = prvPublishCommandCallback;
      xCommandParams.pCmdCompleteCallbackContext = &xCommandContext;

      /* To ensure ulNotification doesn't accidentally hold the expected value
         * as it is to be checked against the value sent from the callback.. */
        ulNotification = ~ulValueToNotify;

        xCommandAdded = MQTTAgent_Publish( &xGlobalMqttAgentContext,
                                           &xPublishInfo,
                                           &xCommandParams );
        configASSERT( xCommandAdded == MQTTSuccess );

        /* For QoS 1 and 2, wait for the publish acknowledgment.  For QoS0,
         * wait for the publish to be sent. */
        ESP_LOGI( TAG,
                  "Task %s waiting for publish %" PRIu32 " to complete.",
                  pcTaskName,
                  ulValueToNotify );

        //prvWaitForCommandAcknowledgment( &ulNotification );
       xTaskNotifyWait( 0,  0, &ulNotification,  portMAX_DELAY );    

**Additional context**
Include any additional context relevant to the issue here.

xuelix, 2024-08-26:
Thanks for reporting, I'll have the team to review your findings and get back to you.

aggarg, 2024-08-29:
The reason for the crash you observe is -
1. [xCommandContext](https://github.com/FreeRTOS/iot-reference-esp32/blob/main/main/demo_tasks/sub_pub_unsub_demo/sub_pub_unsub_demo.c#L456) is created on stack.
2. This context is passed to the [MQTT_Agent](https://github.com/FreeRTOS/iot-reference-esp32/blob/main/main/demo_tasks/sub_pub_unsub_demo/sub_pub_unsub_demo.c#L490).
3. MQTT Agent stores it -
   * https://github.com/FreeRTOS/coreMQTT-Agent/blob/main/source/core_mqtt_agent.c#L492
   * https://github.com/FreeRTOS/coreMQTT-Agent/blob/main/source/core_mqtt_agent.c#L362
4. This context is later accessed when an ACK is received for the command -
    * https://github.com/FreeRTOS/coreMQTT-Agent/blob/main/source/core_mqtt_agent.c#L688
    * https://github.com/FreeRTOS/coreMQTT-Agent/blob/main/source/core_mqtt_agent.c#L629
    * https://github.com/FreeRTOS/coreMQTT-Agent/blob/main/source/core_mqtt_agent.c#L802

If the function [prvPublishToTopic](https://github.com/FreeRTOS/iot-reference-esp32/blob/main/main/demo_tasks/sub_pub_unsub_demo/sub_pub_unsub_demo.c#L444), returns before the PUBACK is received, this `xCommandContext` created in step 1 goes out of scope and results in crash when it is later accessed during PUBACK handling. `portMAX_DELAY` ensures that `prvPublishToTopic` does not return before PUBACK is received and therefore,  `xCommandContext` is not accessed after gonig out of scope. If you do not want to use `portMAX_DELAY`, you need some other way to ensure that the  `xCommandContext` remains in scope till the time PUBACK is received.

WilliamFrasson, 2024-08-29:
The option is to queue messeges to send into a queue and use a different task for publish them. In this way the main task can continue to work.

Beside all, the publish is very slow (as already discussed in https://github.com/FreeRTOS/iot-reference-esp32/issues/44) also for QoS0 when we should expect to be very fast.

aggarg, 2024-08-30:
> The option is to queue messeges to send into a queue and use a different task for publish them. In this way the main task can continue to work.

Note that MQTT Agent also has a queue, so this will be 2 level of indirection. Is it possible to have only task doing all the publishes? 

> Beside all, the publish is very slow (as already discussed in https://github.com/FreeRTOS/iot-reference-esp32/issues/44) also for QoS0 when we should expect to be very fast.

This is something that needs to be investigated. Can you try to put some logs at the following places for QoS0 - 
* When MQTTAgent_Publish is called - https://github.com/FreeRTOS/iot-reference-esp32/blob/main/main/demo_tasks/sub_pub_unsub_demo/sub_pub_unsub_demo.c#L509.
* When the command is added to the MQTT_Agent's command queue - https://github.com/FreeRTOS/coreMQTT-Agent/blob/main/source/core_mqtt_agent.c#L529. 
* When the command is picked up for processing - https://github.com/FreeRTOS/coreMQTT-Agent/blob/main/source/core_mqtt_agent.c#L569. 
* When the command execution is complete - https://github.com/FreeRTOS/coreMQTT-Agent/blob/main/source/core_mqtt_agent.c#L802. 
* When the command completion callback is called - https://github.com/FreeRTOS/iot-reference-esp32/blob/main/main/demo_tasks/sub_pub_unsub_demo/sub_pub_unsub_demo.c#L336. 
* When MQTTAgent_Publish  returns - https://github.com/FreeRTOS/iot-reference-esp32/blob/main/main/demo_tasks/sub_pub_unsub_demo/sub_pub_unsub_demo.c#L513. 

This will help us understand what part is taking more time.

WilliamFrasson, 2024-09-01:
Hello @aggarg 
I uses these tags: 
LOGW - "**PLACEHOLDER_issues#112_001**" : When MQTTAgent_Publish is called - https://github.com/FreeRTOS/iot-reference-esp32/blob/main/main/demo_tasks/sub_pub_unsub_demo/sub_pub_unsub_demo.c#L509.

LOGW - "**PLACEHOLDER_issues#112_002**": When the command is added to the MQTT_Agent's command queue - https://github.com/FreeRTOS/coreMQTT-Agent/blob/main/source/core_mqtt_agent.c#L529.

LOGW - "**PLACEHOLDER_issues#112_003**": When the command is picked up for processing - https://github.com/FreeRTOS/coreMQTT-Agent/blob/main/source/core_mqtt_agent.c#L569.

LOGW - "**PLACEHOLDER_issues#112_004**": When the command execution is complete - https://github.com/FreeRTOS/coreMQTT-Agent/blob/main/source/core_mqtt_agent.c#L802.

LOGW - "**PLACEHOLDER_issues#112_005**": When the command completion callback is called - https://github.com/FreeRTOS/iot-reference-esp32/blob/main/main/demo_tasks/sub_pub_unsub_demo/sub_pub_unsub_demo.c#L336.

LOGW - "**PLACEHOLDER_issues#112_006**":When MQTTAgent_Publish returns - https://github.com/FreeRTOS/iot-reference-esp32/blob/main/main/demo_tasks/sub_pub_unsub_demo/sub_pub_unsub_demo.c#L513.

+ LOGW - "**PLACEHOLDER_issues#112_007**"  After the not    the wait notification " xReceivedEvent = prvWaitForEvent( xMqttEventGroup, MQTT_PUBLISH_COMMAND_COMPLETED_BIT );

as far as i can log We start at  (29252) and finish at (33472) = taking about 4 seconds
 this is the relevant piece of log

PLACEHOLDER_issues#112_001: 
[2024-09-01 23:06:14.192] I (26542) sub_pub_unsub_demo: Task "SubPub0" sending subscribe request to coreMQTT-Agent for topic filter: /filter/SubPub0 with id 4
[2024-09-01 23:06:14.203] W (26552) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_002
[2024-09-01 23:06:14.620] W (26982) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
[2024-09-01 23:06:14.713] W (27072) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_002
[2024-09-01 23:06:14.719] I (27072) coreMQTT: Ack packet deserialized with result: MQTTSuccess.
[2024-09-01 23:06:14.725] I (27072) coreMQTT: State record updated. New state=MQTTPublishDone.
[2024-09-01 23:06:14.730] W (27082) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_004
[2024-09-01 23:06:14.736] I (27082) temp_sub_pub_and_led_control_demo: Rx'ed ack for QoS1 publish from Tx to /filter/TempSubPubLED (P4:F0).
[2024-09-01 23:06:14.747] I (27102) coreMQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.
[2024-09-01 23:06:14.758] I (27102) coreMQTT: State record updated. New state=MQTTPubAckSend.
[2024-09-01 23:06:14.763] I (27112) temp_sub_pub_and_led_control_demo: Received incoming publish message {"temperatureSensor":{ "taskName": "TempSubPubLED", "temperatureValue": 0.000000, "iteration": 3}}
[2024-09-01 23:06:16.780] W (29142) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
[2024-09-01 23:06:16.892] W (29252) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_004
[2024-09-01 23:06:16.898] I (29252) sub_pub_unsub_demo: Subscribe 4 for topic filter /filter/SubPub0 succeeded for task "SubPub0".
**[2024-09-01 23:06:16.904] I (29252) sub_pub_unsub_demo: Task "SubPub0" sending publish request to coreMQTT-Agent with message "SubPub0" on topic "/filter/SubPub0" with ID 5.**
**[2024-09-01 23:06:16.922] W (29272) sub_pub_unsub_demo: PLACEHOLDER_issues#112_001**
**[2024-09-01 23:06:16.926] W (29272) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_002**
**[2024-09-01 23:06:16.931] W (29282) sub_pub_unsub_demo: PLACEHOLDER_issues#112_006**
**[2024-09-01 23:06:16.937] I (29292) sub_pub_unsub_demo: Task "SubPub0" waiting for publish 5 to complete.**
**[2024-09-01 23:06:18.899] W (31262) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
[2024-09-01 23:06:18.905] W (31262) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_004**
[2024-09-01 23:06:19.381] I (31742) temp_sub_pub_and_led_control_demo: Sending publish request to agent with message "{"temperatureSensor":{ "taskName": "TempSubPubLED", "temperatureValue": 0.000000, "iteration": 4}}" on topic "/filter/TempSubPubLED"
[2024-09-01 23:06:19.396] W (31752) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_002
[2024-09-01 23:06:19.402] I (31752) temp_sub_pub_and_led_control_demo: Task TempSubPubLED waiting for publish 4 to complete.
[2024-09-01 23:06:20.910] W (33272) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
[2024-09-01 23:06:21.099] W (33462) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_002
[2024-09-01 23:06:21.105] I (33462) coreMQTT: Ack packet deserialized with result: MQTTSuccess.
[2024-09-01 23:06:21.110] I (33462) coreMQTT: State record updated. New state=MQTTPublishDone.
[2024-09-01 23:06:21.116] W (33472) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_004
**[2024-09-01 23:06:21.126] W (33472) sub_pub_unsub_demo: PLACEHOLDER_issues#112_005**
[2024-09-01 23:06:21.127] I (33482) sub_pub_unsub_demo: Publish 5 succeeded for task "SubPub0".
[2024-09-01 23:06:21.138] I (33482) coreMQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.
**[2024-09-01 23:06:21.143] W (33492) sub_pub_unsub_demo: PLACEHOLDER_issues#112_007**

WilliamFrasson, 2024-09-01:
others logs
start 27526 --> end  31706  -- 4200 msec
```
> I (27506) sub_pub_unsub_demo: Task "SubPub0" sending publish request to coreMQTT-Agent with message "SubPub0" on topic "/filter/SubPub0" with ID 5.
> W (27526) sub_pub_unsub_demo: PLACEHOLDER_issues#112_001
> W (27526) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_002
> W (27536) sub_pub_unsub_demo: PLACEHOLDER_issues#112_006
> I (27546) sub_pub_unsub_demo: Task "SubPub0" waiting for publish 5 to complete.
> W (29516) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (29516) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_004
> W (31526) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (31686) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_002
> I (31686) coreMQTT: Ack packet deserialized with result: MQTTSuccess.
> I (31686) coreMQTT: State record updated. New state=MQTTPublishDone.
> W (31696) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_004
> W (31706) sub_pub_unsub_demo: PLACEHOLDER_issues#112_005
> I (31706) sub_pub_unsub_demo: Publish 5 succeeded for task "SubPub0".
> I (31706) coreMQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.
> W (31726) sub_pub_unsub_demo: PLACEHOLDER_issues#112_007
```

 
start 166776--> end  170986-- 4200 msec

```
> I (166776) sub_pub_unsub_demo: Task "SubPub0" sending publish request to coreMQTT-Agent with message "SubPub0" on topic "/filter/SubPub0" with ID 29.
> W (166786) sub_pub_unsub_demo: PLACEHOLDER_issues#112_001
> W (166796) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_002
> W (166806) sub_pub_unsub_demo: PLACEHOLDER_issues#112_006
> I (166806) sub_pub_unsub_demo: Task "SubPub0" waiting for publish 29 to complete.
> W (168776) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (168776) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_004
> W (170786) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (170956) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_002
> I (170956) coreMQTT: Ack packet deserialized with result: MQTTSuccess.
> I (170956) coreMQTT: State record updated. New state=MQTTPublishDone.
> W (170966) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_004
> W (170966) sub_pub_unsub_demo: PLACEHOLDER_issues#112_005
> I (170976) sub_pub_unsub_demo: Publish 29 succeeded for task "SubPub0".
> I (170976) coreMQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.
> W (170986) sub_pub_unsub_demo: PLACEHOLDER_issues#112_007

```

aggarg, 2024-09-03:
Thank you for capturing and sharing these logs. It seems that there are multiple `sub_pub` tasks running which make the analysis difficult. Would you please change [the number of sub_pub tasks](https://github.com/FreeRTOS/iot-reference-esp32/blob/main/main/demo_tasks/sub_pub_unsub_demo/sub_pub_unsub_demo_config.h#L72) to 1 and change the [QoS](https://github.com/FreeRTOS/iot-reference-esp32/blob/main/main/demo_tasks/sub_pub_unsub_demo/sub_pub_unsub_demo_config.h#L67) to 0. Please capture the log again with these changes and share.

WilliamFrasson, 2024-09-03:
Hello @aggarg 
yes i confim that the log has only one task: #define CONFIG_GRI_SUB_PUB_UNSUB_DEMO_NUM_TASKS_TO_CREATE 1
and all the others demo disabled

ad double check this is a new log 

> 
> > `ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce2810,len:0x11b8
> load:0x403c8700,len:0x4
> load:0x403c8704,len:0xab8
> load:0x403cb700,len:0x2c2c
> entry 0x403c8890
> I (218) cpu_start: Multicore app
> I (227) cpu_start: Pro cpu start user code
> I (227) cpu_start: cpu freq: 160000000 Hz
> I (227) app_init: Application information:
> I (230) app_init: Project name:     iot-reference-esp32c3
> I (236) app_init: App version:      v202212.00-111-g6ace727-dirty
> I (243) app_init: Compile time:     Sep  2 2024 00:00:36
> I (249) app_init: ELF file SHA256:  90472addc...
> I (254) app_init: ESP-IDF:          v5.3-dirty
> I (259) efuse_init: Min chip rev:     v0.0
> I (264) efuse_init: Max chip rev:     v0.99
> I (269) efuse_init: Chip rev:         v0.2
> I (274) heap_init: Initializing. RAM available for dynamic allocation:
> I (281) heap_init: At 3FCA3010 len 00046700 (281 KiB): RAM
> I (287) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
> I (293) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
> I (299) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
> I (306) spi_flash: detected chip: generic
> I (310) spi_flash: flash io: dio
> I (314) sleep: Configure to isolate all GPIO pins in sleep state
> I (321) sleep: Enable automatic switching of GPIO sleep configuration
> I (328) coexist: coex firmware version: dab85ae96
> I (356) coexist: coexist rom version e7ae62f
> I (356) main_task: Started on CPU0
> I (366) main_task: Calling app_main()
> I (366) esp_secure_cert: Pre-provisioned partition information:
> I (366) esp_secure_cert: partition format: CUST_FLASH_TLV, partition name: esp_secure_cert
> I (376) main:
> Device Cert:
> Length: 1221
> -----BEGIN CERTIFICATE-----
> xxxxxxxxxxxxxxxxx
> -----END CERTIFICATE-----
> 
> I (486) main:
> CA Cert:
> Length: 1207
> -----BEGIN CERTIFICATE-----
> xxxxx
> -----END CERTIFICATE-----
> I (596) main:
> Private Key:
> Length: 1676
> -----BEGIN RSA PRIVATE KEY-----
> xxxx
> -----END RSA PRIVATE KEY-----
> 
> I (786) pp: pp rom version: e7ae62f
> I (786) net80211: net80211 rom version: e7ae62f
> I (796) wifi:wifi driver task: 3fcb01a8, prio:23, stack:6656, core=0
> I (806) wifi:wifi firmware version: 0caa81945
> I (806) wifi:wifi certification version: v7.0
> I (806) wifi:config NVS flash: enabled
[... 85 lines cut ...]
> W (24126) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_004
> I (24126) sub_pub_unsub_demo: Subscribe 4 for topic filter /filter/SubPub0 succeeded for task "SubPub0".
> I (24136) sub_pub_unsub_demo: Task "SubPub0" sending publish request to coreMQTT-Agent with message "SubPub0" on topic "/filter/SubPub0" with ID 5.
> W (24146) sub_pub_unsub_demo: PLACEHOLDER_issues#112_001
> W (24156) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_002
> W (24156) sub_pub_unsub_demo: PLACEHOLDER_issues#112_006
> I (24166) sub_pub_unsub_demo: Task "SubPub0" waiting for publish 5 to complete.
> W (26136) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (26136) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_004
> W (28146) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (28306) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_002
> I (28306) coreMQTT: Ack packet deserialized with result: MQTTSuccess.
> I (28306) coreMQTT: State record updated. New state=MQTTPublishDone.
> W (28316) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_004
> W (28316) sub_pub_unsub_demo: PLACEHOLDER_issues#112_005
> I (28326) sub_pub_unsub_demo: Publish 5 succeeded for task "SubPub0".
> W (28336) sub_pub_unsub_demo: PLACEHOLDER_issues#112_007
> I (28336) coreMQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.
> I (28346) coreMQTT: State record updated. New state=MQTTPubAckSend.
> I (28356) sub_pub_unsub_demo: Task "SubPub0" received: SubPub0
> I (28356) sub_pub_unsub_demo: Task "SubPub0" sending unsubscribe request to coreMQTT-Agent for topic filter: /filter/SubPub0 with id 6
> W (28376) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_002
> W (30366) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (30366) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_004
>

WilliamFrasson, 2024-09-03:
just  checking with QoS= 0   --> 44836  - 40836  = 4000 msec again

> I (40816) sub_pub_unsub_demo: Task "SubPub0" sending publish request to coreMQTT-Agent with message "SubPub0" on topic "/filter/SubPub0" with ID 8.
> W (40836) sub_pub_unsub_demo: PLACEHOLDER_issues#112_001
> W (40836) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_002
> W (40846) sub_pub_unsub_demo: PLACEHOLDER_issues#112_006
> I (40846) sub_pub_unsub_demo: Task "SubPub0" waiting for publish 8 to complete.
> W (42816) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (42816) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_004
> W (44826) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (44826) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_004
> W (44826) sub_pub_unsub_demo: PLACEHOLDER_issues#112_005
> I (44826) sub_pub_unsub_demo: Publish 8 succeeded for task "SubPub0".
> W (44836) sub_pub_unsub_demo: PLACEHOLDER_issues#112_007

[image]

WilliamFrasson, 2024-09-03:
the time lost seem in the double loop into: 
*  LOGW - "PLACEHOLDER_issues#112_003": When the command is picked up for processing - https://github.com/FreeRTOS/coreMQTT-Agent/blob/main/source/core_mqtt_agent.c#L569.
and
LOGW - "PLACEHOLDER_issues#112_004": When the command execution is complete - https://github.com/FreeRTOS/coreMQTT-Agent/blob/main/source/core_mqtt_agent.c#L802.

aggarg, 2024-09-04:
> #define CONFIG_GRI_SUB_PUB_UNSUB_DEMO_QOS_LEVEL 0
#define CONFIG_GRI_SUB_PUB_UNSUB_DEMO_NUM_TASKS_TO_CREATE 1

Did you change this flags manually? I think you are supposed to use `idf.py menuconfig`?

Can you try to increase the [MQTT Agent task's priority](https://github.com/FreeRTOS/iot-reference-esp32/blob/main/main/networking/mqtt/core_mqtt_agent_manager_config.h#L107)?

WilliamFrasson, 2024-09-04:
i used menuconfig (project is in espress-ide )
as double check i printed now the Qos in PLACEHOLDER_issues#112_001

I (38776) sub_pub_unsub_demo: Subscribe 7 for topic filter /filter/SubPub0 succeeded for task "SubPub0".
I (38786) sub_pub_unsub_demo: Task "SubPub0" sending publish request to coreMQTT-Agent with message "SubPub0" on topic "/filter/SubPub0" with ID 8.
> W (38806) sub_pub_unsub_demo: PLACEHOLDER_issues#112_001, QoS =0
W (38806) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_002
W (38816) sub_pub_unsub_demo: PLACEHOLDER_issues#112_006
I (38816) sub_pub_unsub_demo: Task "SubPub0" waiting for publish 8 to complete.
W (40786) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
W (40786) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_004
W (42796) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
W (42796) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_004
> W (42796) sub_pub_unsub_demo: PLACEHOLDER_issues#112_005
I (42796) sub_pub_unsub_demo: Publish 8 succeeded for task "SubPub0".
W (42806) sub_pub_unsub_demo: PLACEHOLDER_issues#112_007

Also changed the CONFIG_GRI_MQTT_AGENT_TASK_PRIORITY to 6 (using menuconfig) but the result seem the same

I (11886) sub_pub_unsub_demo: Unsubscribe 3 for topic filter /filter/SubPub0 succeeded for task "SubPub0".
I (11896) sub_pub_unsub_demo: Task "SubPub0" completed a loop. Delaying before next loop.
W (13906) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
W (15926) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
W (17946) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
W (19966) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
I (21896) sub_pub_unsub_demo: Task "SubPub0" sending subscribe request to coreMQTT-Agent for topic filter: /filter/SubPub0 with id 4
W (21896) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_002
W (21976) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
W (22086) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_004
I (22096) sub_pub_unsub_demo: Subscribe 4 for topic filter /filter/SubPub0 succeeded for task "SubPub0".
I (22096) sub_pub_unsub_demo: Task "SubPub0" sending publish request to coreMQTT-Agent with message "SubPub0" on topic "/filter/SubPub0" with ID 5.
W (22106) sub_pub_unsub_demo: PLACEHOLDER_issues#112_001, QoS =0
W (22116) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_002
W (22126) sub_pub_unsub_demo: PLACEHOLDER_issues#112_006
I (22126) sub_pub_unsub_demo: Task "SubPub0" waiting for publish 5 to complete.
W (24106) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
W (24106) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_004
W (24106) sub_pub_unsub_demo: PLACEHOLDER_issues#112_005
I (24106) sub_pub_unsub_demo: Publish 5 succeeded for task "SubPub0".
W (24116) sub_pub_unsub_demo: PLACEHOLDER_issues#112_007
W (24326) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_002

WilliamFrasson, 2024-09-04:
hello @aggarg 
going deeper seem that there is a forced timeout of 2000 msec between #2 to #3 and 3# to #4

[image]

so changing recvTimeoutMs  in timeouts of
path	/iot-reference-esp32c3/components/esp-aws-iot/libraries/coreMQTT/port/network_transport/network_transport.c

> line 14 from 2000 to  Timeouts_t timeouts = { .connectionTimeoutMs = 4000, .sendTimeoutMs = 10000, .recvTimeoutMs = 500/*WFR 2000*/ };

now the Publish take 83026 - 82026  = 1000 sec instead of 4000 msec

**so it seem something related to the network_transport.c recvTimeoutMs value.**

- @aggarg  Why this received always have to goes in timeout? 
- @aggarg  is it reliabled to set it to a value like 100 msec?

 
here the log 

> I (71456) sub_pub_unsub_demo: Task "SubPub0" completed a loop. Delaying before next loop.
> W (71956) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (71956) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_004
> W (72476) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (72996) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (73516) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (74036) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (74556) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (75076) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (75596) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (76116) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (76636) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (77156) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (77676) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (78196) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (78716) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (79236) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (79756) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (80276) CORE_M QTT_AGENT: PLACEHOLDER_issues#112_003
> W (80796) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003
> W (81316) CORE_MQTT_AGENT: PLACEHOLDER_issues#112_003

aggarg, 2024-09-09:
> going deeper seem that there is a forced timeout of 2000 msec between https://github.com/FreeRTOS/iot-reference-esp32/pull/2 to https://github.com/FreeRTOS/iot-reference-esp32/pull/3 and 3# to https://github.com/FreeRTOS/iot-reference-esp32/pull/4

Thank you for diving deep and finding that! 

> so it seem something related to the network_transport.c recvTimeoutMs value.
    @aggarg Why this received always have to goes in timeout?
    @aggarg is it reliabled to set it to a value like 100 msec?

This should be fine and this PR seems to be doing that - https://github.com/FreeRTOS/iot-reference-esp32/pull/104. 

Does this PR solve your issue?

WilliamFrasson, 2024-09-11:
Hello. Confirmed PR  https://github.com/FreeRTOS/iot-reference-esp32/pull/104 resolve the issue.
now QoS=0 publish  take 100-200 msec max
thanks for the support

closed 2024-09-11 by WilliamFrasson

aggarg, 2024-09-11:
Thank you for doing all the debugging!