FreeRTOS/iot-reference-esp32 #118: [DOC]Unable to boot firmware with the Secure Boot enabled bootloader

GASEN1216, 2024-10-11:
I followed the steps in the document(https://github.com/FreeRTOS/iot-reference-esp32/blob/main/UseSecurityFeatures.md), but when I got to step 6, something went wrong. After burning it in, it prompted the following error message. I don't know why? Can anyone help me?
```
invalid header: 0x8b2289f0
invalid header: 0x8b2289f0
invalid header: 0x8b2289f0
invalid header: 0x8b2289f0
invalid header: 0x8b2289f0
invalid header: 0x8b2289f0
invalid header: 0x8b2289f0
invalid header: 0x8b2289f0
invalid header: 0x8b2289f0
invalid header: 0x8b2289f0
invalid header: 0x8b2289f0
invalid header: 0x8b2289f0
invalid header: 0x8b2289f0
invalid header: 0x8b2289f0
invalid header: 0x8b2289f0
invalid header: 0x8b2289f0
invalid header: 0x8b2289f0
invalid header: 0x8ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x2d (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048841
0x40048841: uart_tx_one_char_uart in ROM
```

rawalexe, 2024-10-11:
Can you please provide more logs and the steps you followed?

jasonpcarroll, 2024-10-11:
HI @GASEN1216, 
Originally this repository was created for the C3 - although I imagine it should work for the S3 as well. The errors you are showing are similar to ones one would see if they only flashed the application and not the bootloader. That could possibly be your problem. I will try to source an S3 myself and follow the same steps. This may take time.

Best,
Jason Carroll

GASEN1216, 2024-10-12:
> Can you please provide more logs and the steps you followed?
Hi, @rawalexe ,
The following logs are the ones that keep looping and reporting the errors I gave above.
Step 1: I have successfully run the example without secure mode

Step 2: I turned on the two options mentioned in the document

Step 3: I turned on the flash encrypted project

Step 4: I burned the encrypted certificate as the document said

Step 5: I generated the certificate and set up secure boot

Step 6: I built and burned the project

Is it because I did not build the bootloader with the certificate and secure boot set up in step 5, and the bootloader burned was the bootloader before the secure boot was built, which caused my error?

Thanks,
Gasen

GASEN1216, 2024-10-12:
> HI @GASEN1216, Originally this repository was created for the C3 - although I imagine it should work for the S3 as well. The errors you are showing are similar to ones one would see if they only flashed the application and not the bootloader - see 5.1 step 2. That could possibly be your problem. I will try to source an S3 myself and follow the same steps. This may take time.
> 
> Best, Jason Carroll

Hi @jasonpcarroll ,
Thank you!
Is it because I did not build the bootloader with the certificate and secure boot set up in step 5, and the bootloader burned was the bootloader before the secure boot was built, which caused my error?
In other words, is it because I only ran the idf.py bootloader and did not build the project that caused the error?

jasonpcarroll, 2024-10-14:
After running idf.py bootloader you need to run the command it prints out with (PORT) and (BAUD) replaced with your port and baud rate. In our example, it shows the command, but it will probably print out a slightly different command for you since you are on an s3. When you use secure boot - idf.py flash will not flash the bootloader. You need to manually do it using the command like the one below that it prints out when you run idf.py bootloader. At least that is my understanding. I hope this helps.
```
C:/Users/user/.espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe`  C:/Users/user/Desktop/esp-idf-6/components/esptool_py/esptool/esptool.py --chip esp32c3 --port=(PORT) --baud=(BAUD) --before=default_reset --after=no_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 C:/FreeRTOS-Repositories/lab-iot-reference-esp32c3/build/bootloader/bootloader.bin
```

jasonpcarroll, 2024-10-14:
@dhavalgujar - would you mind taking a look here?

dhavalgujar, 2024-10-15:
Hi @GASEN1216,
Based on your log, it is likely that the first boot up sequence (with flash encryption enabled) that involves encrypting each partition got stopped abruptly.
```
I (168) boot: Checking flash encryption...
I (173) efuse: Batch mode of writing fields is enabled
I (179) flash_encrypt: Generating new flash encryption key...
I (188) efuse: Writing EFUSE_BLK_KEY0 with purpose 4
W (194) flash_encrypt: Not disabling UART bootloader encryption
I (197) flash_encrypt: Disable UART bootloader cache...
I (203) flash_encrypt: Disable JTAG...
I (212) efuse: Batch mode. Prepared fields are committed
I (214) esp_image: segment 0: paddr=00000020 vaddr=3fcd0270 size=02598h (  9624)
I (223) esp_image: segment 1: paddr=000025c0 vaddr=403b6000 size=00878h (  2168)
I (230) esp_image: segment 2: paddr=00002e40 vaddr=403ba000 size=03dd4h ( 15828)
I (534) flash_encrypt: bootloader encrypted successfully
I (578) flash_encrypt: partition table encrypted and loaded successfully
I (578) flash_encrypt: Encrypting partition 1 at offset 0x10000 (length 0x1000)...
I (628) flash_encrypt: Done encrypting
I (629) esp_image: segment 0: paddr=00020020 vaddr=3c020020 size=08118h ( 33048) map
I (636) esp_image: segment 1: paddr=00028140 vaddr=3fc8fa30 size=023f4h (  9204)
I (640) esp_image: segment 2: paddr=0002a53c vaddr=40374000 size=05adch ( 23260)
I (651) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=1a710h (108304) map
I (675) esp_image: segment 4: paddr=0004a738 vaddr=40379adc size=05f48h ( 24392)
I (679) esp_image: segment 5: paddr=00050688 vaddr=600fe000 size=00010h (    16)
I (680) flash_encrypt: Encrypting partition 2 at offset 0x20000 (length 0x100000)...
I (11571) flash_encrypt: Done encrypting
I (11571) flash_encrypt: Encrypting partition 3 at offset 0x120000 (length 0x1000)...
I (11617) flash_encrypt: Done encrypting
I (11618) flash_encrypt: Flash encryption completed
I (11623) boot: Resetting with flash encryption enabled...
```

It should be possible to recover the device by re-flashing all the binaries with the `--encrypt` flag.

But, in order to better assess the situation, can you please provide the following:
1. Your sdkconfig file.
2. The exact commands that you ran.
3. The efuse summary of your ESP32-S3 device obtained by running `espefuse.py summary --port (PORT)`

GASEN1216, 2024-10-16:
> So first you need to run idf.py menuconfig and set the appropriate flags for secure boot to be enabled. THEN, you need to build the bootloader with idf.py bootloader. After running idf.py bootloader you need to run the command it prints out with (PORT) and (BAUD) replaced with your port and baud rate. In our example, it shows the command below, but it will probably print out a slightly different command for you since you are on an s3. When you use secure boot - idf.py flash will not flash the bootloader. You need to manually do it using the command like the one below that it prints out when you run idf.py bootloader. At least that is my understanding. I hope this helps.因此，首先您需要运行 idf.py menuconfig 并设置适当的标志以启用安全启动。然后，您需要使用 idf.py bootloader 构建 bootloader。运行 idf.py 引导加载程序后，您需要运行它打印出来的命令，其中 （PORT） 和 （BAUD） 替换为您的端口和波特率。在我们的示例中，它显示了下面的命令，但它可能会为您打印出一个略有不同的命令，因为您使用的是 s3。当您使用安全启动时 - idf.py flash 不会刷写 bootloader。您需要使用下面的命令手动执行此操作，当您运行 bootloader 时，它会打印 idf.py 该命令。至少我是这么理解的。我希望这有帮助。
> 
> ```
> C:/Users/user/.espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe`  C:/Users/user/Desktop/esp-idf-6/components/esptool_py/esptool/esptool.py --chip esp32c3 --port=(PORT) --baud=(BAUD) --before=default_reset --after=no_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 C:/FreeRTOS-Repositories/lab-iot-reference-esp32c3/build/bootloader/bootloader.bin
> ```

Hi @jasonpcarroll ,
Thank you very much for your help. I also flashed manually before but I only ran the second line of command under "Secure boot enabled, so bootloader not flashed automatically." This time I tried to sign the bootloader first and then flash it manually and the chip worked.

jasonpcarroll, 2024-10-17:
Awesome, was there any discrepancy in the documentation or was it simply a missed step?

GASEN1216, 2024-10-17:
> Awesome! Just to be clear, was there any discrepancy in the documentation?

The documentation is correct. I don't know if I misunderstood it, but when I first read it I thought it was only the second block that needed to be run. It would be nice if you could add a reminder to the documentation to remind people not to miss the first block.

closed 2024-10-21 by GASEN1216