FreeRTOS/iot-reference-esp32 #119: [Feature Request] I want to use wifi and ethernet at the same time, what should I do?

GASEN1216, 2024-10-14:
I read the source code about wifi in core_mqtt_agent_manager.c, but I am not sure how to modify it to adapt to Ethernet to run the three tasks in demo_tasks. Can anyone give me some advice? Thank you very much

rawalexe, 2024-10-14:
Hello @GASEN1216, what board are you using?

GASEN1216, 2024-10-16:
> Hello @GASEN1216, what board are you using?

Hi @rawalexe ,
I use esp32s3

GASEN1216, 2024-10-22:
I tried to modify the code a little bit, now it can support Ethernet connection.
```
#define ETHERNET_CONNECTED_BIT              (1 << 4)
#define ETHERNET_DISCONNECTED_BIT           (1 << 5)

static void prvEthernetEventHandler(void *pvHandlerArg, esp_event_base_t xEventBase, int32_t lEventId, void *pvEventData);

static void prvEthernetEventHandler(void *pvHandlerArg, esp_event_base_t xEventBase, int32_t lEventId, void *pvEventData)
{
    (void) pvHandlerArg;
    (void) pvEventData;

    if (xEventBase == ETH_EVENT) {
        switch (lEventId) {
            case ETHERNET_EVENT_DISCONNECTED:
                ESP_LOGI(TAG, "Ethernet disconnected.");

                xEventGroupClearBits(xNetworkEventGroup, ETHERNET_CONNECTED_BIT);
                break;

            default:
                break;
        }
    } else if (xEventBase == IP_EVENT) {
        switch (lEventId) {
            case IP_EVENT_ETH_GOT_IP:
                ESP_LOGI(TAG, "Ethernet connected.");

                xEventGroupSetBits(xNetworkEventGroup, ETHERNET_CONNECTED_BIT);
                break;

            default:
                break;
        }
    } else {
        ESP_LOGE(TAG, "Ethernet event handler received unexpected event base.");
    }
}

BaseType_t xCoreMqttAgentManagerStart( NetworkContext_t * pxNetworkContextIn )
{

     // Other codes......
     
     if( xRet != pdFAIL )
    {
        // WIFI IP EVENT
        xEspErrRet = esp_event_handler_instance_register( IP_EVENT,
                                                          ESP_EVENT_ANY_ID,
                                                          prvWifiEventHandler,
                                                          NULL,
                                                          NULL );
        if( xEspErrRet == ESP_OK )
        {
            xEspErrRet = esp_event_handler_instance_register( WIFI_EVENT,
                                                          ESP_EVENT_ANY_ID,
                                                          prvWifiEventHandler,
                                                          NULL,
                                                          NULL );    
            if( xEspErrRet != ESP_OK )  {
            ESP_LOGE( TAG,
                      "Failed to register WiFi event handler with WiFi events." );}                                                             
        }else {
            ESP_LOGE( TAG,
                      "Failed to register WiFi event handler with IP events." );            
        }

        // Ethernet IP EVENT
        xEspErrRet = esp_event_handler_instance_register( IP_EVENT,
                                                          ESP_EVENT_ANY_ID,
                                                          prvEthernetEventHandler,
                                                          NULL,
                                                          NULL );
        if( xEspErrRet == ESP_OK )
        {
            xEspErrRet = esp_event_handler_instance_register( ETH_EVENT,
                                                          ESP_EVENT_ANY_ID,
                                                          prvEthernetEventHandler,
                                                          NULL,
                                                          NULL );    
            if( xEspErrRet != ESP_OK )  {
            ESP_LOGE( TAG,
                      "Failed to register ETH event handler with ETH events." );}                                                             
        }else {
            ESP_LOGE( TAG,
                      "Failed to register ETH event handler with IP events." );            
        }        

        // PPP...

        // Check if the network is established
        if( xEspErrRet != ESP_OK )  {
            xRet = pdFAIL;
        } 
    }    
    
    // Other codes......   
    
}
```

Can I write it this way without causing problems in the future? It works fine on my ESP32S3 for now.

closed 2024-10-25 by GASEN1216