FreeRTOS/iot-reference-esp32 #126: [BUG] No rollback when migrating from [202210.01-LTS]

Fienberber, 2025-01-30:
Hello, 

I'm migrating from to the esp-aws-iot [202406.01-LTS] from [202210.01-LTS].
The previous integration was working properly so I know this is not an AWS issue.

We are using an ESP32C3 on our board. 

Using the example works great. But I have issues while transitioning the devices from the old method. 
I think there is an issue with the NVS. After the first boot on the firmware using esp-aws-iot [202406.01-LTS], the OTA flags are at ESP_OTA_IMG_NEW. 

This prevent proper response to AWS and retry the OTA. The device loop OTAs indefinitely. 
To fix this, I added custom flags to set the flag as ESP_OTA_IMG_PENDING_VERIFY if needed. 

My issue now is with the rollback feature. If I OTA a bad firmware (crashing before partition validation) the board restart on the same partition. 

During my tests, flashed using esp-aws-iot [202406.01-LTS] and transitioning to the same version, the rollback function works great. 

Somewhere the version migration does something to the NVS but I can't find where. 

Thanks in advance for you help.

ig15, 2025-01-31:
Thanks @Fienberber for bringing up the issue, we will look into it and get back to you shortly.

Fienberber, 2025-03-24:
Hi, 

Any update on this behavior ? (pun intended)