FreeRTOS/iot-reference-esp32 #127: [BUG] Build Fails

orlyprofili, 2025-03-05:
**Describe the bug**
building main fails. 

**System information**
- Hardware board: esp32c3 (also tried different targets)
- IDE used: vscode w/esp-idf or idf.py in terminal
- Operating System: macos
- Code version: main or commit hash fca362f6494002b54aac26da4f77c9235bf3793c 

steps:
1. clone recursively
2. set target (tried c3, c6, s3)
***NOTE*** 
I skipped this step:
2.3 Provision the ESP32-C3 with the private key, device certificate and CA certificate in Development M
****
3. build

get 3 of these in this same file. 

MQTTFileDownloader_cbor.c:117:50: error: passing argument 2 of 'cbor_value_get_int' from incompatible pointer type [-Wincompatible-pointer-types]
  117 |         cborResult = cbor_value_get_int( &value, ( int32_t * ) fileId );
      |                                                  ^~~~~~~~~~~~~~~~~~~~
      |                                                  |
      |                                                  int32_t * {aka long int *}

I don't believe provisioning is a necessary step to successfully building, maybe I'm missing a key point here?

I noticed that main does fail build in the github workflow, but fca362f6494002b54aac26da4f77c9235bf3793c passed (Sep 9, 2024).  This commit still fails for me.  Whether building with vscode+esp-idf or using idf.py.

closed 2025-03-05 by orlyprofili

imdahisaria, 2025-03-19:
Hi am tried to compile and I am getting this same error.
Any fix for this ?

`development/iot-reference-esp32/com-module/components/esp-aws-iot/libraries/aws-iot-core-mqtt-file-streams-embedded-c/aws-iot-core-mqtt-file-streams-embedded-c/source/MQTTFileDownloader_cbor.c: In function 'CBOR_Decode_GetStreamResponseMessage':

/home/rushil/Desktop/development/iot-reference-esp32/com-module/components/esp-aws-iot/libraries/aws-iot-core-mqtt-file-streams-embedded-c/aws-iot-core-mqtt-file-streams-embedded-c/source/MQTTFileDownloader_cbor.c:117:50: error: passing argument 2 of 'cbor_value_get_int' from incompatible pointer type [-Wincompatible-pointer-types]

  117 |         cborResult = cbor_value_get_int( &value, ( int32_t * ) fileId );

      |                                                  ^~~~~~~~~~~~~~~~~~~~

`

orlyprofili, 2025-03-19:
> Hi am tried to compile and I am getting this same error.
> Any fix for this ?
>
> ` [...]

Sounds like you're using a newer version of esp-idf. Try with 5.2.5