FreeRTOS/iot-reference-esp32 #129: [Feature Request] Device Shadow

brownjar, 2025-05-06:
Demonstrate how to include a device shadow with the led control example.

kar-rahul-aws, 2025-05-14:
Hi @brownjar 
Thank you for your query . We currently do not have an example of device shadow integration with the esp32 -iot-reference.  
As a starting point, you can take a look at the Device shadow [Windows Simulator demo](https://github.com/FreeRTOS/FreeRTOS/tree/main/FreeRTOS-Plus/Demo/AWS/Device_Shadow_Windows_Simulator/Device_Shadow_Demo) ( read the documentation [here](https://www.freertos.org/Documentation/03-Libraries/04-AWS-libraries/03-AWS-IoT-Device-Shadow/04-Device-shadow-demo) ) .

tony-josi-aws, 2025-05-22:
@brownjar 

I'm closing this issue as we dont have immediate plans to include hardware demos for device shadow and there is a working simulated demo as posted by @kar-rahul-aws 

Please feel free to open another issue/reopen if you face any issues.

closed 2025-05-22 by tony-josi-aws